mani-skill/ManiSkill #956: VK_ERROR_INITIALIZATION_FAILED: Shader library compile failed on MacOS

rrzhang139, 2025-03-21:
- OS: macOS version 14.4 Sonoma
- Chip: Apple M3 Pro
- Python: 3.11

I'm testing SAPIEN rendering on macOS (Apple M3 Pro). The example hello_world.py crashes due to shader compilation failure with MoltenVK. Seems related to sampler limits on Metal. 

```
[mvk-error] VK_ERROR_INITIALIZATION_FAILED: Shader library compile failed (Error code 3)
error: 'sampler' attribute parameter is out of bounds: must be between 0 and 15
...
IndexError: unordered_map::at: key not found
```

StoneT2000, 2025-03-21:
From running the `python -m sapien.example.hello_world` command?

StoneT2000, 2025-03-21:
@comoco-xiao any idea? This is from running `python -m sapien.example.hello_world` with the latest mac wheel.

comoco-xiao, 2025-03-21:
Try to install the vulkansdk-1.3.290.0 on Mac before.

# Vulkan Installation
https://sdk.lunarg.com/sdk/download/1.3.290.0/mac/vulkansdk-macos-1.3.290.0.dmg

# Set up the runtime environment (default sdk_path=~/VulkanSDK)
export VULKAN_SDK=sdk_path/1.3.290.0/macOS
export PATH=$VULKAN_SDK/bin:$PATH
export VK_ICD_FILENAMES=$VULKAN_SDK/share/vulkan/icd.d/MoltenVK_icd.json
export VK_LAYER_PATH=$VULKAN_SDK/share/vulkan/explicit_layer.d
export DYLD_LIBRARY_PATH=$VULKAN_SDK/lib:$DYLD_LIBRARY_PATH

# Vulkan guide for MacOS
https://vulkan.lunarg.com/doc/view/1.3.290.0/mac/getting_started.html

lenant, 2025-04-16:
Getting the same error on Intel mac with GPU AMD Radeon Pro 560X. Vulkan is installed.

comoco-xiao, 2025-04-16:
Set environment variable
export MVK_CONFIG_USE_METAL_ARGUMENT_BUFFERS=1

lenant, 2025-04-16:
> Set environment variable export MVK_CONFIG_USE_METAL_ARGUMENT_BUFFERS=1

Didn't change anything for me

mukundkk, 2025-07-24:
Have been running into the same issue (macOS 14.7.1 Sonoma, M1 Pro chip, Python 3.11, Vulkan 1.3.290.0 installed) and setting the `MVK_CONFIG_USE_METAL_ARGUMENT_BUFFERS` env variable didn't change anything for me. Any updates / potential solutions would be really appreciated!

cmcgartoll, 2025-08-01:
@mukundkk Got it working on my M1 Pro Max (macOS 14.4.1 Sonoma, python 3.13, Vulkan 1.3.290.0) with the env vars that @comoco-xiao mentioned. I used the nightly wheel for Sapien (sapien-3.0.0.dev20250303+291f6a77-cp313-cp313-macosx_12_0_universal2.whl). 

Also note, I was running into this error before installing Vulkan, which is different than @rrzhang139, so this might not be helpful for your case:

```
Enabled instance extension VK_KHR_get_physical_device_properties2 is not present at instance level.
2025-07-31 18:25:39,662 - mani_skill  - WARNING - Detected MacOS system, forcing render backend to be sapien_cpu and render device to be MacOS compatible.
[2025-07-31 18:25:39.721] [svulkan2] [error] Validation layer VK_LAYER_KHRONOS_validation is not available and skipped.
[2025-07-31 18:25:39.722] [svulkan2] [error] Extension VK_KHR_external_memory_capabilities is not available. Interop with external library is disabled.
[2025-07-31 18:25:39.722] [svulkan2] [error] Extension VK_KHR_external_semaphore_capabilities is not available. Interop with external library is disabled.
Enabled instance extension VK_KHR_get_physical_device_properties2 is not present at instance level.
[2025-07-31 18:25:39.722] [svulkan2] [error] Your GPU driver does not support Vulkan. You may not use the renderer for rendering.
Traceback (most recent call last):
  ...
  File "/opt/miniconda3/envs/maniskill/lib/python3.13/site-packages/mani_skill/envs/sapien_env.py", line 890, in reset
    self._reconfigure(options)
    ~~~~~~~~~~~~~~~~~^^^^^^^^^
  File "/opt/miniconda3/envs/maniskill/lib/python3.13/site-packages/mani_skill/envs/sapien_env.py", line 724, in _reconfigure
    self._setup_scene()
    ~~~~~~~~~~~~~~~~~^^
  File "/opt/miniconda3/envs/maniskill/lib/python3.13/site-packages/mani_skill/envs/sapien_env.py", line 1171, in _setup_scene
    sapien.Scene([physx_system, sapien.render.RenderSystem(self._render_device)])
                                ~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^
RuntimeError: vk::createInstanceUnique: ErrorIncompatibleDriver
```