mani-skill/ManiSkill #957: Collision detection and planning issue when adding obstacle with mplib in pickcube task

vvviiiccc2001, 2025-03-21:
Hi! I'm trying to add obstacles in the ManiSkill pickcube task following the mplib collision avoidance demo. The obstacle appears visually but collision detection with the robotic arm seems inconsistent.

**Steps:**

1. Added a box obstacle using:

```
builder = env.unwrapped.scene.create_actor_builder()
size = (0.01, 0.9, 0.05) 
pos = [0, 0, size[2]] 

builder.add_box_collision(half_size=size)  # Collision component
builder.add_box_visual(
    half_size=size,
    material=sapien.render.RenderMaterial(base_color=[0.5, 0.5, 0.5, 1])
)
obstacle = builder.build_kinematic(name="wall")
```

2. Used RRT-Connect for motion planning:

` planner.move_to_pose_with_RRTConnect(target_pose)`

3. Verified obstacle visualization by plotting `self.all_collision_pts` - obstacle point cloud appears correctly.

**Observed Behavior:**

1. The planner considers trajectories successful even when the arm visually collides with the obstacle

- `planner.check_for_env_collision()` returns empty collision set

- `env.scene.get_contacts()` shows actual collision contacts, which is correct

2. The path planner can navigate around obstacles to reach the target point, but does not consider collisions between the robotic arm and the environment.

Has anyone had successful experience with obstacle avoidance using mplib in Maniskill?

StoneT2000, 2025-03-25:
What version of mplib are you using?

Maybe @Lexseal (mplib maintainer) has an idea

Lexseal, 2025-03-26:
When you say the arm visually collides with the obstacle, what part of the arm is colliding and by how much?

vvviiiccc2001, 2025-03-26:
> When you say the arm visually collides with the obstacle, what part of the arm is colliding and by how much?

Hi, you can check the video here: 
https://drive.google.com/file/d/1EqM-S52uZlVbsDOC3UeZE6XaIHPZ31VT/view?usp=drive_link
at 8s and 14s, two collisions happened, between the end-effector and obstacle.

Lexseal, 2025-03-28:
Hmm but nothing is falling over it. Are you saying the arm is colliding with the wooden structure you put around the ball?

StoneT2000, 2025-05-20:
Closing for now as this is a mplib related issue.

closed 2025-05-20 by StoneT2000