mani-skill/ManiSkill #958: Question about demo_manual_control.py script (In ManiSkill/mani_skill/examples/demo_manual_control.py)

dongyingyibadao, 2025-03-23:
Hi! I’ve been analyzing the script “demo_manual_control.py” to understand how the robot in Maniskill can be controlled via keyboard. My plan is to create an interface for other devices, such as joysticks, etc. However, when I run the script and press 'w' to move the robot forward, I notice that not only does the robot move forward, but its mechanical arm also moves. Why is the motion of the arm not decoupled from the robot's movement?

Another issue arises in this part of the code:

```python
if has_base:
    if key == "w":  # forward
        base_action[0] = 1
    elif key == "s":  # backward
        base_action[0] = -1
    elif key == "q":  # rotate counterclockwise
        base_action[2] = 1
    elif key == "e":  # rotate clockwise
        base_action[2] = -1
    elif key == "z":  # lift
        body_action[2] = 1
    elif key == "x":  # lower
        body_action[2] = -1
    elif key == "v":  # rotate head left
        body_action[0] = 1
    elif key == "b":  # rotate head right
        body_action[0] = -1
    elif key == "n":  # tilt head down
        body_action[1] = 1
    elif key == "m":  # rotate head up
        body_action[1] = -1
```

Here, `base_action` is defined as `base_action = np.zeros([2])  # hardcoded for fetch robot`, meaning it doesn’t actually have a `base_action[2]`. To fix this, I modified the code as follows:

```python
if has_base:
    if key == "w":  # forward
        base_action[0] = 1
    elif key == "s":  # backward
        base_action[0] = -1
    elif key == "t":  # rotate counterclockwise
        base_action[1] = 1
    elif key == "e":  # rotate clockwise
        base_action[1] = -1
    elif key == "z":  # lift
        body_action[2] = 1
    elif key == "x":  # lower
        body_action[2] = -1
    elif key == "v":  # rotate head left
        body_action[0] = 1
    elif key == "b":  # rotate head right
        body_action[0] = -1
    elif key == "n":  # tilt head down
        body_action[1] = 1
    elif key == "m":  # rotate head up
        body_action[1] = -1
```

Can you confirm if this modification is correct? Also, why does pressing 'w' still affect other part of tobot?

---

StoneT2000, 2025-03-25:
@arth-shukla any idea? for the fetch robot

StoneT2000, 2025-04-03:
Ok this script is pretty crude to begin with (you shouldn't teleoperate with a keyboard anyway) and it is ported from maniskill2 which was a little messy.

One guess we had was the end effector control IK solving is failing to work because it doesn't move with the robot? You might need to hack with this a bit