mani-skill/ManiSkill #964: How to create a hollow object

ThisisXXZ, 2025-03-26:
I tried to simulate picking a cube into a cylinder bin.
However, I can only add collision to the whole cylinder, thus the cube can only be placed on the bin, but not in the bin.
I came up with the following method to subtract a inner bin from the outer bin so that the density of the bin inside is 0, with the wall thickness being the `bin_outer_radius - bin_inner radius`, but it still doesn't work.
```
        builder = self.scene.create_actor_builder()
        builder.add_cylinder_collision(
            radius=self.bin_outer_radius,
            half_length=self.bin_half_length,
            density=1000
        )
        builder.add_cylinder_visual(
            radius=self.bin_outer_radius,
            half_length=self.bin_half_length,
            material=sapien.render.RenderMaterial(
                base_color=np.array([0.2, 0.8, 0.2, 0.6]),
            ),
        )

        self.bin_outer = builder.build_kinematic(name="bin_outer")
        
        builder = self.scene.create_actor_builder()
        builder.add_cylinder_collision(
            radius=self.bin_inner_radius,
            half_length=self.bin_half_length,
            density=-1000
        )
        builder.add_cylinder_visual(
            radius=self.bin_inner_radius,
            half_length=self.bin_half_length,
            material=sapien.render.RenderMaterial(
                base_color=np.array([0, 0, 0, 0]),
            ),
        )
        self.bin_inner = builder.build_kinematic(name="bin_inner")
```
Can you give me some advice on how to achieve that?

StoneT2000, 2025-03-26:
A hollow bin is just 5 box shapes, 4 walls and one bottom part
From a 2d perspective it looks like this (with 3 box shapes):

|_|

If you want a cylinder hollow shape then you need to make a collision mesh in e.g. blender with the desired hollow shape, then load it in here by load collision from mesh and use coacd decomposition. See https://sapien-sim.github.io/docs/api/sapien.wrapper.html#sapien.wrapper.actor_builder.ActorBuilder.add_multiple_convex_collisions_from_file for arguments, you can do decomposition="coacd".

ThisisXXZ, 2025-03-26:
> A hollow bin is just 5 box shapes, 4 walls and one bottom part From a 2d perspective it looks like this (with 3 box shapes):
>
> |_|
>
> If you want a cylinder hollow shape then you need to make a [...]

that'll do. thanks!

closed 2025-03-26 by ThisisXXZ