mani-skill/ManiSkill #965: Accumulting GPU memory when creating new meshes in env.reset

BlGene, 2025-03-26:
Hello,

Thanks for the great work with ManiSkill.

I have a question: I'm resetting an environment very often (200+ times), which is leading to memory being accumulated on the GPU. During my reset, I'm loading new meshes in the  `_load_scene` function. I can "fix" the memory accumulation by deleting and recreating the env. I've tried deleting the objects during the reset, but this does not help, how can I fix this problem?

Below are my snippets from the env.reset and the builder function.

```
    def reset(self, seed: Union[None, int, list[int]] = None, options: Union[None, dict] = None):
        if self.object_dataset == "clevr":
            pass
        elif self.object_dataset == "ycb" or self.object_dataset == "objaverse":
            if self.objects is not None:
                for object in self.objects:
                    object.remove_from_scene()
                    del object
        else:
            raise ValueError
        #set_trace()
        return super().reset(seed=seed, options=options)
 ```

```
    if add_collision:
        collision_file = glb_path
        # builder.add_nonconvex_collision_from_file # This does not support density
        builder.add_multiple_convex_collisions_from_file(
            filename=collision_file,
            scale=[scale] * 3,
            # material=physical_material,
            density=density,
            pose=obj_pose,
        )

    if add_visual:
        visual_file = glb_path
        builder.add_visual_from_file(
            filename=visual_file, scale=[scale] * 3, pose=obj_pose
        )
```

StoneT2000, 2025-03-26:
An environment reset shouldn't accumulate gpu memory, it should stay around the same.

Are you resetting normally or resetting with reconfigure True? Or did you set reconfigure freq to 1?

What do your load scene and initialize episode functions look like?

StoneT2000, 2025-03-27:
Hm I don't see anything that is problematic yet.

Best bet for yourself right now is to manually comment out the code and add them line by line until there is no more gpu memory leaks

fffffq00, 2025-05-12:
@StoneT2000  
I'm using the version `3.0.0b20` of `ManiSkill`.
I have also encountered a similar problem. My environment needs to be initialized frequently due to the high probability of failure. I found that as the epoch increases, my GPU memory usage increases until it overflows. I didn't use reconfigure because I was doing a partial reset. To alleviate this issue, I manually used `gc.collect()` to release GPU memory at the end of each rollout, and found that the memory was significantly reduced. 
I'm not sure what the specific reason is. My task and initialization method are similar to the `pick_cube` environment, only with more frequent resets. Can you provide some help?

StoneT2000, 2025-05-12:
@fffffq00 when running the original PickCube task do you have any memory overflow issues?

What do you mean exactly by frequent resets? Do you have an example reproducible script?

fffffq00, 2025-05-13:
@StoneT2000   Thanks for your reply.
I found no memory overflow when running the original `PickCube` task.
Frequent resets mean that I set the `fail` condition to be too large force between robot joints or collision with the table, so in the early stage of RL, the environment will be frequently reset. Because the project is relatively cumbersome, I have not yet organized the sample code.
However, I tested some situations again today, and I found that when I manually simplified almost all collision bodies into capsules and boxes, the GPU memory usage would be reduced, and memory would rarely accumulate. (Before this, in order to make the collision body similar to the visual, I split the visual mesh into multiple sub-colliders as the collision body during simulation)
Can you analyze the reason through this situation?

StoneT2000, 2025-05-13:
Do you have frequent resets with reconfiguration? Or no reconfiguration?

And what are the more complex collision shapes like? Any picture?

fffffq00, 2025-05-14:
@StoneT2000  Hi, 
I didn't use reconfiguration because I needed a partial reset.
I'm using a shadowhand. The middle part of the finger in this picture is the collision body I used before. 
[image]
Now I have simplified it to a box. After simplification, my code runs normally, but I will check my code further.
Thank you.

StoneT2000, 2025-05-14:
Do you mind sharing your full code for the shadowhand?

A number of people want to use it in our framework so I might try and help model it for people.

fffffq00, 2025-05-15:
@StoneT2000 Glad to hear that, I organized some code and put it together in a repository [shadowhand-maniskill](https://github.com/fffffq00/shadowhand-maniskill) where you can find the original configuration and my modified configuration of shadowhand. Hope it can help others.

StoneT2000, 2025-05-16:
I will move this discussion to a new appropriately named issue for shadow hand modelling in #1073

closed 2025-05-16 by StoneT2000

fffffq00, 2025-05-30:
> I found the cpu memory accumulated slowly when running the original PickCube task. About 0.2GB for 1M steps. It this normal?

@LiuYiKai6  Hi, do you use CPU to train your single environment? I haven't tried this, but when I run this task on a GPU, it initially accumulates some GPU memory, and then it doesn't accumulate anymore.