mani-skill/ManiSkill #967: When can we train sim2real koch arm

rrzhang139, 2025-03-26:
Hello, I am really excited about this new PR with the koch arm [sim2real](https://github.com/haosulab/ManiSkill/pull/942/files). Is it possible to train it on ppo yet? I tried to run this command but I am not sure if the environment is registered. 

```
python ppo_rgb.py  --env_id=KochGraspCube-v1 \
>     --num_envs=1024 --num-steps=16 --update_epochs=8 --num_minibatches=8 \
>     --total_timesteps=100_000_000 \
>     --exp-name="ppo-KochGraspCube-v1-rgb-0"
```

Will this also be supported on SO100 arm soon? Thanks and keep up the good work!

StoneT2000, 2025-03-26:
The PR is not ready, so don't expect it to work. We are cleaning code up and preparing to release a zero shot rgb sim2real tutorial soon.

We are supporting so100 sim2real as well. Mine just came in, testing with it soon:

[image]

StoneT2000, 2025-06-14:
Tutorial for that here: https://github.com/StoneT2000/lerobot-sim2real