mani-skill/ManiSkill #969: Support for Task and Motion Planning

ducmngx, 2025-03-27:
I'm looking to solve problems with the TAMP framework, which requires the use of symbolic languages (STRIP, PDDL, etc) and motion planner + IK solver. I am wondering if ManiSkill supports that. Judging from the documents, it seems like you focus more on end-to-end learning.

Thank you for your answer

ducmngx, 2025-03-27:
mlib seems to set base frame to be the same as world frame. I wonder for mobile manipulator, like Fetch, how exactly the navigation part will look like? How can I use a motion planner to achieve this instead of learning?

StoneT2000, 2025-03-27:
I'm not sure what the best solution is.

I myself have sometimes just wrote heuristics myself.

@lexseal have you tried mplib with mobile manipulators before?

ducmngx, 2025-03-28:
Just to confirm, do you guys support PDDL to define the scene for high level planning? If not, is there any plan for it? I really like what I see from the simulation so far so I hope it can be my main platform for future projects

StoneT2000, 2025-03-28:
PDDL is not something that is simulator specific, you can define your own PDDL as needed. We probably don't have time to provide examples for those at this moment

Lexseal, 2025-03-28:
> mlib seems to set base frame to be the same as world frame. I wonder for mobile manipulator, like Fetch, how exactly the navigation part will look like? How can I use a motion planner to achieve this instead of learning?

there is the ability to move the base frame. if you don't need to move the arm when you are traversing maybe this is enough?