mani-skill/ManiSkill #975: MPLib causing installation to fail.

Nabla7, 2025-04-02:
I'm using an AGX Orion with Ubuntu 22.04 and Python 3.10.12, just running "pip install --upgrade mani_skill torch"

Collecting mplib>=0.1.1
  Downloading mplib-0.2.1.tar.gz (51 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 51.9/51.9 KB 6.3 MB/s eta 0:00:00
  Installing build dependencies ... error
  error: subprocess-exited-with-error
  
  × pip subprocess to install build dependencies did not run successfully.
  │ exit code: 1
  ╰─> [7 lines of output]
      Collecting setuptools>=61.0
        Downloading setuptools-78.1.0-py3-none-any.whl (1.3 MB)
           ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.3/1.3 MB 8.1 MB/s eta 0:00:00
      Collecting setuptools-git-versioning<2
        Downloading setuptools_git_versioning-1.13.6-py3-none-any.whl (10 kB)
      ERROR: Could not find a version that satisfies the requirement libclang==11.0.1 (from versions: 11.1.0, 12.0.0, 13.0.0, 14.0.1, 14.0.6, 15.0.3, 15.0.6, 15.0.6.1, 16.0.0, 16.0.6, 17.0.6, 18.1.1)
      ERROR: No matching distribution found for libclang==11.0.1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× pip subprocess to install build dependencies did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.

Is this because libclang 11.0.1 is not available for my system, how can this be resolved?

Thanks in advance.

StoneT2000, 2025-04-03:
@lexseal can we relax the lib clang version requirement in mplib?

@Nabla7 for the time being only resolution is to pip install from github by cloning the repo, remove the mplib dependency from setup.py
then do a local pip install

Lexseal, 2025-04-04:
sure I can do that

StoneT2000, 2025-05-20:
@Lexseal has mplib relaxed the requirement yet?