mani-skill/ManiSkill #977: The issue about incorrect object's center of mass.

YZY14606, 2025-04-03:
Hello, I found that there might be some issues with obtaining the object's center of mass position.

The code for importing objects into the environment is as follows:
```
        bowl = self.scene.create_actor_builder()
        bowl.add_multiple_convex_collisions_from_file(
        decomposition = "coacd",
        filename = "/home/yzy/nus/database/shapenet/ShapeNetCore.v2/02992529/3a6a3db4a0174fddd2789f496481c83e/models/model_normalized.obj",
        scale=[0.12, 0.12, 0.12] 
        )
```
Then, I used  `env.unwrapped.obj.pose.raw_pose & env.unwrapped.obj.cmass_local_pose.raw_pose` to get the global position of object's center of mass. Meanwhile, I used `env.unwrapped.obj.get_collision_meshes()[0] & mesh.center_mass` to get the mesh's cmass.

I visualized `mesh's cmass`  and `global position of object's center of mass` in a picture. The result is as follows (The green point is `global position of object's center of mass` and the red point is `mesh's cmass`. The gray points are the boundary vertices of the mesh.):
[image]
It can be seen that these two points do not overlap, and the green point should be more accurate. **I cannot figure out the reason because the object's mass is uniform.**

**Note:** There are two .py files and the complete code are as follows.
The first file is script.py.
```
import argparse
import gymnasium as gym
import numpy as np
import env_creation_test
from mani_skill.envs.sapien_env import BaseEnv
from mani_skill.sensors.camera import CameraConfig
import trimesh
import trimesh.scene
import open3d as o3d
def parse_args(args=None):
    parser = argparse.ArgumentParser()
    parser.add_argument("-e", "--env-id", type=str, default="PushCup-yzy-test", help="The environment ID of the task you want to simulate")
    parser.add_argument("--cam-width", type=int, help="Override the width of every camera in the environment")
    parser.add_argument("--cam-height", type=int, help="Override the height of every camera in the environment")
    parser.add_argument(
        "-s",
        "--seed",
        type=int,
        help="Seed the random actions and environment. Default is no seed",
    )
    args = parser.parse_args()
    return args

#Compute the position of object
def compute_cmass_global_pose(cmass_local_pose,object_pose):
    #Trans the object_pose.q to matrix
    qx, qy, qz, qw = object_pose[0][-4:]
    rotation_matrix = np.array([
        [1 - 2*(qy**2 + qz**2), 2*(qx*qy - qz*qw), 2*(qx*qz + qy*qw)],
        [2*(qx*qy + qz*qw), 1 - 2*(qx**2 + qz**2), 2*(qy*qz - qx*qw)],
        [2*(qx*qz - qy*qw), 2*(qy*qz + qx*qw), 1 - 2*(qx**2 + qy**2)]
    ])

    local_position = np.array(cmass_local_pose[:3])

    world_position = np.dot(rotation_matrix, local_position) + np.array(object_pose[0][0:3])
    return world_position

def main(args):
    if args.seed is not None:
        np.random.seed(args.seed)
    sensor_configs = dict()
    if args.cam_width:
        sensor_configs["width"] = args.cam_width
    if args.cam_height:
        sensor_configs["height"] = args.cam_height
    env: BaseEnv = gym.make(
        args.env_id,
        obs_mode="pointcloud",
        reward_mode="none",
[... 224 lines cut ...]
        # some useful observation info for solving the task includes the pose of the tcp (tool center point) which is the point between the
        # grippers of the robot
        obs = dict(
            tcp_pose=self.agent.tcp.pose.raw_pose,
        )
        if self.obs_mode_struct.use_state:
            # if the observation mode requests to use state, we provide ground truth information about where the cube is.
            # for visual observation modes one should rely on the sensed visual data to determine where the cube is
            obs.update(
                goal_pos=self.goal_region.pose.p,
                obj_pose=self.obj.pose.raw_pose,
            )
        return obs

    
    def compute_dense_reward(self, obs: Any, action: Array, info: Dict):
        # We also create a pose marking where the robot should push the cube from that is easiest (pushing from behind the cube)
        reward = 0
        return reward

    def compute_normalized_dense_reward(self, obs: Any, action: Array, info: Dict):
        # this should be equal to compute_dense_reward / max possible reward
        max_reward = 3.0
        return self.compute_dense_reward(obs=obs, action=action, info=info) / max_reward
```

YZY14606, 2025-04-07:
May I ask how the progress on this issue is going?

StoneT2000, 2025-04-22:
I don't immediately know why cmass local pose might be off. Is this gpu or cpu sim?

I guess solution for now is to use the mesh approach

YZY14606, 2025-04-23:
I used CPU simulation.