mani-skill/ManiSkill #982: Question about get_fsr_obj_impulse in allegro_touch.py

JiangBowen0008, 2025-04-04:
Hi, I have been trying to understand how the gpu_query_contact_pair_impulses are set up in allegro_touch.py. I am bit confused by the logic of get_fsr_obj_impulse.

In this line: https://github.com/haosulab/ManiSkill/blob/cc5b8425636c600d42ef194ed176316e4a21ab6d/mani_skill/agents/robots/allegro_hand/allegro_touch.py#L70
It seems to be repeating each object body by 2. Why is it `2` here, instead of `len(self.fsr_links) `?
Thanks!

StoneT2000, 2025-04-04:
I am not sure actually, this was written originally by @yzqin 

The code is somewhat outdated at this point however, we have a different cleaner API for pairwise contact forces now and should probably update that code anyway.

StoneT2000, 2025-04-04:
https://maniskill.readthedocs.io/en/latest/user_guide/tutorials/custom_tasks/advanced.html#contact-forces-on-the-gpu-cpu docs on fetching contact forces to do the same thing with less code

JiangBowen0008, 2025-04-07:
Understood. Yes the code inside the link works for me. Thanks!

closed 2025-04-07 by JiangBowen0008