mani-skill/ManiSkill #987: [Question] How to deploy a pretrained policy?

Frozenkiddo, 2025-04-05:
I have trained my own diffusion policy under the lerobot framework. Is there any script or module to use if I wanna deploy it in maniskill env?

StoneT2000, 2025-04-05:
You will need to write your own code with the gym setup to interact with the environments.

The other thing is if you trained with lerobot, did you train it on the maniskill task or some other task? If it’s not trained on the maniskill task data it probably won’t work.

I recommend reading https://maniskill.readthedocs.io/en/latest/user_guide/learning_from_demos/setup.html for how to setup imitation learning for training and evaluation.

Frozenkiddo, 2025-04-05:
Much appreciation for your prompt reply! To be specific, I trained the lerobot dp with unitree G1 dataset, since I can't find any humanoid dataset in maniskill. I've decided to design a new task, which needs a new env to build. But as a rookie in this field, I really value the convenience. I honestly wanna know that given this situation, do you recommend I go through it with maniskill? Or is it more convenient to do it on mujoco? It will mean so much to me if you could give me some advice on that, thank you so much! @StoneT2000 

> You will need to write your own code with the gym setup to interact with the environments.
>
> The other thing is if you trained with lerobot, did you train it on the maniskill task or some other task? [...]

StoneT2000, 2025-04-05:
The g1 dataset if i recall is real world data. If you train on it and evaluate in simulation without proper sim design your policy will likely not work. We would love to add more real2sim environments where you can evaluate policies in simulation such as one for the G1 dataset.

We have already done that for some environments from the BridgeV2 dataset, see https://simpler-env.github.io/, so you can evaluate some VLAs like Octo and RT-x. But this requires you to build such an environment and do system ID

For mujoco vs maniskill, that is up to you. They have fairly similar features at this point with ManiSkill being a bit more "managed/packaged" in some sense (example environments for RL/IL, real2sim etc.).

Frozenkiddo, 2025-04-07:
Thanks a lot for your reply. I will take your advice seriously.

> The g1 dataset if i recall is real world data. If you train on it and evaluate in simulation without proper sim design your policy will likely not work. We would love to add more real2sim environments where you can evaluate policies in simulation such as one for the G1 dataset but don't have time at the moment to do it.
> 
> We have already done that for some environments from the BridgeV2 dataset, see https://simpler-env.github.io/, so you can evaluate some VLAs like Octo and RT-x. But this requires you to build such an environment and do system ID
> 
> For mujoco vs maniskill, that is up to you. They have fairly similar features at this point with ManiSkill being a bit more "managed/packaged" in some sense (example environments for RL/IL, real2sim etc.).

closed 2025-04-07 by StoneT2000