mani-skill/ManiSkill #989: RGBD Sequence Collection.

BJHYZJ, 2025-04-07:
Thank you for the open-source project. I would like to set up a camera in the ManiSkill simulator to collect RGBD sequences of a scene for 3D reconstruction. Could you please advise how to do this? Is there an open-source script available for collecting trajectories that I could refer to?

StoneT2000, 2025-04-07:
You first need an environment.

There's docs using gym showing you how to create one. You can then specify obs mode to be rgb so that env resets and steps return rgb data

Then you need some trajectory generation code. This can either be motion planning (you can copy example code from the mani_skill/examples/motionplannjng folder) or random actions taken or run some rl training / policy

Then to save the trajectory data we recommend using the RecordEpisode wrapper which auto records env steps and saves trajectory data to a .h5 file

BJHYZJ, 2025-04-08:
Thank you for your response. While your approach relies on a camera mounted on a robot, I’m wondering if it would be possible to construct a human-like perspective camera that can be controlled via mouse and keyboard, and record RGB-D sequences during its movement.

closed 2025-04-27 by BJHYZJ