mani-skill/ManiSkill #992: `TypeError: this type has no size` for act/train_rgbd.py when `--num_eval_envs` > 1

jstmn, 2025-04-08:
Hi,

I'm getting the following error when running `examples/baselines/act/train_rgbd.py`:

```
$ python $ACT_DIR/train_rgbd.py --env-id $TASK   --demo-path $RECORD_DIR/$TASK/motionplanning/maniskill_joint_space.rgbd.$CONTROL_MODE.physx_cpu.h5   --demo_type $TASK --control-mode $CONTROL_MODE --sim-backend "physx_cpu" --num_demos $NUM_DEMOS --max_episode_steps 125 --include_depth   --total_iters 30000 --log_freq 100 --eval_freq 5000   --exp-name=act-$TASK-rgbd-${demos}_motionplanning_demos-$SEED   --track # track training on wandb
/usr/local/lib/python3.10/dist-packages/sapien/_vulkan_tricks.py:37: UserWarning: Failed to find Vulkan ICD file. This is probably due to an incorrect or partial installation of the NVIDIA driver. SAPIEN will attempt to provide an ICD file anyway but it may not work.
  warn(
/usr/local/lib/python3.10/dist-packages/diffusers/utils/outputs.py:63: FutureWarning: `torch.utils._pytree._register_pytree_node` is deprecated. Please use `torch.utils._pytree.register_pytree_node` instead.
  torch.utils._pytree._register_pytree_node(
Traceback (most recent call last):
  File "path/to/ManiSkill/examples/baselines/act/train_rgbd.py", line 474, in <module>
    envs = make_eval_envs(args.env_id, args.num_eval_envs, args.sim_backend, env_kwargs, other_kwargs, video_dir=f'runs/{run_name}/videos' if args.capture_video else None, wrappers=wrappers)
  File "path/to/ManiSkill/examples/baselines/act/act/make_env.py", line 37, in make_eval_envs
    env = vector_cls([cpu_make_env(env_id, seed, video_dir if seed == 0 else None, env_kwargs, other_kwargs) for seed in range(num_envs)])
  File "path/to/ManiSkill/examples/baselines/act/act/make_env.py", line 36, in <lambda>
    vector_cls = gym.vector.SyncVectorEnv if num_envs == 1 else lambda x : gym.vector.AsyncVectorEnv(x, context="forkserver")
  File "/usr/local/lib/python3.10/dist-packages/gymnasium/vector/async_vector_env.py", line 121, in __init__
    _obs_buffer = create_shared_memory(
  File "/usr/lib/python3.10/functools.py", line 889, in wrapper
    return dispatch(args[0].__class__)(*args, **kw)
  File "/usr/local/lib/python3.10/dist-packages/gymnasium/vector/utils/shared_memory.py", line 74, in _create_dict_shared_memory
    [
  File "/usr/local/lib/python3.10/dist-packages/gymnasium/vector/utils/shared_memory.py", line 75, in <listcomp>
    (key, create_shared_memory(subspace, n=n, ctx=ctx))
  File "/usr/lib/python3.10/functools.py", line 889, in wrapper
    return dispatch(args[0].__class__)(*args, **kw)
  File "/usr/local/lib/python3.10/dist-packages/gymnasium/vector/utils/shared_memory.py", line 61, in _create_base_shared_memory
    return ctx.Array(dtype, n * int(np.prod(space.shape)))
  File "/usr/lib/python3.10/multiprocessing/context.py", line 141, in Array
    return Array(typecode_or_type, size_or_initializer, lock=lock,
  File "/usr/lib/python3.10/multiprocessing/sharedctypes.py", line 88, in Array
    obj = RawArray(typecode_or_type, size_or_initializer)
  File "/usr/lib/python3.10/multiprocessing/sharedctypes.py", line 61, in RawArray
    obj = _new_value(type_)
  File "/usr/lib/python3.10/multiprocessing/sharedctypes.py", line 40, in _new_value
    size = ctypes.sizeof(type_)
TypeError: this type has no size
```

Steps to recreate:
```
RECORD_DIR=/tmp
ACT_DIR=$BDAI/projects/maple/ws/src/external/ManiSkill_haosulab/examples/baselines/act
TASK="PickCube-v1"
NUM_DEMOS=10
CONTROL_MODE=pd_ee_delta_pos

# Generate demonstrations.
python -m mani_skill.examples.motionplanning.panda.run \
    --env-id $TASK --obs-mode "rgbd" --sim-backend cpu \
    --traj-name "maniskill_joint_space" --num-traj $NUM_DEMOS --record-dir $RECORD_DIR \
    --save-video --only-count-success

# Convert the 'pd_joint_pos' actions to 'pd_ee_pose' actions
python -m mani_skill.trajectory.replay_trajectory \
    --traj-path $RECORD_DIR/$TASK/motionplanning/maniskill_joint_space.h5 \
    --use-first-env-state --save-traj --target-control-mode $CONTROL_MODE \
    --obs-mode "rgbd" --num-envs $NUM_DEMOS --sim-backend cpu # need to run on cpu to convert action spaces

# train rgbd
SEED=1
python $ACT_DIR/train_rgbd.py --env-id $TASK \
  --demo-path $RECORD_DIR/$TASK/motionplanning/maniskill_joint_space.rgbd.$CONTROL_MODE.physx_cpu.h5 \
  --demo_type $TASK --control-mode $CONTROL_MODE --sim-backend "physx_cpu" --num_demos $NUM_DEMOS --max_episode_steps 125 --include_depth \
  --total_iters 30000 --log_freq 100 --eval_freq 5000 \
  --exp-name=act-$TASK-rgbd-${demos}_motionplanning_demos-$SEED \
  --track # track training on wandb
```
* this error only happens with `num_eval_envs` > 1
* I haven't verified if it works with state based act training version

Any idea why this is happening + how to fix? 

Thanks

StoneT2000, 2025-04-08:
I just tried it myself and could not reproduce the bug, the cpu backend is running fine for me with multiprocessing parallelization. Can you run

```
pip list
```

And are you using maniskill from git or nightly? what version of maniskill?

SumeetBatra, 2025-04-15:
I ran into this issue as well, using maniskill `3.0.0b20`. FYI some of the rgb training commands in the baselines.sh script for ACT have `--sim-backend="physx_cuda"` even though the trajectories were collected with `physx_cpu` as the backend. I'm guessing that's an error?

StoneT2000, 2025-04-16:
Might be an error, only reason i can think of using the cuda backend for cpu backend trajectories is because the evaluation time is much faster without much difference in simulation results between cpu and cuda. ACT training is extremely slow relative to other baselines

I'll take a look later this week

StoneT2000, 2025-04-17:
Ok I don't see the issue exactly when using the latest mani-skill-nightly. I just ran

```
python train_rgbd.py --env-id PickCube-v1 --no_include_depth \
    --demo-path ~/.maniskill/demos/PickCube-v1/motionplanning/trajectory.rgb.pd_ee_delta_pos.physx_cpu.h5 \
    --control-mode "pd_ee_delta_pos" --sim-backend "physx_cuda" --num_demos $demos --max_episode_steps 100 --num_eval_envs 100 --no-capture-video \
    --total_iters 30000 --log_freq 100 --eval_freq 5000 \
    --exp-name=act-PickCube-v1-state-${demos}_motionplanning_demos-$seed \
    --demo_type motionplanning --track
```

both physx_cuda and physx_cpu backends run fine (for cpu i lower num envs from 100 to 4)

Maybe it is a numpy version or gymnasium version problem

Can you run pip list? @SumeetBatra @jstmn

SumeetBatra, 2025-04-18:
Here's mine. I've pasted only a few packages that might be relevant b/c my full pip list is embarrassingly long with other random packages in it from testing out other things. Let me know if there's something else you wanted to see. 

```
gym                        0.23.1
gym-notices                0.0.8
gymnasium                  0.29.1
pytorch-kinematics         0.7.5
pytorch-seed               0.2.0
torch                      2.4.0
torchrl                    0.7.2
torchvision                0.19.0
mani_skill                 3.0.0b20
stable_baselines3          2.3.2
sapien                     3.0.0b1
numpy                      1.26.4
```

StoneT2000, 2025-04-18:
You might need to share the full thing. These all look fine

Can you also try python 3.11, the error is from multiprocessing library.

jstmn, 2025-04-19:
mines pretty massive aswell, i'm on mani_skill 3.0.0b18 though.

```
arm_pytorch_utilities                    0.4.3
pytorch-kinematics                       0.7.5
pytorch-seed                             0.2.0
pytorch3d                                0.7.8
torch                                    2.4.1
torch-model-archiver                     0.12.0
torchaudio                               2.4.1
torcheval                                0.0.7
torchserve                               0.9.0
torchvision                              0.19.1

```

StoneT2000, 2025-04-19:
None of these packages are the source of the issue. Can you try `mani-skill-nightly` package or b20 and use python 3.11?

StoneT2000, 2025-05-05:
Hi @jstmn @SumeetBatra is this still an issue? have you tried python 3.11?

jstmn, 2025-05-15:
Same error with python3.11! This isn't blocking for me though. I can try a different gymnasium version

**Error**
```
(venv311)  jstm ~/Projects/ManiSkillHaosulab python $ACT_DIR/train_rgbd.py --env-id $TASK \
>   --demo-path $RECORD_DIR/$TASK/motionplanning/maniskill_joint_space.rgbd.$CONTROL_MODE.physx_cpu.h5 \
>   --demo_type $TASK --control-mode $CONTROL_MODE --sim-backend "physx_cpu" --num_demos $NUM_DEMOS --max_episode_steps 125 --include_depth \
>   --total_iters 30000 --log_freq 100 --eval_freq 5000 \
>   --exp-name=act-$TASK-rgbd-${demos}_motionplanning_demos-$SEED \
>   --track # track training on wandb
Traceback (most recent call last):
  File "/home/jstm/Projects/ManiSkillJstmn/examples/baselines/act/train_rgbd.py", line 474, in <module>
    envs = make_eval_envs(args.env_id, args.num_eval_envs, args.sim_backend, env_kwargs, other_kwargs, video_dir=f'runs/{run_name}/videos' if args.capture_video else None, wrappers=wrappers)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/jstm/Projects/ManiSkillJstmn/examples/baselines/act/act/make_env.py", line 37, in make_eval_envs
    env = vector_cls([cpu_make_env(env_id, seed, video_dir if seed == 0 else None, env_kwargs, other_kwargs) for seed in range(num_envs)])
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/jstm/Projects/ManiSkillJstmn/examples/baselines/act/act/make_env.py", line 36, in <lambda>
    vector_cls = gym.vector.SyncVectorEnv if num_envs == 1 else lambda x : gym.vector.AsyncVectorEnv(x, context="forkserver")
                                                                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/jstm/Projects/ManiSkillHaosulab/venv311/lib/python3.11/site-packages/gymnasium/vector/async_vector_env.py", line 121, in __init__
    _obs_buffer = create_shared_memory(
                  ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/functools.py", line 909, in wrapper
    return dispatch(args[0].__class__)(*args, **kw)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/jstm/Projects/ManiSkillHaosulab/venv311/lib/python3.11/site-packages/gymnasium/vector/utils/shared_memory.py", line 74, in _create_dict_shared_memory
    [
  File "/home/jstm/Projects/ManiSkillHaosulab/venv311/lib/python3.11/site-packages/gymnasium/vector/utils/shared_memory.py", line 75, in <listcomp>
    (key, create_shared_memory(subspace, n=n, ctx=ctx))
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/functools.py", line 909, in wrapper
    return dispatch(args[0].__class__)(*args, **kw)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/jstm/Projects/ManiSkillHaosulab/venv311/lib/python3.11/site-packages/gymnasium/vector/utils/shared_memory.py", line 61, in _create_base_shared_memory
    return ctx.Array(dtype, n * int(np.prod(space.shape)))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/multiprocessing/context.py", line 141, in Array
    return Array(typecode_or_type, size_or_initializer, lock=lock,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/multiprocessing/sharedctypes.py", line 88, in Array
    obj = RawArray(typecode_or_type, size_or_initializer)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/multiprocessing/sharedctypes.py", line 61, in RawArray
    obj = _new_value(type_)
          ^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/multiprocessing/sharedctypes.py", line 40, in _new_value
    size = ctypes.sizeof(type_)
           ^^^^^^^^^^^^^^^^^^^^
TypeError: this type has no size
```

**Version**
```
(venv311)  jstm ~/Projects/ManiSkillHaosulab pip show mani_skill
Name: mani_skill
Version: 3.0.0b20
Summary: ManiSkill3: A Unified Benchmark for Generalizable Manipulation Skills
Home-page: https://github.com/haosulab/ManiSkill
Author: ManiSkill contributors
Author-email: 
License: 
Location: /home/jstm/Projects/ManiSkillHaosulab/venv311/lib/python3.11/site-packages
Editable project location: /home/jstm/Projects/ManiSkillHaosulab
Requires: dacite, fast_kinematics, GitPython, gymnasium, h5py, huggingface_hub, imageio, imageio, IPython, mplib, numpy, pynvml, pytorch_kinematics, pyyaml, sapien, scipy, tabulate, tqdm, transforms3d, trimesh, tyro
Required-by: act

===================

(venv311)  jstm ~/Projects/ManiSkillHaosulab git log | head
[... 112 lines cut ...]
six                      1.17.0
smmap                    5.0.2
stack-data               0.6.3
sympy                    1.14.0
tabulate                 0.9.0
tensorboard              2.19.0
tensorboard-data-server  0.7.2
toppra                   0.6.3
torch                    2.7.0
torchvision              0.22.0
tqdm                     4.67.1
traitlets                5.14.3
transforms3d             0.4.2
trimesh                  4.6.9
triton                   3.3.0
typeguard                4.4.2
typing_extensions        4.13.2
typing-inspection        0.4.0
tyro                     0.9.20
urllib3                  2.4.0
wandb                    0.19.11
wcwidth                  0.2.13
Werkzeug                 3.1.3
zipp                     3.21.0
```

akushonkamen, 2026-10-08:
I dug into the root cause of this while looking at the ACT baselines, sharing in case it helps converge:

The traceback matches gymnasium 0.29.1's `_create_base_shared_memory` (`gymnasium/vector/utils/shared_memory.py:57-61`), which passes `space.dtype.char` straight into `multiprocessing.Array`. When that char is not a valid multiprocessing typecode (the table only covers `?cubBhHiIlLqQfd`), `RawArray` falls back via `typecode_to_type.get(x, x)` and ends up doing `str * size`, so `ctypes.sizeof` raises exactly `TypeError: this type has no size`. A minimal repro on 0.29.1:

```python
import numpy as np, gymnasium.spaces as spaces
from gymnasium.vector.utils.shared_memory import create_shared_memory
sp = spaces.Dict({"x": spaces.Box(-np.inf, np.inf, (4,), np.float16)})  # char 'e' is not a typecode
create_shared_memory(sp, n=2)  # -> TypeError: this type has no size, same 5-frame traceback as here
```

On current main the ACT rgbd eval obs space comes out as `{state: float32, rgb: uint8, depth: int16}` — all valid typecodes — and `create_shared_memory` succeeds on both gymnasium 0.29.1 and 1.4.0 for me, which is consistent with @StoneT2000 not being able to reproduce back then. My guess is the affected installs (b18/b20-era) had some leaf obs space with a non-typecode dtype, possibly from a local version combo, though I can't pin that down from the info available.

Noting that #1492 is reworking this exact `make_env.py` path. If anyone can still trigger this on current main, a minimal repro script would be very helpful; otherwise it may make sense to close this once #1492 lands, treating the error itself as an upstream gymnasium 0.29.1 shared-memory limitation.