mani-skill/ManiSkill #994: UnitreeG1Stand-v1 only supports state (no sensor)

AlexandreBrown, 2025-04-08:
Hi,  
When testing the [UnitreeG1Stand-v1](https://maniskill.readthedocs.io/en/latest/tasks/humanoid/index.html#unitreeg1stand-v1) task I do not get any sensor information when using the obs mode `state+rgb`.  
[image]  
The same issue does not occur for other tasks related to the G1 robot (eg: `UnitreeG1TransportBox-v1` is fine).

StoneT2000, 2025-04-08:
We probably do not include any cameras mounted or otherwise by default for that task. What do you plan to do?

That task is more for a demo showcase than anything that you can kind of train the g1 (prior to the manipulation tasks we built with G1).

AlexandreBrown, 2025-04-08:
Ok I see, I was just testing out the tasks, in my case I don't think I'll actually use G1 Stand I was just curious to see why it would not have a camera.  
I will use the G1 Transport Box task which has all I need already.  
Thanks for answering.

closed 2025-04-08 by AlexandreBrown