mani-skill/ManiSkill #998: cam2world_gl while be 误差较大的，当camera_head 角度发生变化时，cam2world_gl是有较大误差的

BJHYZJ, 2025-04-10:
Hi, thanks for the great work!

I wrote a custom data collection script based on `demo_manual_control.py`, and added keyboard controls to adjust the`fetch head camera` angles. Here's the part of the code that controls the head camera:

```bash
          # body: head
          elif key == "j":  # rotate head left
              body_action[0] = 1
          elif key == "l":  # rotate head right
              body_action[0] = -1
          elif key == "k":  # tilt head down
              body_action[1] = 1
          elif key == "i":  # rotate head up
              body_action[1] = -1
```

To visualize the `.h5` recordings, I used the following script to reconstruct a point cloud scene:

```bash
import h5py
import numpy as np
import open3d as o3d

def create_point_cloud(rgb, depth, intrinsic, cam2world, depth_scale=1000, seg=None):
    fx, fy = intrinsic[0, 0], intrinsic[1, 1]
    cx, cy = intrinsic[0, 2], intrinsic[1, 2]

    xmap, ymap = np.arange(depth.shape[1]), np.arange(depth.shape[0])
    xmap, ymap = np.meshgrid(xmap, ymap)

    z = depth / depth_scale
    # z_mask = np.logical_and(z > 0, z < 1.2)
    z_mask = np.ones_like(z, dtype=bool)
    x = (xmap - cx) / fx * z
    y = (ymap - cy) / fy * z
    z = -z  # OpenCV -> OpenGL
    points_cam = np.stack((x, y, z), axis=-1)[z_mask].reshape(-1, 3)

    colors = rgb[z_mask].reshape(-1, 3) / 255.0

    ones = np.ones((points_cam.shape[0], 1))
    points_hom = np.concatenate([points_cam, ones], axis=1)

    # points_world = np.dot(cam2world, points_hom.T).T[..., :3].reshape(points_cam.shape)
    points_world = (cam2world @ points_hom.T).T[:, :3]

    pcd = o3d.geometry.PointCloud()
    pcd.points = o3d.utility.Vector3dVector(points_world)
    pcd.colors = o3d.utility.Vector3dVector(colors)
    return pcd

if __name__ == "__main__":
    h5_file_path = "/home/yanzj/workspace/code/mshab/recordings/20250410_131313/rgbdseg_sequence.h5"
    target_view = "fetch_head"  # ["fetch_head", "fetch_hand"]

    with h5py.File(h5_file_path, "r") as f:
        group = f["traj_0"]
        obs = group["obs"]
        sensor_data = obs["sensor_data"][target_view]
        sensor_param = obs["sensor_param"][target_view]

        rgbs = sensor_data["rgb"][:]                # (T, H, W, 3)
        depths = sensor_data["depth"][:]            # (T, H, W, 1)
        segs = sensor_data["segmentation"][:]       # (T, H, W, 1)
        cam2world_gl = sensor_param["cam2world_gl"][:]  # (T, 4, 4)
        intrinsic_cv = sensor_param["intrinsic_cv"][:]  # (T, 3, 4) --> use [:3]

        merged_pcd = o3d.geometry.PointCloud()

        for i in range(len(rgbs)):
            rgb = rgbs[i]
            depth = depths[i][..., 0]
            seg = segs[i][..., 0]
            intrinsic = intrinsic_cv[i][:, :3]
            cam2world = cam2world_gl[i]

            pcd = create_point_cloud(rgb, depth, intrinsic, cam2world, seg=seg)
            merged_pcd += pcd

    o3d.visualization.draw_geometries([merged_pcd])

```

I noticed that when I collect data without moving the `fetch head camera`, the reconstructed scene looks good. However, when I move the head camera during data collection, the reconstruction quality significantly degrades.

I recorded a short video (**8 seconds**):

- The `first 4 seconds`: the `fetch head camera` stays still.

- The `last 4 seconds`: I manually move the `fetch head camera`.

Here is the video:
https://github.com/user-attachments/assets/df1239c3-1dd6-497b-91f4-5d2ae9829cb6

**Reconstruction results:**

The first half (no camera movement) reconstructs well:

[image]

The second half (with head movement) shows large misalignment, suggesting that cam2world_gl has noticeable errors:

[image]

BJHYZJ, 2025-04-10:
The `obs_mode` I'm currently using is `rgb+depth+segmentation`. When I switch it to `pointcloud`, the isssue mentioned above no longer exists, since the pointcloud is inherently in the world coordinate system. I would like to ask whether the error in `cam2world_gl` was introduced intentionally to make it more like the real world, or if it's a bug in the underlying logic of the simulation itself.

[image]

StoneT2000, 2025-04-19:
pointcloud uses cam2world_gl, in fact internally pointcloud is just rgb+depth+segmentation mode but we do preprocessing for you. Maybe you conversions code from rgbd to pcd world frame isn't right. You can check the maniskill code for how it is handled in mani_skill/envs/utils/observation

StoneT2000, 2025-05-20:
Is this still an issue @BJHYZJ ?

closed 2025-06-15 by StoneT2000