mani-skill/ManiSkill #1000: Controlling randomness in parallel environments

abadithela, 2025-04-13:
Hi, 

I'm trying to control the randomness across parallel environments for a simple pick and place task. I've checked the following [documentation](https://maniskill.readthedocs.io/en/latest/user_guide/concepts/rng.html) and I am setting the same seed when calling env.reset(seed= [x1, x1, ..., x1]) for some seed x1. I did not set any other parameters. I still observe a randomness in the outcome; is there any other seed I should be fixing. Based on the documentation, I thought that passing the same seed during reset will control for the randomness across parallel environments. 

Is there anything else I should be fixing?

Thanks,
Apurva

StoneT2000, 2025-04-16:
What is your task randomization implementation like?

We recommend
1. `_batched_episode_rng` for any non env state related randomizations (use only in def _load_scene), such as colors of objects, object meshes etc.
2. torch RNG for env state related randomizations (object poses, velocities, qpos, qvels, any other info) (it is already seeded at an episode level using the first number provided, in this case, `x1`: ). Torch RNG is dependent on batch size and the episode level seed (`x1`). This is faster RNG and recommended as a result since env state related randomizations occur most frequently (during def _initialize_episode)

You should be getting the same env state (object pose, velocities) if you keep calling env.reset(seed=[x1, x1, ..., x1]) as long as you use torch rng for randomizations in def _initialize_episode

abadithela, 2025-04-16:
Our initialize_episode function looks like the following:

```
def _initialize_episode(self, env_idx: torch.Tensor, options: dict):
        # use the torch.device context manager to automatically create tensors on CPU or CUDA depending on self.device, the device the environment runs on
        super()._initialize_episode(env_idx, options)
        with torch.device(self.device):
            if self.do_motion_planning:
                self.manip_obj_pcds = []
                self.goal_obj_pcds = []
                if len(self.manip_obj_meshes_raw) > 1:
                    for i in range(len(self.manip_obj_meshes_raw)):
                        manip_mesh = self.manip_obj_meshes_raw[i].copy()
                        goal_mesh = self.goal_obj_meshes_raw[i].copy()
                        manip_pose = self.manip_obj.pose[i]
                        goal_pose = self.goal_obj.pose[i]

                        manip_mesh.apply_transform(
                            manip_pose.sp.to_transformation_matrix()
                        )
                        goal_mesh.apply_transform(
                            goal_pose.sp.to_transformation_matrix()
                        )

                        samples = trimesh.sample.sample_surface_even(
                            manip_mesh, NUM_SAMPLES
                        )
                        pcd = PointCloud(
                            xyz_pts=samples[0].__array__().copy(),
                            normals=manip_mesh.face_normals[samples[1]].copy(),
                        )
                        self.manip_obj_pcds.append(pcd)

                        # visualize_trimesh(manip_mesh, extra_pts=(samples[0].__array__().copy(), manip_mesh.face_normals[samples[1]].copy()))

                        samples = trimesh.sample.sample_surface_even(
                            goal_mesh, NUM_SAMPLES
                        )
                        pcd = PointCloud(
                            xyz_pts=samples[0].__array__().copy(),
                            normals=goal_mesh.face_normals[samples[1]].copy(),
                        )
                        self.goal_obj_pcds.append(pcd)
                else:
                    manip_mesh = self.manip_obj_meshes_raw.copy()
                    goal_mesh = self.goal_obj_meshes_raw.copy()
                    manip_pose = self.manip_obj.pose
                    goal_pose = self.goal_obj.pose

                    manip_mesh.apply_transform(manip_pose.sp.to_transformation_matrix())
                    goal_mesh.apply_transform(goal_pose.sp.to_transformation_matrix())

                    samples = trimesh.sample.sample_surface_even(
                        manip_mesh, NUM_SAMPLES
                    )
                    pcd = PointCloud(
                        xyz_pts=samples[0].__array__().copy(),
                        normals=manip_mesh.face_normals[samples[1]].copy(),
                    )
                    self.manip_obj_pcds.append(pcd)

                    samples = trimesh.sample.sample_surface_even(goal_mesh, NUM_SAMPLES)
                    pcd = PointCloud(
                        xyz_pts=samples[0].__array__().copy(),
                        normals=goal_mesh.face_normals[samples[1]].copy(),
                    )
                    self.goal_obj_pcds.append(pcd)

        print("Initialized episode.")
```

StoneT2000, 2025-04-16:
How are you checking object poses?

Moreover this looks fine to me? In GPU simulation objects in reality are in one big plane, so floating point imprecision will occur. The difference in poses if you read it from actor.pose property after setting it will happen but are quite small

At most, object i initialized at pose p in parallel env j will always be at the same exact pose to infinite precision? But this might not be guaranteed in physx cuda backend if physx does something strange we don't know about.

abadithela, 2025-04-16:
Yes, I agree that these initial poses are fine. Please see the video here: [https://drive.google.com/file/d/1ODqhOOn5Mn-1r0_BRLAkkNZmTe4O3Xwj/view?usp=sharing](https://drive.google.com/file/d/1ODqhOOn5Mn-1r0_BRLAkkNZmTe4O3Xwj/view?usp=sharing). Here, I have set the same initial condition, and I have also set the self._batched_episode_rng to be initialized to the same state. 

Interestingly, the evals diverge after it drops the tomato for the first time.

StoneT2000, 2025-04-16:
That's the sim, right? I got fooled for a good minute, nicely designed env!

Re: the divergence, this is normal unfortunately because of floating point imprecisions which can compound and cause divergences, especially if objects are falling / hitting multiple things.

If your goal is to just ensure determinism, it really is sufficient to just do what you are doing already by ensuring the initial object states are close enough. Each time you run `n` parallel envs with seeds `x1, x2, ..., xn` with the same policy you will always get the same result in env `1, 2, ..., n`. There might not be perfect consistency between pairwise envs but for all intents and purposes the evaluation is valid. 

Taking an action sequence applied in env `i` and then applying it to a different env `j` may lead to some divergence, but if you are evaluating a policy this should not be an issue (if it fails due to some small floating point imprecisions and cannot react accordingly then policy is not robust at all).The randomness inherent to simulation and gpu parallelized simulation is very small and should not affect results unless the action generation is from an extremely unrobust policy.

abadithela, 2025-04-23:
Thank you. Yes, the policy is indeed not very robust.

closed 2025-04-23 by abadithela

StoneT2000, 2025-04-30:
Can you open a new issue for this?

And in the new issue can you describe within polymetis what exact controller you are using? How does it transform action to joint positions?