mani-skill/ManiSkill #999: TypeError: 'NoneType' object is not callable from running ppo.py

HxLyn3, 2025-04-13:
My ManiSkill version is `mani_skill-3.0.0b20`. While I run ppo.py by
`python ppo.py --env_id="PushCube-v1" 
  --num_envs=2048 --update_epochs=8 --num_minibatches=32
  --total_timesteps=2_000_000 --eval_freq=10 --num-steps=20`
There will be a TypeError after several training steps, like
`Traceback (most recent call last):
  File "/home/linhaoxin/miniconda3/envs/lhx/lib/python3.10/site-packages/gymnasium/vector/vector_env.py", line 330, in __del__
  File "/home/linhaoxin/miniconda3/envs/lhx/lib/python3.10/site-packages/mani_skill/vector/wrappers/gymnasium.py", line 161, in close
  File "/home/linhaoxin/miniconda3/envs/lhx/lib/python3.10/site-packages/gymnasium/core.py", line 475, in close
  File "/home/linhaoxin/miniconda3/envs/lhx/lib/python3.10/site-packages/gymnasium/core.py", line 475, in close
  File "/home/linhaoxin/miniconda3/envs/lhx/lib/python3.10/site-packages/gymnasium/core.py", line 475, in close
  File "/home/linhaoxin/miniconda3/envs/lhx/lib/python3.10/site-packages/mani_skill/envs/sapien_env.py", line 1126, in close
  File "/home/linhaoxin/miniconda3/envs/lhx/lib/python3.10/site-packages/mani_skill/envs/sapien_env.py", line 1123, in _clear
TypeError: 'NoneType' object is not callable`

StoneT2000, 2025-04-13:
Did training complete? If so you can probably ignore this.

I think sometimes when we explicitly run garbage collection in python when a process is about to close some objects/modules are unloaded and become None.

HxLyn3, 2025-04-13:
The training was not complete. Just after 48 epochs, the error occurred.

StoneT2000, 2025-04-13:
48 epochs is near 2 million steps (48 * 2048 envs * 20 steps per env ~= 2 million). The code completed all training epochs actually, so it went to the final line where it called env.close and attempted to exit the python process.

You can just increase the total timesteps up from 2 million if you want to train longer.

HxLyn3, 2025-04-13:
Get it! Thank you very much for your response!

closed 2025-04-13 by StoneT2000