mani-skill/ManiSkill #1002: A minor issue regarding the impact of adding <disable_collisions> in URDF on simulation speed

fffffq99, 2025-04-15:
Hi, thanks for the great work!
I am creating my own robot, and as I have many joints, I need to carefully set the collision relationships between these joints. I am curious because I cannot be sure if some joints will collide during movement. In order to prevent collisions that should not occur, I may need to set a lot of`<disable_comllisions>`in URDF. Will the simulation speed slow down?

Let's clarify my question. In URDF, I have set many`<disable_comllision>`settings, some of which may actually collide while others hardly collide. 
I would like to ask if both of these`<disable_comllision>`settings will affect the simulation speed? Or only the parts that will actually collide but are covered up by me with`<disabling collisions>`will affect the simulation speed? 
If it's the latter, I can confidently set more`<disabling collisions>`to avoid some unreasonable collisions.

StoneT2000, 2025-04-16:
More disabled collisions generally means faster sim. You should check with running random actions on the robot and checking which contacts are recorded and shouldn't be to determine what to disable.

fffffq99, 2025-04-16:
> More disabled collisions generally means faster sim. You should check with running random actions on the robot and checking which contacts are recorded and shouldn't be to determine what to disable.

Thank you for your prompt reply.
I understand what you mean and I am currently working on it. I found that using `pd_joint_delta_pos` generates more collisions than using `pd_joint_pos`. I speculate that this is because `pd_joint_pos` does not produce actions that exceed joint limits, while `pd_joint_delta_pos` can cause certain actions to exceed limits and result in unreasonable collisions. So should I use the `pd_joint_pos` mode? 
Another question, why don't you clip the actions in `pd_joint_delta_pos` to ensure that they are all within joint limits? Is there any special reason?

StoneT2000, 2025-04-16:
The controller type has nothing much to do with collisions. it simply decides how your input actions are translated into target positions for the robot joints. pd_joint_pos is absolute joint space and pd_joint_delta_pos applies the unnormalized user action input to the current joint position to get the new target position.

Both will already clip the generated target positions sent to the sim to control the robot to be within joint limits.

fffffq99, 2025-04-16:
```
def set_action(self, action: Array):
    action = self._preprocess_action(action)
    self._step = 0
    self._start_qpos = self.qpos
    if self.config.use_delta:
        if self.config.use_target:
            self._target_qpos = self._target_qpos + action
        else:
            self._target_qpos = self._start_qpos + action
    else:
        # Compatible with mimic controllers. Need to clone here otherwise cannot do in-place replacements in the reset function
        self._target_qpos = torch.broadcast_to(
            action, self._start_qpos.shape
        ).clone()
    if self.config.interpolate:
        self._step_size = (self._target_qpos - self._start_qpos) / self._sim_steps
    else:
        self.set_drive_targets(self._target_qpos)
```

I apologize for my incorrect description. I have read the source code and my understanding is as follows:
If I set `use_delta=True`, The `target_qpos` seems to be equal to the `current_qpos + action`, although the `action` does not exceed the limit, the `current_qpos + action` may exceed the joint's limit ?

closed 2025-04-16 by fffffq99