mani-skill/ManiSkill #1003: How to get target qpos and target qvel

yjang43, 2025-04-15:
Hi! Thank you for the great simulator.

I would like to do something similar to #144.

The following methods `get_drive_target` and `get_drive_velocity_target` do not seem to be supported by maniskill3. 
Is there an alternative way to compute torque on joints?

StoneT2000, 2025-04-16:
we can add it, it is stored in one of the gpu buffers

yjang43, 2025-04-16:
Great! At the moment, I found a workaround at least for PD position control. Controller state for `PDJointPosController` provides `target_qpos` information and this can be retrieved with `get_proprioception` method.

```python
            # joint_pos = PDJointPosControllerConfig(..., use_target=True, ...)

            joints = self.agent.robot.get_active_joints()
            stiffness = torch.stack([j.stiffness for j in joints], dim=-1)
            damping = torch.stack([j.damping for j in joints], dim=-1)
            force_limit = torch.stack([j.force_limit for j in joints], dim=-1)

            proprio = self.agent.get_proprioception()
            qpos = proprio["qpos"]
            qvel = proprio["qvel"]
            target_qpos = torch.concat([
                proprio["controller"]["free"]["target_qpos"],
                proprio["controller"]["hand"]["target_qpos"]
            ], dim=-1)
            target_qvel = torch.zeros_like(target_qpos)

            torque = stiffness * (target_qpos - qpos) + damping * (target_qvel - qvel)
```

closed 2025-04-27 by StoneT2000