mani-skill/ManiSkill #1004: Low Success Rate When Using Diffusion Policy on the StackCube-v1 Task

DumbleTang, 2025-04-15:
My replay_trajectory settings are as follows:
```
"--use-first-env-state",
"-b", "physx_cpu",
"-c", "pd_ee_delta_pos", 
"-o", "rgb",  
"--save-traj"
```

My train_rgbd.py settings are:
```
"--control-mode", "pd_joint_delta_pos",
"--sim-backend", "physx_cpu",
"--num-demos", "100",
"--max_episode_steps", "200",
"--total_iters", "50000",
"--batch_size", "256",
"--obs-mode", "rgb"
lr: float = 1e-4
obs_horizon: int = 2
act_horizon: int = 8
pred_horizon: int = 16
diffusion_step_embed_dim: int = 64
unet_dims: List[int] = [64, 128, 256]
n_groups: int = 8
```

With these settings, my success rate is only 23%.
May I ask why? Is there a problem with my hyperparameter configuration?

StoneT2000, 2025-04-16:
Possibly. DId you run our baseline first?

Data gen: https://github.com/haosulab/ManiSkill/blob/main/scripts/data_generation/replay_for_il_baselines.sh
Training: https://github.com/haosulab/ManiSkill/blob/main/examples/baselines/diffusion_policy/baselines.sh#L27-L33

should work relatively well.

DumbleTang, 2025-04-16:
Yes, I trained according to your baseline. Both the trajectory replay and training were followed step by step based on your example, the commands used are also based on the commands in the two files you just provided, but the success rate is still low

DumbleTang, 2025-04-16:
> Possibly. DId you run our baseline first?
> 
> Data gen: https://github.com/haosulab/ManiSkill/blob/main/scripts/data_generation/replay_for_il_baselines.sh Training: https://github.com/haosulab/ManiSkill/blob/main/examples/baselines/diffusion_policy/baselines.sh#L27-L33
> 
> should work relatively well.

Thanks for your reply!
Yes, I trained according to your baseline. Both the trajectory replay and training were followed step by step based on your example, the commands used are also based on the commands in the two files you just provided, but the success rate is still low

StoneT2000, 2025-04-16:
These are all our results: https://wandb.ai/stonet2000/ManiSkill/workspace?nw=3d49kizu62e

For stack cube is it not similar? Are you using the latest maniskill commit or what versiona re you using?

DumbleTang, 2025-04-16:
The version I am using is 3.0.0b19

DumbleTang, 2025-04-16:
> These are all our results: https://wandb.ai/stonet2000/ManiSkill/workspace?nw=3d49kizu62e
> 
> For stack cube is it not similar? Are you using the latest maniskill commit or what versiona re you using?

The version I am using is 3.0.0b19. So I am quite confused as to why there is such a big difference in success rate between state and rgb

StoneT2000, 2025-04-16:
Oh you are using vision based input. That task is just much harder in general. If you tune it a lot maybe you can solve the task, but it's simply a baseline.

You can try generating more demos to see if it improves success rate.

closed 2025-04-27 by StoneT2000

Vector215, 2025-12-05:
> > My replay_trajectory settings are as follows:
> > ```
> > "--use-first-env-state",
> > "-b", "physx_cpu",
> > "-c", "pd_ee_delta_pos", 
> > "-o", "rgb",  
> > "--save-traj"
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > My train_rgbd.py settings are:
> > ```
> > "--control-mode", "pd_ee_delta_pos",
> > "--sim-backend", "physx_cpu",
> > "--num-demos", "100",
> > "--max_episode_steps", "200",
> > "--total_iters", "30000",
> > "--batch_size", "256",
> > "--obs-mode", "rgb"
> > lr: float = 1e-4
> > obs_horizon: int = 2
> > act_horizon: int = 8
> > pred_horizon: int = 16
> > diffusion_step_embed_dim: int = 64
> > unet_dims: List[int] = [64, 128, 256]
> > n_groups: int = 8
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > With these settings, my success rate is only 23%. May I ask why? Is there a problem with my hyperparameter configuration?
> 
> Hello, I’m writing to ask how you achieved a 23% success rate. Based on the authors’ reported results (~7%) and my own runs (close to 0%), I haven’t been able to reproduce it. I’d be very grateful if you could tell me some idea that can help me reach 23%. From what I can tell, the parameters you used don’t seem unusual. Thanks!!

I am facing the same problem, the success rate of diffusion policy trained using RGB in Maniskill is quite low. The results I obtained from training with 500 demos are still close to 0%, but according to some papers, stackcube should not be a very difficult task for DP.
If you have identified the problem, I hope you can share your experience