mani-skill/ManiSkill #1005: "Create window failed: Renderer does not support display." with dual GPUs.

fffffq99, 2025-04-16:
I am running `mani_stkill` on a remote server with two NVIDIA 4090 GPUs. And I use `xrdp` to show the destop and windows.  Due to the first GPU being occupied, I used `export CUDA_VISIBLEDEVICES=1` on the terminal. When I run my script, the following message appears:

```
  File "/home/xxx/.local/lib/python3.10/site-packages/mani_skill/envs/sapien_env.py", line 672, in _reconfigure
    self._viewer = create_viewer(self._viewer_camera_config)
  File "/home/xxx/.local/lib/python3.10/site-packages/mani_skill/viewer/__init__.py", line 30, in create_viewer
    viewer = Viewer(
  File "/home/xxx/.local/lib/python3.10/site-packages/sapien/utils/viewer/viewer.py", line 73, in __init__
    self.window = RenderWindow(*resolution, self.shader_dir)
RuntimeError: Create window failed: Renderer does not support display.

```

the script:
```
import gymnasium as gym
import mani_skill
import mani_skill.envs
import mani_skill.agents

env = gym.make(
    "Empty-v1",
    robot_uids="panda",
    control_mode="pd_joint_pos",
    num_envs=1,
    sim_backend="gpu",
    render_backend="gpu",
    render_mode="human",
)
```

No matter what I use
```
sim_backend="gpu",
render_backend="gpu",
```
or
```
sim_backend="cpu",
render_backend="cpu",
```

This problem always occurs.
But everything works fine when I use `export CUDA_VISIBLE_DEVICES=0`.

StoneT2000, 2025-04-16:
I'm surprised even one of the GPUs work.

You can't use render mode human if there isn't a display attached. That mode will always opens a viewer upon env creation.

fffffq99, 2025-04-17:
> I'm surprised even one of the GPUs work.
> 
> You can't use render mode human if there isn't a display attached. That mode will always opens a viewer upon env creation.

Yes, this is so strange, because when I test locally, no matter which GPU I use, it can work properly, but when I connect remotely with `xrdp`, only the first GPU can use render mode human.

By the way, I tested `Vulkan` through `vkcube` and the results seemed to be the same. When I tested remotely, I specified `vkcube -- gpu_number 1` and it would fail with the prompt `Could not find both graphics and presentation queues`. 

So I think that the reason why the viewer is not working properly is related to `Vulkan`, but I don't understand the principle of `Vulkan`, so I cannot solve it.

closed 2025-04-17 by fffffq99