mani-skill/ManiSkill #1006: about initial action when eval

awesomeNabi, 2025-04-17:
Hello, first of all, thank you very much for your excellent work. 

I have a question. I'm currently integrating my model into the ManiSkill baseline. My model is based on the diffusion policy framework, and during evaluation, it requires the initial action from the simulation environment as a _condition to predict an action chunk._

However, in the baseline's evaluation code, I only see how the initial observation is obtained using **obs, info = eval_envs.reset()**. I don't see any code related to getting the initial action. So I’d like to ask:

Are the initial actions always fixed and deterministic in ManiSkill environments? Or can I randomly generate an initial action using **action = eval_envs.action_space.sample()?**

Thank you!

StoneT2000, 2025-04-17:
I have never heard of an initial action? You can take a look at the example diffusion policy baseline we have maybe? Or use zeros.

Otherwise yes you can just sample a random action like that. Note that this will sample usually a batched set of actions of shape (N, D) for n envs, d dimensions in actions.

StoneT2000, 2025-05-05:
@awesomeNabi is this still an issue?

awesomeNabi, 2025-05-05:
> [@awesomeNabi](https://github.com/awesomeNabi) is this still an issue?

Thanks, I don't think I need to think about this issue right now.

closed 2025-05-05 by awesomeNabi