mani-skill/ManiSkill #1009: Changing object color for episode reset

dongki-fai, 2025-04-18:
Hello :) Thank you so much for this awesome simulation! Instead of setting the static color of objects (e.g., cube) in `_load_scene()`, I would like to have a dynamic color of an object by changing its color in `_initialize_episode()`. Because setting the color of an object happens in the `actors.build` code, I could not find a way to change its color in the reset function. Would it be possible to ask for your help on this? Thank you.

dongki-fai, 2025-04-18:
By accessing the cube object render shapes in the `_initialize_episode`, I was successfully change the color of cubes (Both for CPU and GPU). 

```python
    def _initialize_episode(self, env_idx: torch.Tensor, options: dict):
        with torch.device(self.device):
            # set color of cubes
            for cube in self.cubes:
                for _obj in cube._objs:
                    render_body_component = _obj.find_component_by_type(RenderBodyComponent)
                    for render_shape in render_body_component.render_shapes:
                        for part in render_shape.parts:
                            part.material.set_base_color(random.choice(self.cube_colors))
```

However, the [tutorial](https://maniskill.readthedocs.io/en/latest/user_guide/tutorials/domain_randomization.html#actor-link-physical-and-visual-randomizations) says that *"Once the GPU simulation is initialized, some properties are fixed and can only be changed again if the environment is reconfigured."* Would dynamically changing the color of the objects result in issues? Thank you.

StoneT2000, 2025-04-19:
I am not so sure if the colors change during episode initialization correctly, you should check a few parallel envs each reset if they work. Usually you should only do this in load scene and if you want to randomize more you can set a reconfiguration frequency to be higher than 0 (although this also slows sim down).

The other thing is that the for loop may slow down your simulation speed since episode initialization occurs often.

StoneT2000, 2025-05-05:
@dongki-fai is this still an issue?

dongki-fai, 2025-05-05:
Hello @StoneT2000. Thank you so much for your kind help and quick response! I have tested the above [code](https://github.com/haosulab/ManiSkill/issues/1009#issuecomment-2816349574) works for a single environment. However, I could not have a chance to verify whether it functions correctly with parallel environments. For my data collection purposes, using a single environment is sufficient, so I'd like to close the issue. Thank you again for this awesome and useful simulator!

closed 2025-05-05 by dongki-fai