mani-skill/ManiSkill #1010: KeyError with PARTNET_MOBILITY for `TurnFaucet-v1`

jstmn, 2025-04-19:
Howdy, I'm getting the following error when trying to run the `TurnFaucet-v1` environment

```
$ python -m mani_skill.examples.motionplanning.panda.run     --env-id "TurnFaucet-v1" --obs-mode "rgbd" --sim-backend "cpu" --traj-name "TurnFaucet-v1__TEMP"     --num-traj 2 --record-dir $BDAI/data/maniskill_TEMP     --only-count-success --num-procs 15 --reward-mode "sparse"       
/home/jstm/.local/lib/python3.10/site-packages/sapien/_vulkan_tricks.py:37: UserWarning: Failed to find Vulkan ICD file. This is probably due to an incorrect or partial installation of the NVIDIA driver. SAPIEN will attempt to provide an ICD file anyway but it may not work.
  warn(
Incompatible version of OpenGL_accelerate found, need at least (3, 1, 9) found (3, 1, 7)
Traceback (most recent call last):
  File "/usr/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/workspaces/bdai/projects/maple/ws/src/external/ManiSkill/mani_skill/examples/motionplanning/panda/run.py", line 200, in <module>
    main(parse_args())
  File "/workspaces/bdai/projects/maple/ws/src/external/ManiSkill/mani_skill/examples/motionplanning/panda/run.py", line 196, in main
    _main(args, start_seed=seed)
  File "/workspaces/bdai/projects/maple/ws/src/external/ManiSkill/mani_skill/examples/motionplanning/panda/run.py", line 87, in _main
    env = gym.make(
  File "/home/jstm/.local/lib/python3.10/site-packages/gymnasium/envs/registration.py", line 802, in make
    env = env_creator(**env_spec_kwargs)
  File "/workspaces/bdai/projects/maple/ws/src/external/ManiSkill/mani_skill/utils/registration.py", line 182, in make
    env = env_spec.make(**kwargs)
  File "/workspaces/bdai/projects/maple/ws/src/external/ManiSkill/mani_skill/utils/registration.py", line 79, in make
    return self.cls(**_kwargs)
  File "/workspaces/bdai/projects/maple/ws/src/external/ManiSkill/mani_skill/envs/tasks/tabletop/turn_faucet.py", line 48, in __init__
    super().__init__(
  File "/workspaces/bdai/projects/maple/ws/src/external/ManiSkill/mani_skill/envs/sapien_env.py", line 309, in __init__
    obs, _ = self.reset(seed=[2022 + i for i in range(self.num_envs)], options=dict(reconfigure=True))
  File "/workspaces/bdai/projects/maple/ws/src/external/ManiSkill/mani_skill/envs/sapien_env.py", line 828, in reset
    self._reconfigure(options)
  File "/workspaces/bdai/projects/maple/ws/src/external/ManiSkill/mani_skill/envs/sapien_env.py", line 666, in _reconfigure
    self._load_scene(options)
  File "/workspaces/bdai/projects/maple/ws/src/external/ManiSkill/mani_skill/envs/tasks/tabletop/turn_faucet.py", line 91, in _load_scene
    builder = articulations.get_articulation_builder(
  File "/workspaces/bdai/projects/maple/ws/src/external/ManiSkill/mani_skill/utils/building/articulations/__init__.py", line 25, in get_articulation_builder
    builder = get_partnet_mobility_builder(
  File "/workspaces/bdai/projects/maple/ws/src/external/ManiSkill/mani_skill/utils/building/articulations/partnet_mobility.py", line 56, in get_partnet_mobility_builder
    urdf_path = PARTNET_MOBILITY["model_urdf_paths"][id] 
KeyError: '5039'
```

Any ideas? Thanks.

jstmn, 2025-04-20:
Some quick investigation:
```
jstm@bdai_docker:~/.maniskill/data/partnet_mobility/dataset$ ls
1000  1002  1005  1007  1014  1017  1021  1025  1027  1030  1032  1034  1036  1039  1041  1044  1046  1049  1052  1056  1060  1062  1064  1066  1068  1075  1077  1079  1082
1001  1004  1006  1013  1016  1018  1024  1026  1028  1031  1033  1035  1038  1040  1042  1045  1047  1051  1054  1057  1061  1063  1065  1067  1073  1076  1078  1081
jstm@bdai_docker:~/.maniskill/data/partnet_mobility/dataset$ cat 1000/meta.json 
{"model_id": "190cb7264781df604da5feafe6f1c8fc", "anno_id": "45194", "user_id": "face++wsl", "time_in_sec": "816", "version": "3", "model_cat": "StorageFurniture"}jstm@bdai_docker:~/.maniskill/data/partnet_mobility/dataset$ cat 1000/semantics
cat: 1000/semantics: No such file or directory
jstm@bdai_docker:~/.maniskill/data/partnet_mobility/dataset$ cat 1000/semantics.txt 
link_0 hinge rotation_door
link_1 hinge rotation_door
link_2 slider drawer
link_3 slider drawer
link_4 heavy furniture_body
```
I'm not sure how to find the right ID (`urdf_path = PARTNET_MOBILITY["model_urdf_paths"][id]`) for the faucet. The directories in `~/.maniskill/data/partnet_mobility/dataset` don't seem to include what part the contain. They do have a `model_id` hash however, but on the [PartNet website](https://sapien.ucsd.edu/browse) I didn't see any way to search by `model_id`.

StoneT2000, 2025-05-20:
does `python mani_skill/examples/demo_random_action.py -e TurnFaucet-v1 --record-dir=videos` also fail?

The metadata for the faucet assets should be in ManiSkill. There is a `info_faucet_train.json` file that we have ported from the previous ManiSkill2 that should be in the dataset

jstmn, 2025-05-20:
Yep, still fails

```
jstm ~/Projects/mpcm2 uv run python -m mani_skill.utils.download_asset partnet_mobility_cabinet
Uninstalled 2 packages in 19ms
Installed 2 packages in 30ms
/usr/lib/python3.10/runpy.py:126: RuntimeWarning: 'mani_skill.utils.download_asset' found in sys.modules after import of package 'mani_skill.utils', but prior to execution of 'mani_skill.utils.download_asset'; this may result in unpredictable behaviour
  warn(RuntimeWarning(msg))
Downloading assets for partnet_mobility_cabinet: 1/57

-------------------------
/home/jstm/.maniskill/data does not exist. Create?
(y|n): y
Downloading https://storage1.ucsd.edu/datasets/ManiSkill2022-assets/partnet_mobility/dataset/1064.zip
4.23MiB [00:00, 50.6MiB/s]
8840.90it/s]
...

 jstm ~/Projects/mpcm2 uv run python thirdparty/ManiSkill/mani_skill/examples/demo_random_action.py -e TurnFaucet-v1 --record-dir=videos
Traceback (most recent call last):
  File "/home/jstm/Projects/mpcm2/thirdparty/ManiSkill/mani_skill/examples/demo_random_action.py", line 130, in <module>
    main(parsed_args)
  File "/home/jstm/Projects/mpcm2/thirdparty/ManiSkill/mani_skill/examples/demo_random_action.py", line 85, in main
    env: BaseEnv = gym.make(
  File "/home/jstm/Projects/mpcm2/.venv/lib/python3.10/site-packages/gymnasium/envs/registration.py", line 742, in make
    env = env_creator(**env_spec_kwargs)
  File "/home/jstm/Projects/mpcm2/thirdparty/ManiSkill/mani_skill/utils/registration.py", line 182, in make
    env = env_spec.make(**kwargs)
  File "/home/jstm/Projects/mpcm2/thirdparty/ManiSkill/mani_skill/utils/registration.py", line 79, in make
    return self.cls(**_kwargs)
  File "/home/jstm/Projects/mpcm2/thirdparty/ManiSkill/mani_skill/envs/tasks/tabletop/turn_faucet.py", line 48, in __init__
    super().__init__(
  File "/home/jstm/Projects/mpcm2/thirdparty/ManiSkill/mani_skill/envs/sapien_env.py", line 312, in __init__
    obs, _ = self.reset(seed=[2022 + i for i in range(self.num_envs)], options=dict(reconfigure=True))
  File "/home/jstm/Projects/mpcm2/thirdparty/ManiSkill/mani_skill/envs/sapien_env.py", line 890, in reset
    self._reconfigure(options)
  File "/home/jstm/Projects/mpcm2/thirdparty/ManiSkill/mani_skill/envs/sapien_env.py", line 728, in _reconfigure
    self._load_scene(options)
  File "/home/jstm/Projects/mpcm2/thirdparty/ManiSkill/mani_skill/envs/tasks/tabletop/turn_faucet.py", line 91, in _load_scene
    builder = articulations.get_articulation_builder(
  File "/home/jstm/Projects/mpcm2/thirdparty/ManiSkill/mani_skill/utils/building/articulations/__init__.py", line 25, in get_articulation_builder
    builder = get_partnet_mobility_builder(
  File "/home/jstm/Projects/mpcm2/thirdparty/ManiSkill/mani_skill/utils/building/articulations/partnet_mobility.py", line 56, in get_partnet_mobility_builder
    urdf_path = PARTNET_MOBILITY["model_urdf_paths"][id]
KeyError: '5039'

```

StoneT2000, 2025-05-20:
Are you on a fork of ManiSkill or something? There is this code in our codebase to load the metadata for these files:

```
def _load_partnet_mobility_dataset():
    global PARTNET_MOBILITY
    """loads preprocssed partnet mobility metadata"""
    PARTNET_MOBILITY = {
        "model_data": load_json(
            PACKAGE_ASSET_DIR / "partnet_mobility/meta/info_cabinet_drawer_train.json"
        ),
    }
    for data_file in ["info_cabinet_door_train.json", "info_faucet_train.json"]:
        PARTNET_MOBILITY["model_data"].update(
            load_json(PACKAGE_ASSET_DIR / "partnet_mobility/meta" / data_file)
        )
```
at https://github.com/haosulab/ManiSkill/blob/6bfccf372255ffa652283e39d46a17ad757e6d4c/mani_skill/utils/building/articulations/partnet_mobility.py

It has the faucet train json listed, and it should raise an error if it can't find it.

StoneT2000, 2025-05-20:
The file is there too: https://github.com/haosulab/ManiSkill/blob/6bfccf372255ffa652283e39d46a17ad757e6d4c/mani_skill/assets/partnet_mobility/meta/info_faucet_train.json

(Try main branch / nightly?)

StoneT2000, 2025-05-28:
Oh I find a small issue with the download script. It tells you to run the wrong command to download the faucet assets... the given command says to download `parnet_mobility_cabinet`. You also need to run

```
python -m mani_skil.utils.download_asset partnet_mobility_faucet
```

Or just download all of them (its not that big)

```
python -m mani_skil.utils.download_asset parnet_mobility
```

jstmn, 2025-05-28:
Ok great, it's fixed now. Thanks!