mani-skill/ManiSkill #1012: Why is numpy<2.0.0 required? Any plan to support numpy 2.x?

songyuc, 2025-04-20:
Hi, I noticed that ManiSkill currently requires `numpy<2.0.0`. Is there any particular reason for not supporting numpy 2.x? Are there known compatibility issues, or is it just not tested yet?

Are there any plans to support numpy 2.x in the near future?  
Thanks for your great work!

(My environment is Python 3.13, Ubuntu 22.04, and I'm using the latest pip.)

StoneT2000, 2025-04-22:
We can support numpy 2 but I have avoided it because it would take some time and majority of users aren't on numpy 2 still.

What packages / reasons require you to use numpy 2

songyuc, 2025-04-25:
Thank you for your prompt and detailed response!

To clarify, I currently do not have specific packages or reasons that require `numpy 2.x`. My motivation is simply to stay up-to-date with the latest version of NumPy and maintain a modern and compatible development environment. Using the latest versions of foundational libraries like NumPy often helps ensure better performance, access to new features, and compatibility with future updates of other tools and libraries.

I fully understand that most users are still on `numpy 1.x`, and upgrading might not be an immediate priority. However, I believe adding support for `numpy 2.x` sooner rather than later could help avoid potential compatibility issues down the road, especially as the ecosystem gradually transitions to newer versions. 

If there is anything I can do to help, such as testing ManiSkill with `numpy 2.x` or providing feedback on potential issues, I’d be happy to assist.

Thank you again for your fantastic work on ManiSkill, and I really appreciate your consideration!

Best regards,  
songyuc

StoneT2000, 2025-05-28:
It seems one of our dependencies requires < 2 right now preventing an upgrade here: https://github.com/UM-ARM-Lab/pytorch_kinematics/issues/54

StoneT2000, 2025-09-14:
Thanks for helping bump that up. I'll put out a new release

cvoelcker, 2025-10-08:
Hey @StoneT2000 , any news on this? We need this as well to make our updated REPPO fully compatible with maniskill

StoneT2000, 2025-10-21:
testing this now! dependencies have all upgraded so we can too

jstmn, 2025-10-21:
awesome!!

jstmn, 2025-12-18:
Hey, FYI it seems like 'numpy>2' is causing issues for mplib: https://github.com/haosulab/ManiSkill/issues/1359#issuecomment-3668588035