mani-skill/ManiSkill #1014: How to step env with "pd_joint_pose" using delta ee action?

Jimntu, 2025-04-21:
If I have a 7-dimensional end-effector delta action (x, y, z, roll, pitch, yaw, gripper), how can I use this to step an environment with control_mode = "pd_joint_pos"?

As I understand it, this control mode expects an 8-dimensional action vector representing the target positions for each joint (7 arm joints + 1 gripper). So to use my 7D delta action, I would need to:

1. Use forward kinematics (FK) to get the current end-effector pose.

2. Apply the delta action to compute the new target pose.

3. Use inverse kinematics (IK) to convert the new target pose into joint positions.

4. Combine the resulting joint positions with the gripper value to form the 8D action vector.

5. Pass it to env.step().

Are there any built-in functions in ManiSkill or related libraries that can help perform FK and IK for this process? 

Also, does the 8-dim action in "pd_joint_pos" represent 7 revolute joints' rotation in radian + 1 gripper?

Thanks in advance!

103240, 2025-04-21:
我也有同样的疑惑！

StoneT2000, 2025-04-21:
The pd_ee_delta_pose controller of the panda arm for example does exactly what you propose already. You can look at the panda robot code to see how that is done

And for panda it is 7 resolute joints + one prismatic joint (that mimic controls 2 fingers)

StoneT2000, 2025-04-22:
For that you can do what you proposed. From a delta pose apply it to current pose, then use IK with initial condition being the current qpos to find the next qpos and send that as action.

StoneT2000, 2025-04-22:
The green ball in PickCube is hidden, only for user to see, not policy.

The goal info is provided as part of the extras key in the observation (if using visual obs mode or state dict) as a 3d position

Jimntu, 2025-04-22:
> The green ball in PickCube is hidden, only for user to see, not policy.
> 
> The goal info is provided as part of the extras key in the observation (if using visual obs mode or state dict) as a 3d position
So for a VLA model, isn't it quite difficult to reach the goal if the image doesn't contain any visual information about the goal?

Thanks!

Jimntu, 2025-04-22:
So for a VLA model, isn't it quite difficult to reach the goal if the image doesn't contain any visual information about the goal?

closed 2025-05-13 by StoneT2000