mani-skill/ManiSkill #1018: pd_joint_pos转换成pd_ee_delta_pose

103240, 2025-04-22:
您好，我想把数据集的pd_joint_pos转换成pd_ee_delta_pose，然后作为自定义数据集微调openvla，请问我该如何实现？

Lr-2002, 2025-04-22:
可以尝试用官网上的replay episode，但是实际效果不行，任务复杂的话，成功率更低

103240, 2025-04-23:
谢谢你的回复，我依然有点疑问，是说openvla微调后成功率不好吗，replay episode转化成功率如何呢，我的任务是pickcube和stackcube

> 可以尝试用官网上的replay episode，但是实际效果不行，任务复杂的话，成功率更低

StoneT2000, 2025-06-14:
@Lr-2002 is correct. For very complex tasks it is very difficult to convert pd joint pos action space to pd ee delta pose. This is an open unsolved problem at the moment, would also be an interesting research project.

In general you can use our replay trajectory tool to try to convert action spaces, after that then modify the saved dataset to the RLDS format openvla uses.

closed 2025-06-14 by StoneT2000