mani-skill/ManiSkill #1021: Unable to replay trajectory when --num_envs > 1

xucj98, 2025-04-24:
I want to replay trajectory with `physx_cpu`, it works when `num_envs` = 1, however fails when `num_envs` > 1. Error code:
```
RuntimeError: Cannot set the sim backend to 'cpu' and have multiple environments.
            If you want to do CPU sim backends and have environment vectorization you must use multi-processing across CPUs.
            This can be done via the gymnasium's AsyncVectorEnv API
```
I got the trajectory by motionplanning:
```
python -m mani_skill.examples.motionplanning.panda.run -e PegInsertionSide-v1 -n 1000 --only-count-success --num-procs 20
```
Then I want to replay trajectory to get observation and action:
```
python -m mani_skill.trajectory.replay_trajectory --traj-path demos/PegInsertionSide-v1/motionplanning/20250424_114737.h5 --save-traj -c pd_ee_delta_pose -o rgbd --no-allow-failure --num_envs 10
```

StoneT2000, 2025-04-24:
what maniskill version?

xucj98, 2025-04-24:
ubuntu 22.04
python 3.12
mani_skill 3.0.0b20

StoneT2000, 2025-05-03:
I see the bug, I will add a fix soon

PR #1034 (2025-05-03): [BugFix] fix bug for cpu sim replay
Closes #1021