mani-skill/ManiSkill #1027: Visuo-tactile demonstrations sequences for a few tasks in maniskill

pankhurivanjani, 2025-04-28:
Hello!

Do we have demonstrations with visual and tactile modality that could be used for training and evaluation in the Maniskill3 framework? 
I checked out mani-vitac challenge as well, but didn't find dataset yet.

Thanks,

pankhurivanjani, 2025-04-28:
I can see the demonstrations here for peg insertion task : [https://huggingface.co/datasets/haosulab/ManiSkill_PegInsertionSide](https://huggingface.co/datasets/haosulab/ManiSkill_PegInsertionSide) but It is not clear if it has tactile information also, and if it is the same task as the one mani-vitac challenge has.

StoneT2000, 2025-04-30:
This peg insertion task is not the same as the one in ViTac challenge. No tactile information is given in the maniskill3 task.

What kind of tactile information do you want? If you make your own task you can include pairwise contact forces as part of the observation or serve as tactile data. Alternatively I recommend just raising an issue on the ViTac Challenge github

closed 2025-05-13 by StoneT2000