mani-skill/ManiSkill #1037: [Docs] fix action repeat docs to reference correct image path and include in toc tree somewhere

StoneT2000, 2025-05-03:


StoneT2000, 2025-05-03:
@AlexandreBrown didn't catch the issue until now https://maniskill.readthedocs.io/en/latest/user_guide/wrappers/action_repeat.html

this page also does not show up in the TOC anywhere or linked anywhere so not easy to find atm.

AlexandreBrown, 2025-05-07:
I see, the preview was working correctly when checking the PR but when the site is deployed then the relative path does not work.     
Opened a PR to fix this : https://github.com/haosulab/ManiSkill/pull/1046  
Let me know if the fix makes sense, I tried to infer what to do to make it show in the wrappers dropdown.

AlexandreBrown, 2025-05-08:
@StoneT2000 It looks like the PR fixed the menu not having action repeat but the image still does not show. Is there a way to test this locally? What path should I use? https://maniskill.readthedocs.io/en/latest/user_guide/wrappers/action_repeat.html

AlexandreBrown, 2025-05-26:
I guess now that the asset is on main we could just use this path : https://raw.githubusercontent.com/haosulab/ManiSkill/refs/heads/main/docs/source/user_guide/wrappers/images/action_repeat.svg

closed 2025-06-18 by StoneT2000