mani-skill/ManiSkill #1042: Maniskill 3 in rlds format

Jimntu, 2025-05-06:
Hi, thanks for the great work!

I was wondering if there's a way to convert ManiSkill3 datasets into the RLDS format so I can fine-tune OpenVLA on them.

Thanks again!

StoneT2000, 2025-05-06:
I believe @xuanlinli17 has a repository for ManiSkill2 data that you can reference. The RLDS part is not that complicated but I recall that OpenVLA/Octo etc. expect very specific keys and image sizes that aren't so well documented, so a reference repo is needed.

sijitu666, 2025-08-21:
> Hi, thanks for the great work!
> 
> I was wondering if there's a way to convert ManiSkill3 datasets into the RLDS format so I can fine-tune OpenVLA on them.
> 
> Thanks again!

Hi @xuanlinli17,

I'm also very interested in converting ManiSkill datasets to RLDS format. Following @StoneT2000's suggestion, I wanted to ask if you have a conversion script or a reference repository you could share?

It would be a great help. Thank you!