mani-skill/ManiSkill #1050: Digital Twin Environments GPU Sim speed is too slow

QuantuMope, 2025-05-08:
Hello, I'm trying to use the digital twin bridge environments for some RL training, but I found that the GPU step time is extremely slow for these scenes. Below is a slightly modified code excerpt from https://github.com/haosulab/ManiSkill/issues/355#issuecomment-2143320896 for profiling.

```python
from mani_skill.envs.tasks import PickCubeEnv
from mani_skill.envs.tasks.digital_twins.bridge_dataset_eval.put_on_in_scene import PutCarrotOnPlateInScene
from mani_skill.sensors.camera import CameraConfig
from mani_skill.utils import sapien_utils
from mani_skill.utils.registration import register_env
import gymnasium as gym
import mani_skill.envs
import torch
import numpy as np
import time
@register_env("TestPickCube-v1", max_episode_steps=50, override=True)
# class TestPickCubeEnv(PickCubeEnv):
class TestPickCubeEnv(PutCarrotOnPlateInScene):

    """modifies pick cube to use custom camera configs and put two of them for testing"""
    @property
    def _default_sensor_configs(self):
        return []
    # @property
    # def _default_sensor_configs(self):
    #     pose = sapien_utils.look_at(eye=[0.3, 0, 0.6], target=[-0.1, 0, 0.1])
    #     return [CameraConfig("base_camera", pose, 160, 120, np.pi / 2, 0.01, 100)]
num_envs = 128
env = gym.make("TestPickCube-v1", num_envs=num_envs, obs_mode="rgb+segmentation",
               sim_config=dict(sim_freq=200, control_freq=10, scene_config=dict(solver_position_iterations=8)))
env.unwrapped.print_sim_details()
obs, _ = env.reset(seed=0)
done = False
start_time = time.time()
total_rew = 0
TIMESTEPS = 100
for i in range(TIMESTEPS):
    # note that env.action_space is now a batched action space
    obs, rew, terminated, truncated, info = env.step(torch.from_numpy(env.action_space.sample()))

N = num_envs * TIMESTEPS
dt = time.time() - start_time
FPS = N / (dt)
print(f"Frames Per Second = {N} / {dt} = {FPS}")
print(f"Parallel Steps Per Second = {TIMESTEPS / dt}")
```

I then perform runs for both `PickCubeEnv` and `PutCarrotOnPlateInScene` (digital twin) on a 4090. Both have just a single 160x120 camera. Notice the massive difference in results:

```
PickCubeEnv
# -------------------------------------------------------------------------- #
Task ID: TestPickCube-v1, 128 parallel environments, sim_backend=physx_cuda
obs_mode=rgb+segmentation, control_mode=pd_joint_delta_pos
render_mode=None, sensor_details=RGBD(160x120)
sim_freq=200, control_freq=10
observation space: Dict('agent': Dict('qpos': Box(-inf, inf, (128, 9), float32), 'qvel': Box(-inf, inf, (128, 9), float32)), 'extra': Dict('is_grasped': Box(False, True, (128,), bool), 'tcp_pose': Box(-inf, inf, (128, 7), float32), 'goal_pos': Box(-inf, inf, (128, 3), float32)), 'sensor_param': Dict('base_camera': Dict('extrinsic_cv': Box(-inf, inf, (128, 3, 4), float32), 'cam2world_gl': Box(-inf, inf, (128, 4, 4), float32), 'intrinsic_cv': Box(-inf, inf, (128, 3, 3), float32))), 'sensor_data': Dict('base_camera': Dict('segmentation': Box(-32768, 32767, (128, 120, 160, 1), int16), 'rgb': Box(0, 255, (128, 120, 160, 3), uint8))))
(single) action space: Box(-1.0, 1.0, (8,), float32)
# -------------------------------------------------------------------------- #
Frames Per Second = 12800 / 4.682096242904663 = 2733.818216444688
Parallel Steps Per Second = 21.357954815974125
```

```
PutCarrotOnPlateInScene
# -------------------------------------------------------------------------- #
Task ID: TestPickCube-v1, 128 parallel environments, sim_backend=physx_cuda
obs_mode=rgb+segmentation, control_mode=arm_pd_ee_target_delta_pose_align2_gripper_pd_joint_pos
render_mode=None, sensor_details=RGBD(160x120)
sim_freq=200, control_freq=10
observation space: Dict('agent': Dict('qpos': Box(-inf, inf, (128, 8), float32), 'qvel': Box(-inf, inf, (128, 8), float32), 'controller': Dict('arm': Dict('target_pose': Box(-inf, inf, (128, 7), float32)))), 'extra': Dict(), 'sensor_param': Dict('3rd_view_camera': Dict('extrinsic_cv': Box(-inf, inf, (128, 3, 4), float32), 'cam2world_gl': Box(-inf, inf, (128, 4, 4), float32), 'intrinsic_cv': Box(-inf, inf, (128, 3, 3), float32))), 'sensor_data': Dict('3rd_view_camera': Dict('rgb': Box(0, 255, (128, 120, 160, 3), uint8), 'segmentation': Box(-32768, 32767, (128, 120, 160, 1), int16))))
(single) action space: Box([-1.        -1.        -1.        -1.5707964 -1.5707964 -1.5707964
 -1.       ], [1.        1.        1.        1.5707964 1.5707964 1.5707964 1.       ], (7,), float32)
# -------------------------------------------------------------------------- #
Frames Per Second = 12800 / 18.619595766067505 = 687.4477921441678
Parallel Steps Per Second = 5.370685876126311
```

Do you have any idea what could be causing such a large discrepancy?

AlexandreBrown, 2025-05-08:
Hi @QuantuMope , maybe this section has information that could help you with this ? https://maniskill.readthedocs.io/en/latest/user_guide/tutorials/custom_tasks/advanced.html#task-sim-configurations

QuantuMope, 2025-05-08:
I don't think `solver_iterations` is the issue. Both are set to identical values in the above examples. The issue is that using IK results in an Nx slowdown over using joint control.

StoneT2000, 2025-05-08:
Hi @QuantuMope 

The GPU sim speed is slow for those environments due to a few reasons
- IK solver used in ManiSkill for absolute end-effector control is slow. Even if you lower the solver iterations of the IK solver it is still not that fast really. @jstmn brought this up and has a solution #955. I have been actively looking for some better IK libraries / write some of our own but have not found anything of good enough quality yet that doesn't depend on jax (https://github.com/chungmin99/pyroki/ looks very good but jax dependency is an issue)
- For various real2sim alignment reasons (although not always needed) the sim to control frequency ratio is 20 (also known as decimation). This means 20 physics steps are executed per action, so the FPS is "20x" lower than perhaps expected. I think the original reasoning for such a high ratio is to ensure the simulator gets to the desired joint positions as closely as possible with minimal error to better match the real world. Normally we default to using a sim freq of 100 control freq of 20 (ratio of 5), and some other 

Last point I would like to point out is that the digital twin environments are not in any way designed for RL training, only for evaluation of any existing models. There are no dense rewards and we have not tuned the real2sim digital twins to run fast for RL like we have for other environments such as PickCube.

QuantuMope, 2025-05-08:
Thanks for the response @StoneT2000. Responding in a similar structure.

- It seems then that IK is quite the bottleneck. I'm curious if this Nx slowdown has been experienced from your benchmarks before?
- The sim to control frequency ratio is something I chose and is different from the actual one used in the real2sim examples (sim_freq=500, control_freq=5). I've tried setups like (100, 20), but those still experience a slowdown of several folds due to IK.

For the last point, I'm not in need of any rewards and won't actually being using ManiSkill's RL framework at all. I just need to extract the observations and apply actions to the environment and everything else will be handled from a different framework. Doing a quick test with replacing the IK with a joint controller allows the digital twin env to reach similar speeds of PickCube.

Though I've recently now opened another issue because GPU parallel joint control for the widow does not seem t work: https://github.com/haosulab/ManiSkill/issues/1052

StoneT2000, 2025-05-08:
I can place improving the IK as a higher priority, implementing @jstmn's solution perhaps. Would also close #884 (the delta ee control has large error in GPU sim at the moment due to poor implementation). 

I will take a look at the joint position control of the widowx arm later. I'm guessing it stems from a poor implementation (copied directly over from maniskill2 and we only ensured that the real2sim envs worked, but did not really tune it otherwise).

For now I will close this issue as this is technically duplicate of #955.

QuantuMope, 2025-05-08:
It seems like from #955, the reason for the slowdown is using a target delta ee pose control. Curious why this is so much slower than delta ee pose. They should be identical I would imagine.

StoneT2000, 2025-05-08:
Delta ee pose uses a fast pseudo inverse IK based solving trick. Similar to some other GPU sim frameworks. It's fast but somewhat inaccurate.

QuantuMope, 2025-05-08:
I think you can use the same pseudo inverse IK for both? Only thing that should change is the initial guess: joint pos from sensor readings vs. previous joint target.