mani-skill/ManiSkill #1052: GPU Joint control doesn't work for WidowX

QuantuMope, 2025-05-08:
GPU parallel joint control doesn't seem to work for the WidowX robot. I've been trying to debug this, but can't seem to find the issue. Below is a modified agent of `WidowX250sBridgeDatasetFlatTable`. The only changes is to use joint position control + adding an extra front view camera.

```python
class Hobot2BridgeFlatTable(WidowX250SBridgeDatasetFlatTable):

    @property
    def _sensor_configs(self):

        scale = 3
        width = 228 * scale
        height = 128 * scale
        intrinsic = np.array([[124.409, 0, 114.0], [0, 124.409, 64.0],
                              [0, 0, 1]])
        intrinsic[:2] *= scale

        return super()._sensor_configs + [
            CameraConfig(
                uid="external_camera",
                # the camera used for real evaluation for the sink setup
                pose=sapien.Pose(
                    [0.80, 0.0, 0.23],
                    [0.0, 0.0, 0.0, 1.0],
                ),
                entity_uid="base_link",
                width=width,
                height=height,
                near=0.01,
                far=2.0,
                intrinsic=intrinsic,
            ),
        ]

    @property
    def _controller_configs(self):
        arm_common_kwargs = dict(
            joint_names=self.arm_joint_names,
            lower=None,
            upper=None,
            stiffness=np.array(self.arm_stiffness) * 2.0,
            damping=self.arm_damping,
            force_limit=self.arm_force_limit,
            friction=self.arm_friction,
            use_delta=False,
            interpolate=True,
            normalize_action=False,
        )
        arm_pd_absolute_joint_target = PDJointPosControllerConfig(
            **arm_common_kwargs)

        gripper_pd_joint_pos = PDJointPosMimicControllerConfig(
            joint_names=self.gripper_joint_names,
            lower=-0.04,
            upper=0.04,
            stiffness=self.gripper_stiffness,
            damping=self.gripper_damping,
            force_limit=self.gripper_force_limit,
            normalize_action=True,
            drive_mode="force",
        )
        controller = dict(arm=arm_pd_absolute_joint_target,
                          gripper=gripper_pd_joint_pos)
        return dict(
            arm_pd_absolute_joint_target_gripper_pd_joint_pos=controller)
```

If I run this in an environment with 1 env, everything works as expected
```python
env = gym.make("Hobot2RedCubeTowelPlateScene",
               obs_mode="rgb+depth+segmentation",
               num_envs=1)
obs, _ = env.reset()

while True:
    img0 = obs["sensor_data"]["external_camera"]["rgb"].cpu().numpy()[0]
    action = env.action_space.sample()

    # Do all 0.1s
    action[:] = 0.1
    obs = env.step(action)[0]

    print(obs["agent"]["qpos"])

    cv2.imshow("Over Shoulder", cv2.cvtColor(img0, cv2.COLOR_RGB2BGR))
    cv2.waitKey(100)

# Print Output
tensor([[0.0165, 0.0600, 0.1745, 0.0291, 0.7961, 0.0331, 0.0240, 0.0240]])
tensor([[0.0412, 0.0732, 0.1446, 0.0519, 0.4830, 0.0545, 0.0162, 0.0162]])
tensor([[0.0586, 0.0819, 0.1267, 0.0673, 0.3107, 0.0691, 0.0150, 0.0150]])
tensor([[0.0709, 0.0878, 0.1160, 0.0777, 0.2159, 0.0790, 0.0150, 0.0150]])
tensor([[0.0795, 0.0918, 0.1096, 0.0848, 0.1638, 0.0857, 0.0150, 0.0150]])
tensor([[0.0856, 0.0945, 0.1057, 0.0897, 0.1351, 0.0903, 0.0150, 0.0150]])
tensor([[0.0898, 0.0963, 0.1034, 0.0930, 0.1193, 0.0934, 0.0150, 0.0150]])
tensor([[0.0928, 0.0975, 0.1021, 0.0952, 0.1106, 0.0955, 0.0150, 0.0150]])
tensor([[0.0950, 0.0983, 0.1012, 0.0968, 0.1058, 0.0970, 0.0150, 0.0150]])
tensor([[0.0965, 0.0989, 0.1007, 0.0978, 0.1032, 0.0979, 0.0150, 0.0150]])
tensor([[0.0975, 0.0992, 0.1004, 0.0985, 0.1018, 0.0986, 0.0150, 0.0150]])
tensor([[0.0982, 0.0995, 0.1003, 0.0990, 0.1010, 0.0990, 0.0150, 0.0150]])
tensor([[0.0988, 0.0997, 0.1002, 0.0993, 0.1005, 0.0993, 0.0150, 0.0150]])
tensor([[0.0991, 0.0998, 0.1001, 0.0996, 0.1003, 0.0995, 0.0150, 0.0150]])
tensor([[0.0994, 0.0998, 0.1001, 0.0997, 0.1002, 0.0997, 0.0150, 0.0150]])
tensor([[0.0996, 0.0999, 0.1000, 0.0998, 0.1001, 0.0998, 0.0150, 0.0150]])
tensor([[0.0997, 0.0999, 0.1000, 0.0999, 0.1000, 0.0999, 0.0150, 0.0150]])
tensor([[0.0998, 0.1000, 0.1000, 0.0999, 0.1000, 0.0999, 0.0150, 0.0150]])
tensor([[0.0998, 0.1000, 0.1000, 0.0999, 0.1000, 0.0999, 0.0150, 0.0150]])
tensor([[0.0999, 0.1000, 0.1000, 0.1000, 0.1000, 0.1000, 0.0150, 0.0150]])
tensor([[0.0999, 0.1000, 0.1000, 0.1000, 0.1000, 0.1000, 0.0150, 0.0150]])
tensor([[0.0999, 0.1000, 0.1000, 0.1000, 0.1000, 0.1000, 0.0150, 0.0150]])
tensor([[0.1000, 0.1000, 0.1000, 0.1000, 0.1000, 0.1000, 0.0150, 0.0150]])
tensor([[0.1000, 0.1000, 0.1000, 0.1000, 0.1000, 0.1000, 0.0150, 0.0150]])

```

[image]

As soon as I enable GPU sim via num_envs>1, the actions get completely ignored and all environments simply go to all zeros. This is the result for num_envs=2

```python

# Print output
tensor([[-0.0091,  0.0179,  0.0765, -0.0023,  0.3861,  0.0007,  0.0240,  0.0240],
        [-0.0091,  0.0179,  0.0765, -0.0023,  0.3861,  0.0007,  0.0240,  0.0240]],
       device='cuda:0')
tensor([[-0.0045,  0.0075,  0.0259, -0.0015,  0.1091,  0.0003,  0.0161,  0.0162],
        [-0.0045,  0.0075,  0.0259, -0.0015,  0.1091,  0.0003,  0.0161,  0.0162]],
       device='cuda:0')
tensor([[-0.0022,  0.0032,  0.0088, -0.0009,  0.0308,  0.0001,  0.0150,  0.0150],
        [-0.0022,  0.0032,  0.0088, -0.0009,  0.0308,  0.0001,  0.0150,  0.0150]],
       device='cuda:0')
tensor([[-1.0797e-03,  1.3620e-03,  2.9706e-03, -5.1347e-04,  8.7176e-03,
          6.2972e-05,  1.5000e-02,  1.5000e-02],
        [-1.0797e-03,  1.3620e-03,  2.9706e-03, -5.1347e-04,  8.7176e-03,
          6.2972e-05,  1.5000e-02,  1.5000e-02]], device='cuda:0')
tensor([[-5.3120e-04,  5.8806e-04,  1.0094e-03, -2.5339e-04,  2.4637e-03,
          2.8457e-05,  1.5000e-02,  1.5000e-02],
        [-5.3120e-04,  5.8806e-04,  1.0094e-03, -2.5339e-04,  2.4637e-03,
          2.8457e-05,  1.5000e-02,  1.5000e-02]], device='cuda:0')
tensor([[-2.6135e-04,  2.5551e-04,  3.4331e-04, -1.1911e-04,  6.9616e-04,
          1.2863e-05,  1.5000e-02,  1.5000e-02],
        [-2.6135e-04,  2.5551e-04,  3.4331e-04, -1.1911e-04,  6.9616e-04,
          1.2863e-05,  1.5000e-02,  1.5000e-02]], device='cuda:0')
tensor([[-1.2858e-04,  1.1153e-04,  1.1682e-04, -5.4509e-05,  1.9668e-04,
          5.8112e-06,  1.5000e-02,  1.5000e-02],
        [-1.2858e-04,  1.1153e-04,  1.1682e-04, -5.4509e-05,  1.9668e-04,
          5.8112e-06,  1.5000e-02,  1.5000e-02]], device='cuda:0')
tensor([[-6.3263e-05,  4.8844e-05,  3.9753e-05, -2.4560e-05,  5.5550e-05,
          2.6228e-06,  1.5000e-02,  1.5000e-02],
        [-6.3263e-05,  4.8844e-05,  3.9753e-05, -2.4560e-05,  5.5550e-05,
          2.6228e-06,  1.5000e-02,  1.5000e-02]], device='cuda:0')
tensor([[-3.1125e-05,  2.1442e-05,  1.3521e-05, -1.0966e-05,  1.5684e-05,
          1.1823e-06,  1.5000e-02,  1.5000e-02],
        [-3.1125e-05,  2.1442e-05,  1.3521e-05, -1.0966e-05,  1.5684e-05,
          1.1823e-06,  1.5000e-02,  1.5000e-02]], device='cuda:0')
tensor([[-1.5314e-05,  9.4292e-06,  4.5938e-06, -4.8709e-06,  4.4255e-06,
          5.3224e-07,  1.5000e-02,  1.5000e-02],
        [-1.5314e-05,  9.4292e-06,  4.5938e-06, -4.8709e-06,  4.4255e-06,
          5.3224e-07,  1.5000e-02,  1.5000e-02]], device='cuda:0')
```

[image]

I've known about this issue for a bit now and decided to use IK controllers instead because those seem to work. After some profiling though in issue https://github.com/haosulab/ManiSkill/issues/1050, I now want to use joint control sim instead for the significant speedup, but it seems broken for this case.

QuantuMope, 2025-05-15:
The reason for this is because of setting `interpolate=True`. Seems like this arg will mess up all control modes for gpu sim.

StoneT2000, 2025-05-16:
thanks for finding that out. Just don't use that option for now (none of our example code does anyway). I created a new issue for that.