mani-skill/ManiSkill #1057: Digital Twin / GreenScreening

abadithela, 2025-05-09:
Hi,

Thank you very much for the excellent tool. I am working on building a digital twin for the Franka Robot setup that we have in our lab in ManiSkill3. For this, I need to do a green-screening of the real-world experimental setup. We already have realistic object and robot visualizations, so the piece we need is visual matching by green screening the background image. I am aware that this is still under development for ManiSkill3, but I thought to follow the existing code to develop it for my setup. I am mainly looking at the files in the digital twins folder: https://github.com/haosulab/ManiSkill/tree/bfeb7ddc51479baa57981d1ed0c2fa6bb1b15874/mani_skill/envs/tasks/digital_twins and sapien_env.py

However, I have a few questions regarding this, so I am opening this thread. 

First, It seems that the [BaseDigitalTwinEnv](https://github.com/haosulab/ManiSkill/blob/bfeb7ddc51479baa57981d1ed0c2fa6bb1b15874/mani_skill/envs/tasks/digital_twins/base_env.py) requries:

Setting up the sensors: [self._sensor_configs](https://github.com/haosulab/ManiSkill/blob/bfeb7ddc51479baa57981d1ed0c2fa6bb1b15874/mani_skill/envs/tasks/digital_twins/base_env.py#L107). In the Bridge example, these seem to be hardcoded in the function: [bridge_sensor_config](https://github.com/haosulab/ManiSkill/blob/bfeb7ddc51479baa57981d1ed0c2fa6bb1b15874/mani_skill/envs/tasks/digital_twins/bridge_dataset_eval/base_env.py#L41C5-L57C10)

At the same time, there is a function in sapien_env.py with a TODO note: [_setup_sensors](https://github.com/haosulab/ManiSkill/blob/b3f2cfdb20e4ff7494d50cf686e93c87a495986c/mani_skill/envs/sapien_env.py#L753)

Is there a quick way to integrate this function to the sensor API, or would you recommend that we hardcode it?

Best,
Apurva

StoneT2000, 2025-05-09:
Greenscreening is quite easy. You can follow the base digital twin env example code. Either inherit that BaseDigitalTwinEnv directly or copy relevant snippets of code. Inheriting the env might be easier, it is the same as BaseEnv with an added `self.remove_object_from_greenscreen` function. 

For example the SIMPLER environments only remove the target objects/to-be-manipulated objects and the robot from the greenscreen

If you choose to not inherit the BaseDigitalTwinEnv you can also just read and copy the code there. It essentially uses the rgb+segmentation observation mode, overrides the original BaseEnv's image obs generation function to perform greenscreening, and will selectively choose to not green-screen certain actors/links.

Ignore the TODO note, that is old. I was thinking about refactoring some old code but decided not to, sensors api is mostly fine as it is, its not related to green-screening.

abadithela, 2025-05-09:
Got it! Thanks!
I am seeing a different issue in the green_screen_rgb method at the following line:

`rgb = rgb * (~mask) + overlay_img * mask`

For some reason, the rgb tensor has four channels while overlay_image has 3:

```
Why am I seeing the following error:

rgb = rgb * (~mask) + overlay_img * mask
*** RuntimeError: The size of tensor a (4) must match the size of tensor b (3) at non-singleton dimension 3

The tensor sizes are as follows:

(Pdb) mask.shape
torch.Size([1, 480, 640, 1])

rgb.shape
torch.Size([1, 480, 640, 4])

overlay_img.shape
torch.Size([480, 640, 3])
```
The object "obs" is from:
`obs = super()._get_obs_sensor_data(apply_texture_transforms)`

StoneT2000, 2025-05-09:
Ah it looks like we didn’t update the get sensor images function for the digital twin envs. We can update the base digital twin envs class to also update and green screen in that function.

Note that that function is not used for policy observation normally, it is purely for visualization. What we might do is add a third image showing the green screen.

Do env.reset or env.get_obs or env.step to get the policy observation which should include the green screened result.

abadithela, 2025-05-09:
So, just trying env.get_obs(), the obs from env.reset() does not change the result. I get the same output. I can setup another function showing the greenscreened image.

StoneT2000, 2025-05-09:
Background mode is the correct one.

Did you remove the objects you wanted to keep out of the green screen process? Maybe share a code snippet?

And are you saying the environment observations are returning two images, the green screen and segmentation as a single image?

abadithela, 2025-05-10:
Sorry, to clarify: the segmentation figure is fine. I meant that earlier, I could not see the green screened image with 50% opacity due to an error on my end which did not change the rib_overlay_mode to debug. I fixed this and I get the image above. 

Now, I am trying to resolve:
1. Yes, I am wondering why the tomato and pot are not visible after green screening, and what I must do to fix it. This is the code that sets objects:

```
    def _load_scene(self, options: dict):
        # we use a prebuilt scene builder class that automatically loads in a floor and table.
        self.table_scene = Real2SimTabletopSceneBuilder(
            env=self,
            cfg=self.base_cfg.scene_builder,
            robot_init_qpos_noise=self.base_cfg.scene_builder.robot_init_qpos_noise,
        )
        self.table_scene.build(options)
        self.cam_mount = [
            self.scene.create_actor_builder().build_kinematic(f"camera_mount_{i}")
            for i in range(len(self.base_cfg.camera.sensor.eye))
        ]
        # self.wrist_mount = self.scene.create_actor_builder().build_kinematic(
        #     "wrist_mount"
        # )
        self.build_distractor(options)

        sapien.set_log_level("off")
        self._build_pick_and_place_obj()

        # Excluding objects from greenscreening
        for obj_name in self.objects_excluded_from_greenscreening:
            self.remove_object_from_greenscreen(self.objs[obj_name])
        self.remove_object_from_greenscreen(self.agent.robot)
```
The list `self.objects_excluded_from_greenscreening` is empty.

2. How to update/adjust the camera_config. For the BridgeDataset digital twin example, I see that this was hardcoded. In my code I am trying to call it before the reset instead of hardcoding it. 

```
env_config["camera_pose"] = np.array([[-0.56389985,  0.51223484,  0.5907 , -0.76089205, -0.25114999 ,-0.32208765,0.5042088 ]] )
  env_config.update({"reconfigure": reconfigure})
  obs, info = env.reset(options=env_config)
```

The _setup_sensors method in sapiens_env.py does not change the camera config based on the options passed in..

StoneT2000, 2025-05-12:
1)

env.unwrapped._segmentation_ids_to_keep  could you print that? During reconfiguration we clear out the list you mentioned to release all references to those objects to free memory so it's expected to be empty

2)

For changing camera config you can't do it via env reset options since there isn't code from maniskill that parses that. Following gym convention that reset options dictionary is meant to be primarily user defined/applied (we only use the reconfigure field in the options usually). 

Would it be a useful feature to allow changing default camera configs during env resets? Currently we only support overrides of sensor configs (and sim configs) during gym.make. Usually this kind of extra reset functionality should be defined by  you. 

One way to actually do it is to modify camera poses in the initialize episode function (which receives the options you pass in) although in gpu sim this only works for mounted cameras at the moment. I can probably add support for static cameras as well. (changing camera pose should work completely in cpu sim at least right now).

abadithela, 2025-05-12:
1. When I type `env.unwrapped._segmentation_ids_to_keep`, I get:

```
env.unwrapped._segmentation_ids_to_keep
tensor([ 1,  2,  3,  4,  5,  6,  7,  8,  9, 10, 11, 12, 13, 14, 15],
       device='cuda:0', dtype=torch.int32)
```

2. Got it. No, it's fine if I should just change the camera pose before I call gym.make. I was passing it in the reset phase since I was playing around with the green screening, and thought it might be an easy way to fiddle with the camera pose.

StoneT2000, 2025-05-12:
Oh I know the issue. I think you copied the code from the bridge/simpler example environments directly without adding the other necessary code.

The list self.objects_excluded_from_greenscreening is supposed to be filled by you with the objects you want. Since it is empty you aren't removing anything from the green screen except the robot

StoneT2000, 2025-05-20:
Does the above fix it for you @abadithela ?

StoneT2000, 2025-06-04:
Z is up. I am not sure what down and left are relatively? You can check which orientation x and y are, but without doubt z in your env is the natural "upwards direction".

In any case if you aren't sure try changing x, y sign values or swapping them, theres only 4 possible variations.

Did you check camera intrinsics? There may be some warping causing the images to look off. You can define camera intrinsics in the camera config as an alternative to using the fov argument.

closed 2025-06-06 by abadithela