mani-skill/ManiSkill #1058: pd_joint_delta_pos controller does not handle joint limit

zjhthu, 2025-05-10:
I find [pd_joint_delta_pos](https://github.com/haosulab/ManiSkill/blob/b3f2cfdb20e4ff7494d50cf686e93c87a495986c/mani_skill/agents/controllers/pd_joint_pos.py#L82) controller does not guarantee that the output joint angles stay within the limit range, because it only restricts delta_pos without limiting the final target_qpos. Is this acceptable?

For the [pd_ee_delta_pose](https://github.com/haosulab/ManiSkill/blob/v0.5.3/mani_skill2/agents/controllers/pd_ee_pose.py#L103) controller, when the joint angles exceed the limits, the inverse kinematics (IK) solving will fail, and then the controller will revert to the previous state. I found that this would cause the learned policy to keep the robot arm stuck in a certain position without movement.

I am using Maniskill2, but I suspect that Maniskill3 also has similar issues.

StoneT2000, 2025-05-20:
pd joint delta pos implicitly handles the joint limit. Internally in the physics engine setting joint angles beyond the limits will only cause the robot to try and reach that angle up to the limit, but avoid going over. This is the case in MS2 and MS3.

It is possible for the IK based controller to reach limits and lead to no solution in which case the robot does not move. Sometimes a singularity can occur. There isn't a trivial way to do some kind of RL action masking in the IK control space but you can try and penalize the robot for going too far / outside of a boundary with a reward penalty.

jpizarrom, 2025-06-25:
@StoneT2000 are there also limits in the case of using a real robotic arm such as the so100?