mani-skill/ManiSkill #1067: ppo.py supporting env

XxZzXxxx, 2025-05-15:
I wanna ask if ppo.py in the baseline supports Allegro's RotateSingleObjectInHand environment, as there may be a dimension error in ycb actor merge. But i compare the pick_cube env with the rotate env, donot see something different.

XxZzXxxx, 2025-05-15:
Saving eval videos to runs/RotateSingleObjectInHandLevel0-v1__ppo__1__1747327477/videos
Running training
####
args.num_iterations=390 args.num_envs=512 args.num_eval_envs=8
args.minibatch_size=800 args.batch_size=25600 args.update_epochs=4
####
Epoch: 1, global_step=0
Evaluating
Evaluated 400 steps resulting in 0 episodes
model saved to runs/RotateSingleObjectInHandLevel0-v1__ppo__1__1747327477/ckpt_1.pt
Traceback (most recent call last):
  File "/home/xzx/workbench/2025/ManiGen/ppo.py", line 480, in <module>
    main()
  File "/home/xzx/workbench/2025/ManiGen/ppo.py", line 328, in main
    next_obs, reward, terminations, truncations, infos = envs.step(clip_action(action))
  File "/home/xzx/miniconda3/envs/manigen/lib/python3.9/site-packages/mani_skill/vector/wrappers/gymnasium.py", line 148, in step
    obs, infos = self.reset(options=dict(env_idx=env_idx))
  File "/home/xzx/miniconda3/envs/manigen/lib/python3.9/site-packages/mani_skill/vector/wrappers/gymnasium.py", line 93, in reset
    obs, info = self._env.reset(seed=seed, options=options)
  File "/home/xzx/miniconda3/envs/manigen/lib/python3.9/site-packages/gymnasium/core.py", line 467, in reset
    return self.env.reset(seed=seed, options=options)
  File "/home/xzx/miniconda3/envs/manigen/lib/python3.9/site-packages/gymnasium/wrappers/order_enforcing.py", line 61, in reset
    return self.env.reset(**kwargs)
  File "/home/xzx/miniconda3/envs/manigen/lib/python3.9/site-packages/mani_skill/envs/sapien_env.py", line 855, in reset
    self._initialize_episode(env_idx, options)
  File "/home/xzx/miniconda3/envs/manigen/lib/python3.9/site-packages/mani_skill/envs/tasks/dexterity/rotate_single_object_in_hand.py", line 159, in _initialize_episode
    self._initialize_actors(env_idx)
  File "/home/xzx/miniconda3/envs/manigen/lib/python3.9/site-packages/mani_skill/envs/tasks/dexterity/rotate_single_object_in_hand.py", line 173, in _initialize_actors
    pose.raw_pose[:, 0:3] = new_pos
  File "/home/xzx/miniconda3/envs/manigen/lib/python3.9/site-packages/torch/utils/_device.py", line 104, in __torch_function__
    return func(*args, **kwargs)
RuntimeError: The expanded size of the tensor (512) must match the existing size (7) at non-singleton dimension 0.  Target sizes: [512, 3].  Tensor sizes: [7, 3]
I run the python ppo.py --env_id="RotateSingleObjectInHandLevel0-v1", and here is the error information

StoneT2000, 2025-05-16:
That environment might be too old and probably has a bug or two. I'll take a look after the neurips deadline.

I don't know if that task is even solvable. It was made by Yuzhe previously, but is mostly there as a reference implementation of something that simulates correctly with dexterous hand.

StoneT2000, 2025-05-20:
I also forgot to ask, what command did you run exactly?

XxZzXxxx, 2025-05-25:
sorry for reply late, i run the ppo.py --env_id="RotateSingleObjectInHandLevel0-v1"

XxZzXxxx, 2025-06-07:
maybe the reset function has some problems, and i think the initial functions need to be rewritten because reset part reuses them and didn't consider the parallel envs.

StoneT2000, 2025-09-04:
@CassieLiuuu I ran that command do not have any bugs, what error do you get?

StoneT2000, 2025-09-04:
@XxZzXxxx looks like the issue is that the RotateSingleObjectInHand task does not support partial environment resets at the moment. You should add `--no-partial-reset` to the training arguments.

That being said that suite of tasks is not tuned with RL yet, it is currently unsolved.

closed 2025-09-04 by StoneT2000