mani-skill/ManiSkill #1070: insane memory consumption jump during evaluation

kwang-12, 2025-05-16:
Hello.
I am observing ridiculous memory usage during evaluation which causes sim to stop.
1. Hardware:
nvidia H100 80GB HBM3 with xeon plat 8480+ and 225 GB.
2. code:
simplified unitree go2 env with custom reward functions.
3. nvidia driver versions:
NVIDIA-SMI 570.133.07             Driver Version: 570.133.07     CUDA Version: 12.8 
4. error message:
Epoch: 61, global_step=49152000
Evaluating
Traceback (most recent call last):
  File "/opt/conda/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/opt/conda/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/workspace/test_mskill/ppo.py", line 287, in <module>
    eval_obs, eval_rew, eval_terminations, eval_truncations, eval_infos = eval_envs.step(agent.get_action(eval_obs, deterministic=True))
  File "/opt/conda/lib/python3.9/site-packages/mani_skill/vector/wrappers/gymnasium.py", line 148, in step
    obs, infos = self.reset(options=dict(env_idx=env_idx))
  File "/opt/conda/lib/python3.9/site-packages/mani_skill/vector/wrappers/gymnasium.py", line 93, in reset
    obs, info = self._env.reset(seed=seed, options=options)
  File "/opt/conda/lib/python3.9/site-packages/mani_skill/utils/wrappers/record.py", line 376, in reset
    obs, info = super().reset(*args, seed=seed, options=options, **kwargs)
  File "/opt/conda/lib/python3.9/site-packages/gymnasium/core.py", line 467, in reset
    return self.env.reset(seed=seed, options=options)
  File "/opt/conda/lib/python3.9/site-packages/gymnasium/core.py", line 467, in reset
    return self.env.reset(seed=seed, options=options)
  File "/opt/conda/lib/python3.9/site-packages/gymnasium/wrappers/order_enforcing.py", line 61, in reset
    return self.env.reset(**kwargs)
  File "/opt/conda/lib/python3.9/site-packages/mani_skill/envs/sapien_env.py", line 874, in reset
    info = self.get_info()
  File "/opt/conda/lib/python3.9/site-packages/mani_skill/envs/sapien_env.py", line 1047, in get_info
    info.update(self.evaluate())
  File "/workspace/test_mskill/envs/env_flat.py", line 237, in evaluate
    ret = super().evaluate()
  File "/workspace/test_mskill/envs/env_flat.py", line 83, in evaluate
    is_fallen = self.agent.is_fallen()
  File "/workspace/test_mskill/agents/go2.py", line 75, in is_fallen
    forces = self.robot.get_net_contact_forces(["base",
  File "/opt/conda/lib/python3.9/site-packages/mani_skill/utils/structs/articulation.py", line 427, in get_net_contact_forces
    return self.get_net_contact_impulses(link_names) / self.scene.timestep
  File "/opt/conda/lib/python3.9/site-packages/mani_skill/utils/structs/articulation.py", line 388, in get_net_contact_impulses
    query.cuda_impulses.torch()
RuntimeError: CUDA error: an illegal memory access was encountered
CUDA kernel errors might be asynchronously reported at some other API call, so the stacktrace below might be incorrect.
For debugging consider passing CUDA_LAUNCH_BLOCKING=1
Compile with `TORCH_USE_CUDA_DSA` to enable device-side assertions.

Traceback (most recent call last):
  File "/opt/conda/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/opt/conda/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/workspace/test_mskill/ppo.py", line 287, in <module>
    eval_obs, eval_rew, eval_terminations, eval_truncations, eval_infos = eval_envs.step(agent.get_action(eval_obs, deterministic=True))
  File "/opt/conda/lib/python3.9/site-packages/mani_skill/vector/wrappers/gymnasium.py", line 148, in step
    obs, infos = self.reset(options=dict(env_idx=env_idx))
  File "/opt/conda/lib/python3.9/site-packages/mani_skill/vector/wrappers/gymnasium.py", line 93, in reset
    obs, info = self._env.reset(seed=seed, options=options)
  File "/opt/conda/lib/python3.9/site-packages/mani_skill/utils/wrappers/record.py", line 376, in reset
    obs, info = super().reset(*args, seed=seed, options=options, **kwargs)
  File "/opt/conda/lib/python3.9/site-packages/gymnasium/core.py", line 467, in reset
    return self.env.reset(seed=seed, options=options)
  File "/opt/conda/lib/python3.9/site-packages/gymnasium/core.py", line 467, in reset
    return self.env.reset(seed=seed, options=options)
  File "/opt/conda/lib/python3.9/site-packages/gymnasium/wrappers/order_enforcing.py", line 61, in reset
    return self.env.reset(**kwargs)
  File "/opt/conda/lib/python3.9/site-packages/mani_skill/envs/sapien_env.py", line 874, in reset
    info = self.get_info()
  File "/opt/conda/lib/python3.9/site-packages/mani_skill/envs/sapien_env.py", line 1047, in get_info
    info.update(self.evaluate())
  File "/workspace/test_mskill/envs/env_flat.py", line 237, in evaluate
    ret = super().evaluate()
  File "/workspace/test_mskill/envs/env_flat.py", line 83, in evaluate
    is_fallen = self.agent.is_fallen()
  File "/workspace/test_mskill/agents/go2.py", line 75, in is_fallen
    forces = self.robot.get_net_contact_forces(["base",
  File "/opt/conda/lib/python3.9/site-packages/mani_skill/utils/structs/articulation.py", line 427, in get_net_contact_forces
    return self.get_net_contact_impulses(link_names) / self.scene.timestep
  File "/opt/conda/lib/python3.9/site-packages/mani_skill/utils/structs/articulation.py", line 388, in get_net_contact_impulses
    query.cuda_impulses.torch()
RuntimeError: CUDA error: an illegal memory access was encountered
CUDA kernel errors might be asynchronously reported at some other API call, so the stacktrace below might be incorrect.
For debugging consider passing CUDA_LAUNCH_BLOCKING=1
Compile with `TORCH_USE_CUDA_DSA` to enable device-side assertions.

[2025-05-16 04:47:26.676] [SAPIEN] [critical] /buildAgent/work/eb2f45c4acc808a0/physx/source/gpucommon/src/PxgCudaMemoryAllocator.cpp
[2025-05-16 04:47:26.676] [SAPIEN] [critical] Mem free failed with error code 700!

[2025-05-16 04:47:26.676] [SAPIEN] [critical] /buildAgent/work/eb2f45c4acc808a0/physx/source/gpucommon/src/PxgCudaMemoryAllocator.cpp
[2025-05-16 04:47:26.676] [SAPIEN] [critical] /buildAgent/work/eb2f45c4acc808a0/physx/source/gpucommon/src/PxgCudaMemoryAllocator.cpp
[2025-05-16 04:47:26.676] [SAPIEN] [critical] /buildAgent/work/eb2f45c4acc808a0/physx/source/gpucommon/src/PxgCudaMemoryAllocator.cpp
[2025-05-16 04:47:26.676] [SAPIEN] [critical] /buildAgent/work/eb2f45c4acc808a0/physx/source/gpucommon/src/PxgCudaMemoryAllocator.cpp
[2025-05-16 04:47:26.677] [SAPIEN] [critical] /buildAgent/work/eb2f45c4acc808a0/physx/source/gpucommon/src/PxgCudaMemoryAllocator.cpp
CUDA error at /__w/SAPIEN/SAPIEN/3rd_party/sapien-vulkan-2/src/core/buffer.cpp 103: an illegal memory access was encountered
scripts/run_flat_v2.sh: line 15:  3742 Segmentation fault      (core dumped) python3 -m test_mskill.ppo --env_id="go2-flat-reach-v2" --num_envs=4096 --update_epochs=8 --num_minibatches=32 --total_timesteps=204_800_000 --num_steps=200 --num_eval_steps=200 --gamma=0.99 --gae_lambda=0.95 --control_mode="pd_joint_delta_pos" --save_model --track

I'm also observing similar behavior on 4090. reducing num_eval_steps to 100 seems to somehow solve the problem but that just doesn't make sense to me (only 8 evaluation envs by default). Any help is appreciated.

kwang-12, 2025-05-16:
Also worth mentioning that I have to reboot the pc once this error happens. Starting a new sim after the error will just result in simulation freezing with the script stuck at the initial evaluation.
nvidia-smi shows that the gpu is used by 100% but nothing actually happens:

Every 0.1s: nvidia-smi                                                       192-222-52-71: Fri May 16 05:03:34 2025

Fri May 16 05:03:34 2025
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 570.133.07             Driver Version: 570.133.07     CUDA Version: 12.8     |
|-----------------------------------------+------------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA H100 80GB HBM3          On  |   00000000:06:00.0 Off |                    0 |
| N/A   27C    P0            115W /  700W |    6205MiB /  81559MiB |    100%      Default |
|                                         |                        |             Disabled |
+-----------------------------------------+------------------------+----------------------+

+-----------------------------------------------------------------------------------------+
| Processes:                                                                              |
|  GPU   GI   CI              PID   Type   Process name                        GPU Memory |
|        ID   ID                                                               Usage      |
|=========================================================================================|
|    0   N/A  N/A           75814    C+G   python3                                6164MiB |
+-----------------------------------------------------------------------------------------+

console output:
```
Running training
obs shape: (61,), act shape: (16,)
####
args.num_iterations=250 args.num_envs=4096 args.num_eval_envs=8
args.minibatch_size=25600 args.batch_size=819200 args.update_epochs=8
####
Epoch: 1, global_step=0
Evaluating
^C^C^C
```

StoneT2000, 2025-05-16:
Using 100% of the GPU is normal and expected.

Do the other PPO baseline scripts work fine? Or is it just this task that is breaking.

kwang-12, 2025-05-17:
Thank you for your reply!
I meant to say that I observed the VRAM usage jumped to 100% during evaluation.
This training session had 4096 simplified go2 envs. num_steps=200. 
The main script is `examples/baselines/ppo/ppo.py`
And the occurrence seems to be quite random. I mean, my platform has 80GB of VRAM and normally the VRAM usage 

[image]

StoneT2000, 2025-05-17:
First let's verify if it's just an environment problem or GPU issue with eg bad drivers

Are you able to run any of the other PPO baselines? eg run the PickCube training script in baselines.

Let it run for a long time and let me know about how the memory behaves

kwang-12, 2025-05-19:
I did some testing using only the provided code from ManiSkill. 
Specifically, `ManiSkill/examples/baselines/ppo/ppo.py` and `AnymalC-Reach-v1`.

Running the baseline example with different numbers of training envs yielded same error behavior due to CUDA OOM errors.

Note that this error only happens if evaluation is enabled. Test runs completes without any problem if evaluation is disabled.

1. test run with 1024 training envs:
```
(base) root@5a9cae070a1f:/workspace#  cd /workspace ; /usr/bin/env /opt/conda/bin/python /root/.vscode-server/extensions/ms-python.debugpy-2025.8.0-linux-x64/bundled/libs/debugpy/adapter/../../debugpy/launcher 39365 -- /workspace/ManiSkill/examples/baselines/ppo/ppo.py --env_id AnymalC-Reach-v1 --seed 1 --num_envs 1024 --update_epochs 8 --num_minibatches 32 --gamma 0.99 --gae_lambda 0.95 --total_timesteps 50000000 --num-steps 16 --num-eval-steps 200 --num_eval_envs 16 --track 
2025-05-19 05:43:27,290 - mani_skill  - WARNING - No initial pose set for actor builder of goal, setting to default pose q=[1,0,0,0], p=[0,0,0]. Not setting reasonable initial poses may slow down simulation, see https://github.com/haosulab/ManiSkill/issues/421.
2025-05-19 05:43:28,441 - mani_skill  - WARNING - No initial pose set for actor builder of goal, setting to default pose q=[1,0,0,0], p=[0,0,0]. Not setting reasonable initial poses may slow down simulation, see https://github.com/haosulab/ManiSkill/issues/421.
Saving eval videos to runs/AnymalC-Reach-v1__ppo__1__1747633403/videos
Running training
2025-05-19 05:43:31,604 - mani_skill  - WARNING - No initial pose set for actor builder of goal, setting to default pose q=[1,0,0,0], p=[0,0,0]. Not setting reasonable initial poses may slow down simulation, see https://github.com/haosulab/ManiSkill/issues/421.
####
args.num_iterations=3051 args.num_envs=1024 args.num_eval_envs=16
args.minibatch_size=512 args.batch_size=16384 args.update_epochs=8
####
Epoch: 1, global_step=0
Evaluating
2025-05-19 05:43:32,022 - mani_skill  - WARNING - No initial pose set for actor builder of goal, setting to default pose q=[1,0,0,0], p=[0,0,0]. Not setting reasonable initial poses may slow down simulation, see https://github.com/haosulab/ManiSkill/issues/421.
2025-05-19 05:43:38,808 - mani_skill  - WARNING - No initial pose set for actor builder of goal, setting to default pose q=[1,0,0,0], p=[0,0,0]. Not setting reasonable initial poses may slow down simulation, see https://github.com/haosulab/ManiSkill/issues/421.
Evaluated 3200 steps resulting in 16 episodes
eval_success_once_mean=0.0
eval_fail_once_mean=0.0
eval_return_mean=64.66410827636719
eval_episode_len_mean=200.0
eval_reward_mean=0.32332056760787964
eval_success_at_end_mean=0.0
eval_fail_at_end_mean=0.0
model saved to runs/AnymalC-Reach-v1__ppo__1__1747633403/ckpt_1.pt
SPS: 1814
Epoch: 2, global_step=16384
SPS: 3184
Epoch: 3, global_step=32768
SPS: 4190
Epoch: 4, global_step=49152
SPS: 5087
Epoch: 5, global_step=65536
SPS: 5863
Epoch: 6, global_step=81920
SPS: 6405
Epoch: 7, global_step=98304
SPS: 6833
Epoch: 8, global_step=114688
SPS: 7220
Epoch: 9, global_step=131072
SPS: 7529
Epoch: 10, global_step=147456
SPS: 7815
Epoch: 11, global_step=163840
SPS: 8053
Epoch: 12, global_step=180224
SPS: 8277
Epoch: 13, global_step=196608
SPS: 8486
Epoch: 14, global_step=212992
SPS: 8656
Epoch: 15, global_step=229376
SPS: 8791
Epoch: 16, global_step=245760
SPS: 8920
Epoch: 17, global_step=262144
SPS: 9037
Epoch: 18, global_step=278528
SPS: 9151
Epoch: 19, global_step=294912
SPS: 9222
Epoch: 20, global_step=311296
[... 244 lines cut ...]
  File "/workspace/ManiSkill/mani_skill/utils/structs/articulation.py", line 456, in get_net_contact_impulses
    self.px.gpu_query_contact_body_impulses(query)
RuntimeError: CUDA error: out of memory
```

Both cases were ran with vscode's python debugger so I was able to check VRAM usage when OOM error is triggered and the VRAM usage is not  anywhere near the capacity:
```
|  0%   45C    P5              15W / 200W |   5249MiB / 12282MiB |      7%      Default |

```

Another behavior I have noticed is that the OOM error happens at the same evaluation epoch step given the same training args.

This is really puzzling since the test runs only have 16 test envs so I don't think they should cause any oom issue. 

Here's a screenshot in vscode when OOM is triggered:

[image]

My driver versions are:
```
Driver Version: 535.230.02   CUDA Version: 12.2
```

For now, I am going to run with evaluation disabled I guess.

StoneT2000, 2025-05-20:
I wonder if this is an issue of using the contact force APIs and reconfiguration. I recall that the AnymalC-Reach-v1 environment would have this particular crash when `query.cuda_impulses.torch()` is called if the environment is reconfigured (which we only do by default for eval envs, originally because we want to ensure everything is randomized including object geometries if the task does randomize it).

Could you try turning reconfiguration frequency to 0 and try again?

kwang-12, 2025-05-23:
Can confirm that modifying reconfiguration frequency to 0 makes the problem go away.
Is this a problem on the SAPIEN's side?

StoneT2000, 2025-05-27:
It might be. Maybe some issue with memory not being reallocated for contact buffers once deleting the env, unsure.

For now just avoid reconfiguration (you probably don't need it anyway for the most part)

jpizarrom, 2025-06-25:
Hi @StoneT2000,

I had the same OOM issue following https://github.com/StoneT2000/lerobot-sim2real, and also the OOM was solved with `--ppo.eval_reconfiguration_freq=0`

NVIDIA-SMI 550.67
Driver Version: 550.67
CUDA Version: 12.4

jpizarrom, 2025-06-27:
in _after_reconfigure when rgb_overlay_mode="none", the self._objects_to_remove_from_greenscreen is not being empty, so all the refs are kept in the array

StoneT2000, 2025-07-15:
Some progress on debugging from my side so far:

I tried running the AnymalC-Reach-v1 example and it seems that after multiple reconfigures I will get a illegal memory access error when trying to get the impulses/forces of the "base" link specifically. Several other links seem fine like the "RH_FOOT" one.