mani-skill/ManiSkill #1081: Why qpos can exceed joint limits and seem very large?

fffffq00, 2025-05-20:
Hi, I have found that sometimes the robot joint values exceed the joint limits during simulation, which can even be very large.
The second finger in this picture looks really strange.
[image]

This is my test code (mani_skill version: 3.0.0b20):
```Python
import torch
from mani_skill.agents.robots import AllegroHandRight
from mani_skill.utils import common
from mani_skill.utils.structs.pose import Pose

if __name__ == "__main__":

    import gymnasium as gym
    num_envs = 2
    parallel_in_single_scene = True

    env = gym.make(
        "Empty-v1",
        robot_uids="allegro_hand_right",
        num_envs=num_envs,
        parallel_in_single_scene=parallel_in_single_scene if num_envs > 1 else False,
        control_mode="pd_joint_pos",
        render_mode="human",
        enhanced_determinism=True,
        obs_mode="state",
    )

    agent: AllegroHandRight = env.unwrapped.agent
    robot = agent.robot

    robot.set_root_pose(Pose.create_from_pq(p=[0., 0., 2.0]))
    obs, _ = env.reset(seed=0)
    limits = torch.stack([joint.get_limits() for joint in agent.controller.joints], dim=1)
    while True:
        action = env.action_space.sample()
        action = common.to_tensor(action, env.device)
        #print("action:")
        #print((action >= limits[..., 0]) & (action <= limits[..., 1]))
        env.step(action)
        qpos = robot.get_qpos()
        print("qpos:")
        exceed_low = qpos < limits[..., 0]
        exceed_high = qpos > limits[..., 1]
        print("exceed_low:", qpos[exceed_low] - limits[..., 0][exceed_low])
        print("exceed_high:", qpos[exceed_high] - limits[..., 0][exceed_high])
        env.render()
```

And this is the output:
```Python
qpos:
exceed_low: tensor([ -0.2773,  -2.6457, -23.7976,  -0.1709,  -0.2147,  -6.0707,  -3.4652,
         -0.3866], device='cuda:0')
exceed_high: tensor([ 1.2655, 10.6572,  4.8132,  1.8735,  1.6010, 20.9730], device='cuda:0')
qpos:
exceed_low: tensor([-4.3213e-06, -8.3966e-02, -5.9668e-02, -2.5599e+01, -1.5123e+01,
        -3.3073e+00, -1.0605e-01], device='cuda:0')
exceed_high: tensor([ 1.1589,  8.0738,  5.0748,  1.1114,  9.8585, 16.2163], device='cuda:0')
qpos:
exceed_low: tensor([ -0.3783, -28.1994,  -0.6031,  -0.0288,  -0.0475, -11.5003,  -3.4101,
         -0.2659], device='cuda:0')
exceed_high: tensor([ 0.9566,  4.3562,  4.8136,  7.4705, 12.6272], device='cuda:0')
qpos:
exceed_low: tensor([-4.8248e-01, -3.5644e+01, -1.0257e+01, -1.8634e-02, -3.7698e+00,
        -3.3790e+00, -4.2397e-01], device='cuda:0')
exceed_high: tensor([1.0449, 5.5153, 6.9922, 0.9496, 8.4656], device='cuda:0')
qpos:
exceed_low: tensor([-1.6359e+01, -3.4741e+01, -9.3036e+00, -1.5985e-02, -4.0418e+00,
        -3.6810e+00], device='cuda:0')
exceed_high: tensor([7.2369, 7.6276], device='cuda:0')
qpos:
exceed_low: tensor([-3.2416e-02, -4.4205e+00, -2.6805e+01, -3.5313e+01, -8.7894e+00,
        -1.6505e-02, -4.6995e+00, -3.7462e+00, -3.5167e+00, -1.0328e-01],
       device='cuda:0')
exceed_high: tensor([8.7536, 4.6152, 4.0593, 1.2739], device='cuda:0')
qpos:
exceed_low: tensor([ -0.5419, -21.8486, -45.5664,  -4.4191,  -9.3226,  -3.3544],
       device='cuda:0')
exceed_high: tensor([3.8198, 2.4153, 8.0610], device='cuda:0')
qpos:
exceed_low: tensor([ -0.3076,  -0.0639, -17.8434, -47.3704,  -3.7227,  -0.1102,  -6.2122,
         -3.8880,  -0.0594], device='cuda:0')
exceed_high: tensor([1.5535, 1.8391, 5.4043], device='cuda:0')
qpos:
exceed_low: tensor([-9.3836e-02, -8.9055e-02, -1.5817e+01, -4.7934e+01, -1.0282e+01,
        -4.0286e-02, -1.7495e-01, -1.6422e+01, -3.2812e+00], device='cuda:0')
exceed_high: tensor([1.5276, 1.3979, 5.2389], device='cuda:0')
qpos:
exceed_low: tensor([-15.8948, -47.4670, -10.4357,  -0.1557, -31.0846,  -3.6346],
       device='cuda:0')
exceed_high: tensor([4.7855, 1.1833], device='cuda:0')
qpos:
exceed_low: tensor([-15.8635, -44.8271, -11.2137,  -0.3750,  -0.1421, -34.7589,  -1.1776,
         -0.1039], device='cuda:0')
exceed_high: tensor([21.3528], device='cuda:0')
```

StoneT2000, 2025-05-20:
You might be taking very large and unrealistic actions which is causing simulation to break. I see that you are using the absolute action space and taking random actions, which can lead to this happening for more complex robots.

fffffq00, 2025-05-21:
Thanks for your reply. To avoid unrealistic actions, I now use `pd_joint_delta_pos` mode and still take random actions to watch the results. But I found this problem still exists.
By the way, when I specify `parallel_in_single_scene = False`, these situations will be different. 

This is my current code:
```Python
import torch
from mani_skill.agents.robots import AllegroHandRight
from mani_skill.utils import common
from mani_skill.utils.structs.pose import Pose

if __name__ == "__main__":

    import gymnasium as gym
    num_envs = 2
    parallel_in_single_scene = False
    env = gym.make(
        "Empty-v1",
        robot_uids="allegro_hand_right",
        num_envs=num_envs,
        parallel_in_single_scene=parallel_in_single_scene if num_envs > 1 else False,
        control_mode="pd_joint_delta_pos",
        render_mode="human",
        enhanced_determinism=False,
        obs_mode="state",
        sim_backend="physx_cuda",
    )

    agent: AllegroHandRight = env.unwrapped.agent
    robot = agent.robot

    robot.set_root_pose(Pose.create_from_pq(p=[0., 0., 2.0]))
    obs, _ = env.reset(seed=0)
    limits = torch.stack([joint.get_limits() for joint in agent.controller.joints], dim=1)
    i = 0
    while True:
        i+=1
        action = env.action_space.sample()
        action = common.to_tensor(action, env.device)
        qpos1 = robot.get_qpos()
        env.step(action)
        qpos = robot.get_qpos()
        print(i, (qpos- qpos1).norm())
        exceed_low = qpos < limits[..., 0]
        exceed_high = qpos > limits[..., 1]
        print("exceed_low:", (qpos[exceed_low] - limits[..., 0][exceed_low]).norm())
        print("exceed_high:", (qpos[exceed_high] - limits[..., 1][exceed_high]).norm())
        env.render()
```

Set `parallel_in_single_scene = False`, as the step length increases, it suddenly starts to become abnormal:
```Python
...
exceed_low: tensor(0.1216, device='cuda:0')
exceed_high: tensor(1.5871, device='cuda:0')
878 tensor(0.3458, device='cuda:0')
exceed_low: tensor(0.2209, device='cuda:0')
exceed_high: tensor(1.6108, device='cuda:0')
879 tensor(0.2817, device='cuda:0')
exceed_low: tensor(0.3084, device='cuda:0')
exceed_high: tensor(1.5337, device='cuda:0')
880 tensor(0.2693, device='cuda:0')
exceed_low: tensor(0.3085, device='cuda:0')
exceed_high: tensor(1.5149, device='cuda:0')
881 tensor(0.2403, device='cuda:0')
exceed_low: tensor(0.2312, device='cuda:0')
exceed_high: tensor(1.5253, device='cuda:0')
882 tensor(0.2489, device='cuda:0')
exceed_low: tensor(0.1599, device='cuda:0')
exceed_high: tensor(1.5177, device='cuda:0')
883 tensor(0.3338, device='cuda:0')
[... 56 lines cut ...]
902 tensor(4.7827, device='cuda:0')
exceed_low: tensor(32.9519, device='cuda:0')
exceed_high: tensor(1.3713, device='cuda:0')
903 tensor(5.2762, device='cuda:0')
exceed_low: tensor(36.1172, device='cuda:0')
exceed_high: tensor(1.2924, device='cuda:0')
904 tensor(3.7248, device='cuda:0')
exceed_low: tensor(33.2415, device='cuda:0')
exceed_high: tensor(1.1580, device='cuda:0')
905 tensor(4.3478, device='cuda:0')
exceed_low: tensor(33.5969, device='cuda:0')
exceed_high: tensor(1.3048, device='cuda:0')
906 tensor(11.5348, device='cuda:0')
exceed_low: tensor(25.6275, device='cuda:0')
exceed_high: tensor(1.5219, device='cuda:0')
...
```
I uploaded the video clips of the beginning and after a while to make it easier to observe this:

https://github.com/user-attachments/assets/5dccff8c-618e-46fc-95a8-115d4ca58e3c

https://github.com/user-attachments/assets/040d1ffb-7a14-4927-9072-48f75b93ea77

Is this just because I took random actions? 
Is there any parameter that can be set to avoid this? I am worried that this situation will affect the training of the agent.

fffffq00, 2025-07-05:
Hi, don't worry. 
After using maniskill for a few days, I just found a bug in the above code, 
```python
limits = torch.stack([joint.get_limits() for joint in agent.robot.joints], dim=1)
```
This line is wrong because the controller joints and robot joints are in the wrong order (I'm sorry for confusing the two).

Then I updated the code to test:
<details>
<summary>Test Code</summary>

```python
import torch
from mani_skill.agents.robots import AllegroHandRight
from mani_skill.utils import common
from mani_skill.utils.structs.pose import Pose
AllegroHandRight.disable_self_collisions = True
if __name__ == "__main__":

    import gymnasium as gym

    num_envs = 8
    parallel_in_single_scene = False
    env = gym.make(
        "Empty-v1",
        robot_uids="allegro_hand_right",
        num_envs=num_envs,
        parallel_in_single_scene=parallel_in_single_scene if num_envs > 1 else False,
        control_mode="pd_joint_delta_pos",
        render_mode="human",
        enhanced_determinism=False,
        obs_mode="state",
        sim_backend="physx_cuda",
    )

    agent: AllegroHandRight = env.unwrapped.agent
    robot = agent.robot
    robot.set_root_pose(Pose.create_from_pq(p=[0.0, 0.0, 2.0]))
    obs, _ = env.reset(seed=0)
    limits = robot.get_qlimits()
    i = 0
    while True:
        i += 1
        action = env.action_space.sample()
        action = common.to_tensor(action, env.device)
        qpos1 = robot.get_qpos()
        env.step(action)
        qpos = robot.get_qpos()
        print(i, (qpos - qpos1).norm())
        exceed_low = qpos < limits[..., 0]
        exceed_high = qpos > limits[..., 1]
        print("exceed_low:", (qpos[exceed_low] - limits[..., 0][exceed_low]).norm(p=2).rad2deg())
        print("exceed_high:", (qpos[exceed_high] - limits[..., 1][exceed_high]).norm(p=2).rad2deg())
        env.render()
```
</details>

You are right, after I set `disable_self_collisions=True`, the problem disappeared.

There is a setting that can stabilize the dexterous hand.
Looking back at the [video](https://github.com/user-attachments/assets/040d1ffb-7a14-4927-9072-48f75b93ea77) I uploaded above, I guessed that the instability came from the excessive rebound force, so I checked the properties of these links and found that `max_contact_impulse` and `max_depenetration_velocity` were both `1.0000000331813535e+32`.  This means that the engine tries to correct the penetration with a huge force in one step, and the link pops out of the penetration at a very high speed, which causes a link to collide at one moment, and then collide with a link in almost the opposite direction at the next moment, resulting in the above situation.
So, I set these two properties like this (just for testing):
```python
...
robot = agent.robot
for link in robot.links_map.values():
    for b in link._bodies:
        b.set_max_contact_impulse(1000 * env.scene.timestep)  #assume that the maximum force that the link can withstand is 1e4N
        b.set_max_depenetration_velocity(0.02 / env.scene.timestep)  # 0.02, based on the default contact offset.
...
```
It will be much more stable now. But I don't know if this setting will have some negative impact, so it needs to be discussed.

StoneT2000, 2025-07-05:
Nice trick to stabilize the hand, although it is more of a hack since the root cause is a self collision somewhere

The change to max depenetration is fine if that self collision is impossible to avoid due to the design of the hand and are bound to happen occasionally in sim and real.

Can you figure out which links are self colliding? If they do self collide you can disable collisions between those links.