mani-skill/ManiSkill #1082: PDEEPosController (delta=True) with actions all zero will still move robot arm.

dddab, 2025-05-20:
I initialize the arm with some qpos and qvel all zero. When I pass an action of all zeros deltas, the robot arm still moves.

I confirmed the target pos is the same as the given pos. Why would this happen?

dddab, 2025-05-20:
Debug print
```
after reset self.agent.robot.qpos tensor([[ 0.0000,  0.3927,  0.0000, -1.9635,  0.0000,  2.3562,  0.0400,  0.0400]])
after reset self.agent.robot.qvel tensor([[0., 0., 0., 0., 0., 0., 0., 0.]])
action [ 0.  0. -0. -0.]
%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%% _step_action [ 0.  0. -0. -0.]
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!! set_action True
self.agent.set_action(action) with tensor([[0., 0., -0., -0.]])
!!!!!!!!!!!!!! self.articulation.qpos tensor([[ 0.0000,  0.3927,  0.0000, -1.9635,  0.0000,  2.3562,  0.0400,  0.0400]])
@@@@@@@@@@@@@@@@@@@@@@@ setting PDEEPosController action tensor([[0., 0., -0.]])
@@@@@@@@@@@@@@@@@@@@@@@ setting PDEEPosController action after preprocess tensor([[0., 0., 0.]])
@@@@@@@@@@@@@@@@@@@@@@@ setting PDEEPosController start qpos tensor([[ 0.0000,  0.3927,  0.0000, -1.9635,  0.0000,  2.3562]])
self.config.use_target False
@@@@@@@@@@@@@@@@@@@@@@@ setting PDEEPosController prev_ee_pose_at_base Pose(raw_pose=tensor([[-1.4338e-01,  1.0568e-08, -2.5826e-03,  1.4645e-01,  3.5355e-01,
          3.5355e-01, -8.5355e-01]]))
@@@@@@@@@@@@@@@@@@@@@@@ setting PDEEPosController target pose Pose(raw_pose=tensor([[-1.4338e-01,  1.0568e-08, -2.5826e-03,  1.4645e-01,  3.5355e-01,
          3.5355e-01, -8.5355e-01]]))
@@@@@@@@@@@@@@@@@@@@@@@ setting PDEEPosController target qpos tensor([[ 0.0000,  0.3927,  0.0000, -1.9635,  0.0000,  2.3562]])
!!!!!!!!!!!!!! self.articulation.qpos tensor([[ 0.0000,  0.3927,  0.0000, -1.9635,  0.0000,  2.3562,  0.0400,  0.0400]])
!!!!!!!!!!!!!!!!!!!!!!!! self._sim_steps_per_control 5
before step self.agent.robot.qpos tensor([[ 0.0000,  0.3927,  0.0000, -1.9635,  0.0000,  2.3562,  0.0400,  0.0400]])
before step self.agent.robot.qvel tensor([[0., 0., 0., 0., 0., 0., 0., 0.]])
after step self.agent.robot.qpos tensor([[ 0.0235,  0.9328,  0.0070, -1.8872, -0.4545,  3.0711,  0.0440,  0.0440]])
after step self.agent.robot.qvel tensor([[ -42.0577,   62.4869,  -31.7927,   68.3693,   80.2184, -107.9720,
           38.4164,    8.5076]])
before step self.agent.robot.qpos tensor([[ 0.0235,  0.9328,  0.0070, -1.8872, -0.4545,  3.0711,  0.0440,  0.0440]])
before step self.agent.robot.qvel tensor([[ -42.0577,   62.4869,  -31.7927,   68.3693,   80.2184, -107.9720,
           38.4164,    8.5076]])
after step self.agent.robot.qpos tensor([[ 2.3749, -2.3574,  1.4868, -1.5179,  0.9743,  2.6374,  0.0402,  0.0440]])
after step self.agent.robot.qvel tensor([[603.1045,  92.7904,   7.9487, 132.7648, 249.5032,  70.8810, -96.0444,
          28.1165]])
before step self.agent.robot.qpos tensor([[ 2.3749, -2.3574,  1.4868, -1.5179,  0.9743,  2.6374,  0.0402,  0.0440]])
before step self.agent.robot.qvel tensor([[603.1045,  92.7904,   7.9487, 132.7648, 249.5032,  70.8810, -96.0444,
          28.1165]])
after step self.agent.robot.qpos tensor([[ 2.2825, -2.7344,  1.2016, -1.2576,  0.9352,  2.6324,  0.0440,  0.0440]])
after step self.agent.robot.qvel tensor([[-16.5582, -77.6817, -75.2619, 102.0969,  -0.2842, -55.2259,   2.1075,
          10.4559]])
before step self.agent.robot.qpos tensor([[ 2.2825, -2.7344,  1.2016, -1.2576,  0.9352,  2.6324,  0.0440,  0.0440]])
before step self.agent.robot.qvel tensor([[-16.5582, -77.6817, -75.2619, 102.0969,  -0.2842, -55.2259,   2.1075,
          10.4559]])
after step self.agent.robot.qpos tensor([[ 1.7155, -2.7909,  1.3667, -1.4818,  0.8817,  2.6087,  0.0439,  0.0440]])
after step self.agent.robot.qvel tensor([[ -3.7859, -34.1630,  12.7031, -73.3687, -49.9843,  62.4783, -10.4484,
          16.6678]])
before step self.agent.robot.qpos tensor([[ 1.7155, -2.7909,  1.3667, -1.4818,  0.8817,  2.6087,  0.0439,  0.0440]])
before step self.agent.robot.qvel tensor([[ -3.7859, -34.1630,  12.7031, -73.3687, -49.9843,  62.4783, -10.4484,
          16.6678]])
after step self.agent.robot.qpos tensor([[ 1.8675, -2.0635,  1.9968, -1.5725,  0.8108,  2.5855,  0.0440,  0.0440]])
after step self.agent.robot.qvel tensor([[ 48.2026,  59.0796,  45.6989,  18.5862,  -4.0287, -17.5475,   2.9200,
           1.3152]])
```

dddab, 2025-05-20:
I also tried with panda_wristcam as that is an arm included in the repo:

```after reset self.agent.robot.qpos tensor([[ 0.0000,  0.3927,  0.0000, -1.9635,  0.0000,  2.3562,  0.7854,  0.0400,
          0.0400]])
after reset self.agent.robot.qvel tensor([[0., 0., 0., 0., 0., 0., 0., 0., 0.]])
action [-0. -0. -0. -0.]
%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%% _step_action [-0. -0. -0. -0.]
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!! set_action True
self.agent.set_action(action) with tensor([[-0., -0., -0., -0.]])
!!!!!!!!!!!!!!!!!!!!!!!! self._sim_steps_per_control 5
before step self.agent.robot.qpos tensor([[ 0.0000,  0.3927,  0.0000, -1.9635,  0.0000,  2.3562,  0.7854,  0.0400,
          0.0400]])
before step self.agent.robot.qvel tensor([[0., 0., 0., 0., 0., 0., 0., 0., 0.]])
after step self.agent.robot.qpos tensor([[ 0.5896,  0.5047, -0.6589, -1.9373,  1.3598,  2.3222,  0.0280,  0.0400,
          0.0400]])
after step self.agent.robot.qvel tensor([[  95.1430,   17.2247,  -87.7723,  -25.7976, -130.2239,   15.3620,
          -90.7096,   -0.7482,   24.1257]])
before step self.agent.robot.qpos tensor([[ 0.5896,  0.5047, -0.6589, -1.9373,  1.3598,  2.3222,  0.0280,  0.0400,
          0.0400]])
before step self.agent.robot.qvel tensor([[  95.1430,   17.2247,  -87.7723,  -25.7976, -130.2239,   15.3620,
          -90.7096,   -0.7482,   24.1257]])
after step self.agent.robot.qpos tensor([[ 0.9960,  1.5641, -1.7869, -1.8596,  0.1968,  3.9369,  0.1144,  0.0000,
          0.0400]])
after step self.agent.robot.qvel tensor([[ -48.7507,   66.8063,   32.0744,    1.9052, -739.7488,  397.5370,
          618.1490,  -18.5780,  135.1873]])
before step self.agent.robot.qpos tensor([[ 0.9960,  1.5641, -1.7869, -1.8596,  0.1968,  3.9369,  0.1144,  0.0000,
          0.0400]])
before step self.agent.robot.qvel tensor([[ -48.7507,   66.8063,   32.0744,    1.9052, -739.7488,  397.5370,
          618.1490,  -18.5780,  135.1873]])
after step self.agent.robot.qpos tensor([[ 1.0600,  1.7664, -2.0863, -1.6509, -1.1640,  3.7370,  3.1982,  0.0400,
          0.0400]])
after step self.agent.robot.qvel tensor([[-19.9358,   2.3098, -14.6727,  53.2552, -87.9410, -39.9553, 229.9137,
          20.7057,  10.0000]])
before step self.agent.robot.qpos tensor([[ 1.0600,  1.7664, -2.0863, -1.6509, -1.1640,  3.7370,  3.1982,  0.0400,
          0.0400]])
before step self.agent.robot.qvel tensor([[-19.9358,   2.3098, -14.6727,  53.2552, -87.9410, -39.9553, 229.9137,
          20.7057,  10.0000]])
after step self.agent.robot.qpos tensor([[ 1.0320,  1.7627, -2.0946, -1.5821, -1.1790,  3.6053,  2.8971,  0.0400,
          0.0400]])
after step self.agent.robot.qvel tensor([[ -3.2957,  -0.4197,  -0.8255,   8.2171,  -4.4069, -12.0280,  -0.0310,
           1.3138,   0.9084]])
before step self.agent.robot.qpos tensor([[ 1.0320,  1.7627, -2.0946, -1.5821, -1.1790,  3.6053,  2.8971,  0.0400,
          0.0400]])
before step self.agent.robot.qvel tensor([[ -3.2957,  -0.4197,  -0.8255,   8.2171,  -4.4069, -12.0280,  -0.0310,
           1.3138,   0.9084]])
after step self.agent.robot.qpos tensor([[ 1.0291,  1.7629, -2.0963, -1.5702, -1.0911,  3.4863,  2.6969,  0.0383,
          0.0382]])
after step self.agent.robot.qvel tensor([[  0.0621,   0.0843,  -0.4180,   0.5661,  10.1358, -11.5016, -19.0645,
          -0.2531,  -0.1125]])
obs tensor([[  1.0291,   1.7629,  -2.0963,  -1.5702,  -1.0911,   3.4863,   2.6969,
           0.0383,   0.0382,   0.0621,   0.0843,  -0.4180,   0.5661,  10.1358,
         -11.5016, -19.0645,  -0.2531,  -0.1125]])
```

dddab, 2025-05-20:
My environment has some objects in it. When then environment is completely empty, the arm still moves, but much less:
```
after reset self.agent.robot.qpos tensor([[ 0.0000,  0.3927,  0.0000, -1.9635,  0.0000,  2.3562,  0.7854,  0.0400,
          0.0400]])
after reset self.agent.robot.qvel tensor([[0., 0., 0., 0., 0., 0., 0., 0., 0.]])
action [ 0.  0. -0. -0.]
%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%% _step_action [ 0.  0. -0. -0.]
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!! set_action True
self.agent.set_action(action) with tensor([[0., 0., -0., -0.]])
!!!!!!!!!!!!!!!!!!!!!!!! self._sim_steps_per_control 5
before step self.agent.robot.qpos tensor([[ 0.0000,  0.3927,  0.0000, -1.9635,  0.0000,  2.3562,  0.7854,  0.0400,
          0.0400]])
before step self.agent.robot.qvel tensor([[0., 0., 0., 0., 0., 0., 0., 0., 0.]])
after step self.agent.robot.qpos tensor([[-3.0213e-08,  3.9270e-01,  3.8688e-08, -1.9635e+00,  1.1057e-07,
          2.3562e+00,  7.8540e-01,  3.7660e-02,  3.7660e-02]])
after step self.agent.robot.qvel tensor([[-2.5478e-07,  5.1627e-08,  2.0451e-07,  8.1232e-08,  1.8954e-07,
         -3.9556e-08,  1.3208e-07, -2.2698e-01, -2.2690e-01]])
before step self.agent.robot.qpos tensor([[-3.0213e-08,  3.9270e-01,  3.8688e-08, -1.9635e+00,  1.1057e-07,
          2.3562e+00,  7.8540e-01,  3.7660e-02,  3.7660e-02]])
before step self.agent.robot.qvel tensor([[-2.5478e-07,  5.1627e-08,  2.0451e-07,  8.1232e-08,  1.8954e-07,
         -3.9556e-08,  1.3208e-07, -2.2698e-01, -2.2690e-01]])
after step self.agent.robot.qpos tensor([[-2.6719e-08,  3.9270e-01,  3.6994e-08, -1.9635e+00,  1.0199e-07,
          2.3562e+00,  7.8540e-01,  3.5507e-02,  3.5507e-02]])
after step self.agent.robot.qvel tensor([[ 4.7099e-07,  4.3704e-09, -1.9066e-07,  1.8706e-08, -8.9014e-07,
          1.8902e-08, -2.7912e-08, -2.0537e-01, -2.0539e-01]])
before step self.agent.robot.qpos tensor([[-2.6719e-08,  3.9270e-01,  3.6994e-08, -1.9635e+00,  1.0199e-07,
          2.3562e+00,  7.8540e-01,  3.5507e-02,  3.5507e-02]])
before step self.agent.robot.qvel tensor([[ 4.7099e-07,  4.3704e-09, -1.9066e-07,  1.8706e-08, -8.9014e-07,
          1.8902e-08, -2.7912e-08, -2.0537e-01, -2.0539e-01]])
after step self.agent.robot.qpos tensor([[-2.0438e-08,  3.9270e-01,  3.1606e-08, -1.9635e+00,  8.4424e-08,
          2.3562e+00,  7.8540e-01,  3.3559e-02,  3.3559e-02]])
after step self.agent.robot.qvel tensor([[ 4.3840e-07, -4.5607e-09, -1.8071e-07,  1.6104e-09, -8.4214e-07,
          8.1495e-09, -1.4119e-08, -1.8586e-01, -1.8589e-01]])
before step self.agent.robot.qpos tensor([[-2.0438e-08,  3.9270e-01,  3.1606e-08, -1.9635e+00,  8.4424e-08,
          2.3562e+00,  7.8540e-01,  3.3559e-02,  3.3559e-02]])
before step self.agent.robot.qvel tensor([[ 4.3840e-07, -4.5607e-09, -1.8071e-07,  1.6104e-09, -8.4214e-07,
          8.1495e-09, -1.4119e-08, -1.8586e-01, -1.8589e-01]])
after step self.agent.robot.qpos tensor([[-1.4554e-08,  3.9270e-01,  2.6029e-08, -1.9635e+00,  6.7033e-08,
          2.3562e+00,  7.8540e-01,  3.1796e-02,  3.1796e-02]])
after step self.agent.robot.qvel tensor([[ 3.5838e-07, -4.7037e-09, -1.5672e-07, -4.3547e-09, -6.8865e-07,
          9.1505e-09, -1.8840e-08, -1.6820e-01, -1.6823e-01]])
before step self.agent.robot.qpos tensor([[-1.4554e-08,  3.9270e-01,  2.6029e-08, -1.9635e+00,  6.7033e-08,
          2.3562e+00,  7.8540e-01,  3.1796e-02,  3.1796e-02]])
before step self.agent.robot.qvel tensor([[ 3.5838e-07, -4.7037e-09, -1.5672e-07, -4.3547e-09, -6.8865e-07,
          9.1505e-09, -1.8840e-08, -1.6820e-01, -1.6823e-01]])
after step self.agent.robot.qpos tensor([[-9.6767e-09,  3.9270e-01,  2.1002e-08, -1.9635e+00,  5.1865e-08,
          2.3562e+00,  7.8540e-01,  3.0200e-02,  3.0200e-02]])
after step self.agent.robot.qvel tensor([[ 2.5821e-07, -3.4201e-09, -1.0897e-07, -9.1235e-09, -4.7689e-07,
          1.5636e-08, -3.1701e-08, -1.5222e-01, -1.5225e-01]])
obs tensor([[-9.6767e-09,  3.9270e-01,  2.1002e-08, -1.9635e+00,  5.1865e-08,
          2.3562e+00,  7.8540e-01,  3.0200e-02,  3.0200e-02,  2.5821e-07,
         -3.4201e-09, -1.0897e-07, -9.1235e-09, -4.7689e-07,  1.5636e-08,
         -3.1701e-08, -1.5222e-01, -1.5225e-01]])

StoneT2000, 2025-05-20:
Is there collision occuring? That would cause the qvel to be nonzero which is shown in the first code block

dddab, 2025-05-21:
It turns out the arm was in self collision. Setting disable_self_collisions = True resolved the issue.

closed 2025-05-21 by dddab