mani-skill/ManiSkill #1085: I wonder that how .USD file can use in maniskill. I don't find a method like create_usd_loader etc.

Ethan-Chen-plus, 2025-05-22:
As title, I wonder that how .USD file can use in maniskill. I don't find a method like create_usd_loader etc.

Thanks again.

StoneT2000, 2025-05-22:
We can support the USD file format, it just needs someone to do it. It's currently a more low priority task at the moment.

Ethan-Chen-plus, 2025-05-23:
Thanks a lot. I will try to make it these days. If success, I will make a pull request!

jayant-mil, 2025-05-27:
Honestly I would love for Maniskill to be the go-to robotics simulator. Isaac sim is incredibly frustrating.

Ethan-Chen-plus, 2025-05-28:
Hi @StoneT2000,
We can leverage the `OpenUSD` (often referred to as pxr in Python) repository for the fundamental task of loading and parsing USD files.
However, the more involved part seems to be mapping and assigning all the relevant object properties (like geometry, materials, physics attributes, etc.) into the SAPIEN/ManiSkill systems. I'm currently still digging into the API integration details.
It's great to hear that more people are requesting this! I'd definitely appreciate it if you have some time to look into it next week as well.
Data Test: `GenieSimAssets` dataset on Hugging Face (https://huggingface.co/datasets/agibot-world/GenieSimAssets) could be a really good candidate. It seems to have a nice collection of USD assets relevant to robotics scenes.

StoneT2000, 2025-07-15:
Currently guessing my way around the mjcf2usd package (https://github.com/LightwheelAI/mjcf2usd) and the isaac sim urdf exporter extension (https://github.com/isaac-sim/IsaacSim/blob/main/source/extensions/isaacsim.asset.exporter.urdf/isaacsim/asset/exporter/urdf/exporter.py)

Takes a bit of time to figure out USD, but this is on my priority list though.

Zhouxunzhe, 2025-07-15:
> Currently guessing my way around the mjcf2usd package (https://github.com/LightwheelAI/mjcf2usd) and the isaac sim urdf exporter extension [...]

@StoneT2000 I have now tried to export .USD as .urdf by using the isaac sim urdf exporter extension. However, in the process of exportation, the textures will not be kept. Do you have any idea how to solve this issue? Or is this issue under your consideration?

StoneT2000, 2025-07-15:
I did not know that, that sounds problematic, Ill investigate

yilin404, 2025-09-24:
@StoneT2000 @Ethan-Chen-plus @Zhouxunzhe Do you have any updates? I’d greatly appreciate it if you could share them.