mani-skill/ManiSkill #1095: Code or sim2real transfer

yeshenpy, 2025-05-28:
Is there any code available for sim2real transfer? I read the ManiSkill3 paper and found that the implementation is based on PointNet + PPO, but I haven't been able to find the corresponding code so far.

I would greatly appreciate it if you could provide the relevant code.

StoneT2000, 2025-05-30:
> Thank you very much for your response. I noticed that the current version on arXiv uses point clouds. May I ask why it was switched to RGB? Was it because point clouds were less stable or reliable compared to RGB?

RGB is more accessible to users as not everyone has depth cameras. We targeted something more reproducible so we switched to using LeRobot and RGB inputs. Pointcloud is probably better but harder to set up.

yeshenpy, 2025-06-01:
Thank you very much for your reply. 
I’ve been working on point cloud-related tasks recently, but I found that adding new cameras in ManiSkill seems a bit tricky. I couldn’t find very detailed documentation. One key point is that I’d like to visualize the image from the added camera to make sure the orientation is set correctly. How can I achieve this? For example, is it possible to render multiple camera views in human mode?

StoneT2000, 2025-06-01:
Adding new cameras is fairly simple. See the tutorial for how to make new tasks in ManiSkill: https://maniskill.readthedocs.io/en/latest/user_guide/tutorials/custom_tasks/intro.html#optional-setting-up-cameras-sensors-for-observations-and-recording

It also tells you how to do see how your new camera is positioned by rendering with rgb_array or sensors render mode.

Either create your own environment following the tutorial, or create a new python class that inherits an existing environment class and override whatever you want to override.

closed 2025-06-02 by yeshenpy

StoneT2000, 2025-06-14:
By the way @yeshenpy the code for sim2real transfer demonstrated in the latest maniskill3 paper is open sourced here: https://github.com/StoneT2000/lerobot-sim2real

yeshenpy, 2025-06-14:
Great job! I'll give it a try right away.