mani-skill/ManiSkill #1097: the cpu memory increase slowly when running env step

LiuYiKai6, 2025-05-29:
When I run the PickCube task and just step for several times, the cpu memory increase slowly, about 0.2GB for 1M steps. It this normal?

```
import gymnasium as gym
import mani_skill.envs

env = gym.make(
    "PickCube-v1", # there are more tasks e.g. "PushCube-v1", "PegInsertionSide-v1", ...
    num_envs=63,
    obs_mode="state", # there is also "state_dict", "rgbd", ...
    control_mode="pd_ee_delta_pose", # there is also "pd_joint_delta_pos", ...
    render_mode="all"
)

obs, _ = env.reset(seed=0) # reset with a seed for determinism
for i in range(500000):
    action = env.action_space.sample()
    obs, reward, terminated, truncated, info = env.step(action)
    print(i*63)
```

StoneT2000, 2025-05-30:
What version of python and maniskill are you using?

And how are you measuring memory use, I don't see the code for that

StoneT2000, 2025-06-15:
We recently found that torch 2.6.0 can cause cpu memory increases. This should not have significant memory increases anymore beyond python pecularities.

closed 2025-06-15 by StoneT2000