mani-skill/ManiSkill #1106: How to aquire rgb and depth under obs_mode=pointcloud

yqi19, 2025-06-02:
Hi, thanks for the great work!
If I set 
```python
env = gym.make("table_scene",
               sensor_configs=dict(width=480, height=480),
               human_render_camera_configs=dict(shader_pack='rt'),
               obs_mode="pointcloud"
               )
```
How do I acquire the rgb and depth data from each of the camera in the env?

I can generate point cloud corresponds to each camera when obs_mode='rgb+depth+segmentation", however, how to get rgb and depth when obs_mode="pointcloud"?

Thanks!

yqi19, 2025-06-05:
Is the point cloud mode of obs return the full point cloud of the scene?

StoneT2000, 2025-06-05:
no, only what is observed by the cameras defined in the environment