mani-skill/ManiSkill #1110: Do you have plan to add visual tactile sensor?

yqi19, 2025-06-05:
Hi wonderful team of Maniskill,

May I ask do you have any plan to add visual tactile sensors in simulation like https://binghao-huang.github.io/3D-ViTac/ that we can acquire RGB-based tactile image from it? There is one existing simulation named TacSL (https://iakinola23.github.io/tacsl/) which is builed base on issac gym. I wonder if Maniskill plans to have such sensor in the recent future.

Thank you very much!

StoneT2000, 2025-06-05:
ManiSkill ViTAC is a specifically separate sim with different physics engine for doing visual tactile simulation work: https://ai-workshops.github.io/maniskill-vitac-challenge-2024/

Alternatively depending on what kind of visual tactile information you want, you could make a sensor of sorts that uses pairwise contact forces to generate a image maybe.

yqi19, 2025-06-05:
Thanks for the response! 👍