mani-skill/ManiSkill #1127: "No low value can be equal to `np.nan`" when using `pd_ee_pose`

jstmn, 2025-06-13:
Hi,

I'm getting an error whenever I create a gym env using `control_mode="pd_ee_pose"`. 

With the following code:
```
import gymnasium as gym
import mani_skill.envs
gym.make(
    "PickCube-v1",
    num_envs=1,
    control_mode="pd_ee_pose",
)
```
I'm getting this error:
```
jstm ~/Projects/mpcm2 uv run python scripts/temp_ee_pose_control.py 
Traceback (most recent call last):
  File "/home/jstm/Projects/mpcm2/scripts/temp_ee_pose_control.py", line 4, in <module>
    gym.make(
  File "/home/jstm/Projects/mpcm2/.venv/lib/python3.10/site-packages/gymnasium/envs/registration.py", line 742, in make
    env = env_creator(**env_spec_kwargs)
  File "/home/jstm/Projects/mpcm2/thirdparty/ManiSkill/mani_skill/utils/registration.py", line 182, in make
    env = env_spec.make(**kwargs)
  File "/home/jstm/Projects/mpcm2/thirdparty/ManiSkill/mani_skill/utils/registration.py", line 79, in make
    return self.cls(**_kwargs)
  File "/home/jstm/Projects/mpcm2/thirdparty/ManiSkill/mani_skill/envs/tasks/tabletop/pick_cube.py", line 65, in __init__
    super().__init__(*args, robot_uids=robot_uids, **kwargs)
  File "/home/jstm/Projects/mpcm2/thirdparty/ManiSkill/mani_skill/envs/sapien_env.py", line 312, in __init__
    obs, _ = self.reset(seed=[2022 + i for i in range(self.num_envs)], options=dict(reconfigure=True))
  File "/home/jstm/Projects/mpcm2/thirdparty/ManiSkill/mani_skill/envs/sapien_env.py", line 890, in reset
    self._reconfigure(options)
  File "/home/jstm/Projects/mpcm2/thirdparty/ManiSkill/mani_skill/envs/sapien_env.py", line 726, in _reconfigure
    self._load_agent(options)
  File "/home/jstm/Projects/mpcm2/thirdparty/ManiSkill/mani_skill/envs/tasks/tabletop/pick_cube.py", line 82, in _load_agent
    super()._load_agent(options, sapien.Pose(p=[-0.615, 0, 0]))
  File "/home/jstm/Projects/mpcm2/thirdparty/ManiSkill/mani_skill/envs/sapien_env.py", line 409, in _load_agent
    agent: BaseAgent = agent_cls(
  File "/home/jstm/Projects/mpcm2/thirdparty/ManiSkill/mani_skill/agents/base_agent.py", line 112, in __init__
    self.set_control_mode()
  File "/home/jstm/Projects/mpcm2/thirdparty/ManiSkill/mani_skill/agents/base_agent.py", line 261, in set_control_mode
    self.controllers[control_mode] = CombinedController(
  File "/home/jstm/Projects/mpcm2/thirdparty/ManiSkill/mani_skill/agents/controllers/base_controller.py", line 209, in __init__
    self.controllers[uid] = config.controller_cls(
  File "/home/jstm/Projects/mpcm2/thirdparty/ManiSkill/mani_skill/agents/controllers/base_controller.py", line 68, in __init__
    self._initialize_action_space()
  File "/home/jstm/Projects/mpcm2/thirdparty/ManiSkill/mani_skill/agents/controllers/pd_ee_pose.py", line 207, in _initialize_action_space
    self.single_action_space = spaces.Box(low, high, dtype=np.float32)
  File "/home/jstm/Projects/mpcm2/.venv/lib/python3.10/site-packages/gymnasium/spaces/box.py", line 147, in __init__
    self.low, self.bounded_below = self._cast_low(low, dtype_min)
  File "/home/jstm/Projects/mpcm2/.venv/lib/python3.10/site-packages/gymnasium/spaces/box.py", line 214, in _cast_low
    raise ValueError(f"No low value can be equal to `np.nan`, low={low}")
ValueError: No low value can be equal to `np.nan`, low=[nan nan nan nan nan nan]
```

Git version:
```
 jstm ~/Projects/mpcm2/thirdparty/ManiSkill git log | head
commit ee3903c99411ef643839aed8838581b127378417
Author: hu-po <8493427+hu-po@users.noreply.github.com>
Date:   Fri Jun 6 13:25:06 2025 -0500

    [Feature] Support the widowxai robot and provide RL baseline for cube picking, and update task docs (#904)
```

Any idea? Thanks

jstmn, 2025-06-13:
oh weird, I just installed a fresh clone and the issue isn't happening ...

```
 cd /tmp/
 git clone https://github.com/haosulab/ManiSkill.git
 python3.10 -mv venv venv
 python3.10 -m venv venv
 source venv/bin/activate
 pip install ManiSkill/
 python 
# running the gym.make() command here
```

StoneT2000, 2025-06-13:
Oh its because its gymnasium 1.1.1

closed 2025-06-13 by StoneT2000

jstmn, 2025-06-13:
Ahh I see, thanks.

prasuchit, 2025-06-18:
Hi guys, I'm interested in this feature. Just adding a comment here to keep track of the status. Hoping that this PR will be merged soon. Thanks.

prasuchit, 2025-06-18:
I mean the pd_ee_pose controller on GPU. I reached this issue from [here](https://github.com/haosulab/ManiSkill/issues/955).