mani-skill/ManiSkill #1131: Setting up Maniskill on A100 Server

Jaraxxus-Me, 2025-06-14:
Hi, I'm trying to run Maniskill on an A100 headless server, and worked around different solutions, but they all failed. I am wondering if you could provide some suggestions on this. I'm using Lambda instances.
System: A100-SXM4-40GB, Cuda 12.8

- I first tried to use the provided docker: `maniskill/base` (where the environment variables and `.json` files should be good). But it does not work, showing the following errors:
[image]

- I then tried to directly install Maniskill in a Python environment on the machine.
    1. I checked the solutions discussed [here](https://github.com/haosulab/SAPIEN/issues/115#issuecomment-1434899965).
       I added the three files to the correct locations and tried to set the environment variables, but after that `vulkaninfo` can't show the information, see below.

[image]

- Two things I'm unsure about:
    1. If I do not export the variable `VK_ICD_FILENAMES`, `vulkaninfo` does show information, and running the example raises different errors:

[image]

    2. `ldconfig -p | grep libGLX_nvidia` does not show anything on the clean environment.

Thanks a lot if you could provide some guidance/suggestions on this.

StoneT2000, 2025-06-14:
Lemme get back to you on debugging this later.

For a quick workaround, can you try other GPU instances on Lambda? I recommend 4090 as it is the best and fastest GPU to run simulation on.

Jaraxxus-Me, 2025-06-14:
Thanks a lot for the response. Sure, I can run experiments on 3090/4090 for now, they work well.

Jaraxxus-Me, 2025-06-19:
When you mention driver, which driver are you referring to? The nvidia cuda driver or the vulkan driver?
For docker image, I have one cuda 12.2 built here, which works on my local machine, but not A100 servers..
bowenli1024/skill_ref:base

StoneT2000, 2025-06-19:
Nvidia driver installed on the server. I believe before NVIDIA docker can be run, you need to set up the nvidia drivers correctly on the server. Try reinstalling that before running nvidia docker

Jaraxxus-Me, 2025-06-19:
For A100 GPU, is there any suggestions on the driver and CUDA version? I can try reinstalling

AlexandreBrown, 2025-06-19:
Hi @Jaraxxus-Me , I got the same error (ErrorIncompatibleDriver) when trying to run my image on a different cluster.  
I ended up fixing the issue and my image does not even install the GPU drivers, I can use the one from the host to avoid any conflict.   
This stackoverflow post suggests it's better to use the drivers from the host : https://stackoverflow.com/a/73052220  

The major thing I learned is that if your host (eg: Compute node on the cluster) has a different OS than your image, it might lead to Singularity/Docker incorrectly binding the Nvidia drivers.   

For instance I had the following setups :  
**Same OS Scenario**
Cluster A : Compute node has Ubuntu 22.04, my image uses `nvidia/cuda:12.4.1-cudnn-devel-ubuntu22.04`  
Result: Everything works fine   
**Different OS than Host Scenario**  
Cluster B : Compute node has Rocky Linux (significantly different than Ubuntu)  
Result: torch.cuda.is_available() returns true but gym.make would crash with Maniskill resulting in the same error as you (svulkan2 incompatible driver etc).  

The reason is that in my case, the compute node has the nvidia drivers installed under `/usr/lib64/` (eg: `/usr/lib64/libnvidia-...so`)  
But my container is Ubuntu 22.04 which uses a multiarch spec folder structure to store libraries (source: https://unix.stackexchange.com/a/43214 ) so it expects the libs to be under `/usr/lib/x86_64-linux-gnu/`, which is different than what we have on the host for this cluster (`/usr/lib64/` != `/usr/lib/x86_64-linux-gnu/`).  
This results in Singularity not binding the nvidia and vulkan drivers correctly since there is a mismatch. Manually binding `/usr/lib64/` to `/usr/lib/x86_64-linux-gnu/` does not work because this directory has other stuff than just nvidia and vulkan and in my case it would just break things.  

The fix in my case was to manually bind the appropriate lib files individually instead of the entire folder eg: `-B /usr/lib64/libEGL_nvidia.so.0:/usr/lib/x86_64-linux-gnu/libEGL_nvidia.so.0`.   
And I had to add (in the container at build time or runtime) `/usr/lib/x86_64-linux-gnu/` to my `LD_LIBRARY_PATH` so that vulkan/sapien/everyone can find the libs properly.  
Ex: `--env LD_LIBRARY_PATH=/usr/lib/x86_64-linux-gnu/:$LD_LIBRARY_PATH`   
I created a repo to demonstrate a setup that can work https://github.com/AlexandreBrown/Maniskill3-Singularity-Example/tree/main  without needing to install any extra driver in the docker/singularity image.  

I will polish the repo tomorrow so that it only has the strict minimum, right now the recipe has a lot of things that are being installed in the sudo apt install phase which might not be needed (I was just looking to get something that works first, then I'll trim it down).  
Most likely only a few simple dependencies like `libvulkan1` are needed now.  
I will also add proper documentation in the README explaining the issue in more details but thought I'd post this here in case it can help in the meantime.  
  
One final thing, if you manually create the files like the icd and json files, make sure the library path matches the bind you use when running your image because vulkan seems to look exactly at this path.  
In my case I do not manually create and place the json, I just use the one from the host.  
Note that in my case my host had `/usr/share/vulkan/icd.d/nvidia_icd.x86_64.json` which caused issues because vulkan would expect `/usr/share/vulkan/icd.d/nvidia_icd.json` so I had to add a bind as shown here https://github.com/AlexandreBrown/Maniskill3-Singularity-Example/blob/50e2ab4b85015f91d9fcd3d73e8f5f2b014194c3/drac_cluster_sbatch_script_example.sh#L70C10-L70C96

AlexandreBrown, 2025-06-20:
@Jaraxxus-Me No worries, FYI I cleaned up the repo and added proper in-depth documentation : https://github.com/AlexandreBrown/Maniskill3-singularity-example  
Feel free to give it a read and a try.  
The goal of the repo is to be very minimalistic while still showing real-world setup.  
It explain how to setup and run your Maniskill3 workload on clusters and how to handle the 2 cases (Host OS = Container OS and Host OS $\neq$ Container OS).   

I cleaned up the recipe file to be much more minimalistic (we only install a few packages in the sudo apt install now) and I added a small loop in my cluster script to automatically find and  bind the appropriate lib files in the case where the Host OS is not the same as the container OS.    

I added 2 examples, one for the [MILA cluster](https://docs.mila.quebec/) which shows a case where Host OS = Container OS and another example for the [DRAC clusters](https://alliancecan.ca/en/services/advanced-research-computing/national-services/clusters) representing the case where Host OS $\neq$ Container OS.  

The singularity image recipe also ensures that training can work if the cluster has no internet connection.  

Hope this helps,

Cheers!

AlexandreBrown, 2025-07-03:
Let us know @Jaraxxus-Me , FYI I tested my repo on 6 different clusters and the general recipe always worked (it's always a matter of figuring out if the cluster compute node OS = your container OS and depending on the compute node OS you need to check where the json files reside).  
Feel free to point out any issue you have to get Maniskill3 working on your cluster.

Jaraxxus-Me, 2025-07-06:
Hi Stone and Alexandre, thanks for checking back. I'm traveling last two weeks and will get back next week. I'll update here if the repo can resolve the issue or not. Thanks again!

closed 2025-07-07 by Jaraxxus-Me