mani-skill/ManiSkill #1134: Floating assets during GPU simulation with hidden objects

SoshinK, 2025-06-15:
Hello!

I have found interesting behavior of scene assets during GPU simulation. More specifically, when I use sapien viewer with `sim_backend='gpu'` enabled and append some kinematic objects (like `goal_region`) to `self._hidden_objects`, during idle rendering without any steps I can see that my **assets are slightly moving!** I can reproduce this behavior using YCB assets in `PushCubeEnv`. Here example which demonstrates this effect:

```
import gymnasium as gym
from tqdm import tqdm

import sapien
import mani_skill.envs
from mani_skill.envs.tasks.tabletop.push_cube import PushCubeEnv
from mani_skill.utils.building import actors
from mani_skill.utils.registration import register_env

@register_env("BuggyPush", max_episode_steps=50)
class BuggyPush(PushCubeEnv):
    def _load_scene(self, options: dict):
        super()._load_scene(options)
       
        # load ycb object
        model_id='002_master_chef_can'
        builder = actors.get_actor_builder(
            self.scene,
            id=f"ycb:{model_id}",
        )
        builder.set_initial_pose(sapien.Pose(p = [0.2, 0.2, 0.071]))
        self.can = builder.build(name=model_id)

        # hide goal_region
        self._hidden_objects.append(self.goal_region)

def main():
    env_kwargs = dict(
        render_mode='human',
        num_envs=1,
        sim_backend='gpu', # enable GPU sim!
        enable_shadow=True,
        viewer_camera_configs={'shader_pack': 'default'}, 
        human_render_camera_configs={'shader_pack': 'default'},
        parallel_in_single_scene=True,
    )
    env = gym.make(
        'BuggyPush',
        **env_kwargs
    )
    env.reset()
    episode_length = 10

    for i in tqdm(range(episode_length)):
        _, _, _, _, _ = env.step(None)
        
        env.render_human()

    # wait for exit [HERE YCB ASSET BEGIN TO FLOAT!!!]
    viewer = env.render_human()
    while True:
        if viewer.closed:
            exit()
        if viewer.window.key_down("c"):
            break
        env.render_human()

    env.close()

if __name__ == '__main__':
    main()
```

In the example above, the starting position of ycb object (metal can) after steps was:

[image]

After waiting some time, it became:

[image]

Once again, it can be observed with enabled GPU simulation, with custom assets and with at least one hidden object. On CPU or w/o hidden objects, everything works correctly (objects are still). In my cluttered environments, this becomes even worse and objects are actually flying with visible motion during GUI rendering. Although it seems not happen during steps, it is not very convenient in viewing scenes.

Any ideas why this happens? Thank You in advance.

StoneT2000, 2025-06-16:
Can you share a video of what happens with and without the hidden object?

SoshinK, 2025-06-17:
@StoneT2000 

There is with hidden object (speed x80): can moves during render

[image]

There is without hidden object (can stays still):

[image]

In my cluttered scenes with many objects this effect is more visible (original speed):

[image]

StoneT2000, 2025-07-15:
Apologies for the late reply. I just ran the code you shared earlier but I don't observe what you are seeing.

What version of maniskill are you using? And could you try `pip uninstall mani_skill && pip install mani-skill-nightly`?

Moreover my can is always at `Pose([0.198897, 0.199804, 0.0493853], [1.00105, -0.00384372, -0.00605212, 1.96799e-06])`, the z value is lower than yours. Are you possible using outdated assets?

Does this only happen with objects with meshes? Or does this also happen for primitive shapes?

StoneT2000, 2025-09-05:
Ok I can reproduce the can moving thing. It seems the position is changing very slightly over the course of thousands of steps.

This occurs because in a GPU sim you can't actually hide something by hiding the visuals, you have to teleport the object far away (which is what we do at the moment). It appears there is some small floating point imprecision issue when spawning the object back to its original place.

After some debugging I can't determine a easy solution to this other than for you to manually place the object back where it should be if you aren't performing simulation and are only running render_human / recording videos.

Another option is to use the CPU backend which if you are rendering only should not have this issue when hiding/showing objects.

StoneT2000, 2025-09-05:
Maybe @fbxiang you can take a look. For a GPU sim with even just 1 environment, constant calls to apply rigid dynamic and fetch rigid dynamic slowly causes object positions to drift.

fbxiang, 2025-09-05:
The apply and fetch calls have floating point arithmetic in it. Physx may also do some computation inside, so some drifting is expected.