mani-skill/ManiSkill #1150: Observations Turn Green After Reset in "PickSingleYCB-v1" and "PickClutterYCB-v1" Environments on macOS

cinemere, 2025-06-21:
I encountered an issue while testing the `PickSingleYCB-v1` and `PickClutterYCB-v1` environments locally on my Mac. Specifically, after resetting the environment for the second or third time, all observations turn green. This issue does not occur with the `PickCube-v1` environment, which works as expected.

#### Steps to Reproduce:
1. Install ManiSkill on macOS by following the installation guide in the official documentation. Test installation commands work fine.
2. Run the following script:

```python
import gymnasium as gym
import mani_skill.envs
import matplotlib.pyplot as plt

env_name = "PickCube-v1"
# env_name = "PickSingleYCB-v1"
# env_name = "PickClutterYCB-v1"

env = gym.make(env_name,
               num_envs=1, 
               render_mode="rgb_array")

for i in range(10):
    obs, _ = env.reset()
    image = env.render()
    plt.imshow(image[0]); plt.show()

env.close()
```

3. Uncomment and test with the following environments:
   - `"PickCube-v1"`: Works fine.
   - `"PickSingleYCB-v1"`: Observations turn green after 2 resets.
   - `"PickClutterYCB-v1"`: Observations turn green after 1-2 resets.

#### Example of *turning green* on PickSingleYCB-v1

[image]

#### Environment Details:
- **OS**: macOS
- **Python Version**: 3.11

Could you please investigate this issue and provide guidance on how to resolve it? If additional information or debugging steps are needed, feel free to ask.

Thank you for your help!

StoneT2000, 2025-06-24:
What mac is this, intel GPU or M1?

StoneT2000, 2025-06-28:
@comoco-xiao any idea potentially?