mani-skill/ManiSkill #1156: Asset Penetration & Policy Training Issue

yeshenpy, 2025-06-27:
I encountered an environment issue: I loaded a cabinet, but found that the RL-learned policy can directly penetrate the edges of the cabinet and push the door out from the inside to complete the cabinet-opening task. How can this issue be resolved? The cabinet asset was loaded from PartNet.

[image]

In addition, I found that during evaluation, if the task is completed but not terminated immediately, the robotic arm tends to swing violently or even twist unnaturally. Is this because the success condition is too simplistic? Currently, a reward of 10 is given as long as the cabinet door is opened.

https://github.com/user-attachments/assets/8e1fbcd8-a07b-440e-82d3-18459c109bc3

StoneT2000, 2025-06-28:
re the violent behavior yes it is likely because your reward function doesn't say anything about the robot having to be still after solving the task

For the "penetration" it is possible, do you have a picture of what the collision mesh looks like?  If i remember correctly a lot of cabinets do not have a collision face modelled on the top.

moreover is the cabinet inside the table?

yeshenpy, 2025-06-30:
Thank you for the reply. I have resolved the violent behavior.

This cabinet is not placed under the table; rather, it is a long and narrow cabinet, which makes it appear as if it is embedded underneath the table.

Is there any way to detect the collision mesh? I would like to extract it separately for visualization. I’ve noticed that the agent is still grasping through the mesh (i.e., penetrating the object). The gripper clearly inserts into the door in the video (if you slow it down enough). This makes it difficult for me to determine which aspects to adjust.

https://github.com/user-attachments/assets/d4c832a4-59f5-4e0b-8f78-9e48a10e5178

closed 2025-07-08 by StoneT2000