mani-skill/ManiSkill #1166: How to filter camera and state observations in demonstration datasets

branyang02, 2025-07-02:
Hi, I downloaded a demo dataset with 

```
python -m mani_skill.trajectory.replay_trajectory   --traj-path ~/.maniskill/demos/StackCube-v1/motionplanning/trajectory.h5   --use-first-env-state -c pd_ee_delta_pos -o rgb   --save-traj --num-envs 10 -b physx_cpu
```
However, I'm struggling to find the image and state observation keys in trajectory.h5.

StoneT2000, 2025-07-03:
What's the code you are using to load data? There should be an observation key and in that containing some keys for sensor data and more

branyang02, 2025-07-14:
Hi, sorry for the late reply, but I was trying to use a modified version of this:

```
from typing import Union
import h5py
import numpy as np
from torch.utils.data import Dataset
from tqdm import tqdm

from mani_skill.utils.io_utils import load_json
from mani_skill.utils import sapien_utils
from mani_skill.utils import common

# loads h5 data into memory for faster access
def load_h5_data(data):
    out = dict()
    for k in data.keys():
        if isinstance(data[k], h5py.Dataset):
            out[k] = data[k][:]
        else:
            out[k] = load_h5_data(data[k])
    return out

class ManiSkillTrajectoryDataset(Dataset):
    """
    A general torch Dataset you can drop in and use immediately with just about any trajectory .h5 data generated from ManiSkill.
    This class simply is a simple starter code to load trajectory data easily, but does not do any data transformation or anything
    advanced. We recommend you to copy this code directly and modify it for more advanced use cases

    Args:
        dataset_file (str): path to the .h5 file containing the data you want to load
        load_count (int): the number of trajectories from the dataset to load into memory. If -1, will load all into memory
        success_only (bool): whether to skip trajectories that are not successful in the end. Default is false
        device: The location to save data to. If None will store as numpy (the default), otherwise will move data to that device
    """

    def __init__(self, dataset_file: str, load_count=-1, success_only: bool = False, device = None) -> None:
        self.dataset_file = dataset_file
        self.device = device
        self.data = h5py.File(dataset_file, "r")
        json_path = dataset_file.replace(".h5", ".json")
        self.json_data = load_json(json_path)
        self.episodes = self.json_data["episodes"]
        self.env_info = self.json_data["env_info"]
        self.env_id = self.env_info["env_id"]
        self.env_kwargs = self.env_info["env_kwargs"]

        self.obs = None
        self.actions = []
        self.terminated = []
        self.truncated = []
        self.success, self.fail, self.rewards = None, None, None
        if load_count == -1:
            load_count = len(self.episodes)
        for eps_id in tqdm(range(load_count)):
            eps = self.episodes[eps_id]
            if success_only: 
                assert "success" in eps, "episodes in this dataset do not have the success attribute, cannot load dataset with success_only=True"
                if not eps["success"]:
                    continue
            trajectory = self.data[f"traj_{eps['episode_id']}"]
            trajectory = load_h5_data(trajectory)
            eps_len = len(trajectory["actions"])
            
            # exclude the final observation as most learning workflows do not use it
            obs = common.index_dict_array(trajectory["obs"], slice(eps_len))
            if eps_id == 0:
                self.obs = obs
            else:
                self.obs = common.append_dict_array(self.obs, obs)
[... 58 lines cut ...]
            if self.fail is not None:
                self.fail = sapien_utils.to_tensor(self.truncated, device=device)
    def __len__(self):
        return len(self.actions)

    def __getitem__(self, idx):
        action = sapien_utils.to_tensor(self.actions[idx], device=self.device)
        obs = common.index_dict_array(self.obs, idx, inplace=False)

        res = dict(
            obs=obs,
            action=action,
            terminated=self.terminated[idx],
            truncated=self.truncated[idx],
        )
        if self.rewards is not None:
            res.update(reward=self.rewards[idx])
        if self.success is not None:
            res.update(success=self.success[idx])
        if self.fail is not None:
            res.update(fail=self.fail[idx])
        return res
```

Both `obs` and `env_states` in each trajectory do not contain any specific obs keys, but rather concatenated numpy arrays.

branyang02, 2025-07-16:
TLDR: Can't get `sensor_data.hand_camera.Color` when replaying from RL based demos.

Hi,

Yep I was using an outdated file instead of the new replay trajectory tool. 

I am in progress on porting the replay trajectory tool as a script to generate LeRobotDataset formatted Maniskill Demos. It works perfectly great for motionplanning demos. Basically, I am using `obs_mode: str = "sensor_data"` to get all camera data from the replayed datasets, and using those as `obs` for LeRobot. Here's my LeRobot dataset setup for motionplanning demos:

```
DATASET_FEATURES = {
    "observation.images.base_camera": {
        "dtype": "video",
        "shape": (128, 128, 3),
        "names": ["height", "width", "channels"],
        "info": None,
    },
    "observation.images.hand_camera": {
        "dtype": "video",
        "shape": (128, 128, 3),
        "names": ["height", "width", "channels"],
        "info": None,
    },
    "observation.state": {
        "dtype": "float32",
        "shape": (18,),
        "names": [
            "s_0",
            "s_1",
            "s_2",
            "s_3",
            "s_4",
            "s_5",
            "s_6",
            "s_7",
            "s_8",
            "s_9",
            "s_10",
            "s_11",
            "s_12",
            "s_13",
            "s_14",
            "s_15",
            "s_16",
            "s_17",
        ],
    },
    "action": {
        "dtype": "float32",
        "shape": (8,),
        "names": [
            "a_0",
            "a_1",
            "a_2",
            "a_3",
            "a_4",
            "a_5",
            "a_6",
            "a_7",
        ],
    },
}

```

where the `state = concat(agent.qpod, agent.qvel)`.

However, when using rl based demos (such as `LiftPegUpright-v1/rl/trajectory.none.pd_ee_delta_pose.physx_cuda.h5`), I am not getting and robot joint camera (`hand_camera`). 

This is what is returned as `obs` when I try to step through the replayed demo:

```
agent.qpos: Tensor, shape=(1, 9), dtype=torch.float32, device=cuda:0
agent.qvel: Tensor, shape=(1, 9), dtype=torch.float32, device=cuda:0
extra.tcp_pose: Tensor, shape=(1, 7), dtype=torch.float32, device=cuda:0
sensor_param.base_camera.extrinsic_cv: Tensor, shape=(1, 3, 4), dtype=torch.float32, device=cuda:0
sensor_param.base_camera.cam2world_gl: Tensor, shape=(1, 4, 4), dtype=torch.float32, device=cuda:0
sensor_param.base_camera.intrinsic_cv: Tensor, shape=(1, 3, 3), dtype=torch.float32, device=cuda:0
sensor_data.base_camera.Color: Tensor, shape=(1, 128, 128, 4), dtype=torch.uint8, device=cuda:0
sensor_data.base_camera.PositionSegmentation: Tensor, shape=(1, 128, 128, 4), dtype=torch.int16, device=cuda:0
```

I was hoping that I could also extract the `hand_camera` for rl based demos. I'm not sure if this is by design, or if there is a way to retrieve that. 

Thanks for all the help!